INTI-CMNB/KiBot #184: Transitive 'extends' (inherit options of parents parents)

richardbarlow, 2022-04-14:
It looks like the `extends` feature currently only inherits the options from the direct parent (i.e. the output named in the `extends` field). If the parent also `extends` a third output, these options are not copied across. It would be helpful if the inheritance was transitive.

By example:

```yaml
- name: "base"
  type: render_3d
  run_by_default: false
  options:
    width: 2000
    height: 1000
- name: "top"
  type: render_3d
  extends: "base"
  options:
    rotate_x: 3
    rotate_z: -2
- name: "bottom"
  type: render_3d
  extends: "top"
  options:
    view: "bottom"
```

I would expect that the `bottom` output has the fields `width`, `height`, `rotate_x`, `rotate_y` and `view`. However, it looks like it does not have `width` and `height`.

set-soft, 2022-04-14:
Hi @richardbarlow !
Note that if you generate `top` and then `bottom` it will work as expected.

set-soft, 2022-04-14:
Now you can just ask for `bottom` without generating `top` and will work.
Also: now if you try to extend different output types you'll get an error.

commit 8c1d6e53 (2022-04-14): Now an output can extend another output that extends ...

- Even when the other outputs aren't yet configured
- Also added an error when the types doesn't match

Closes #184
closed 2022-04-14 by set-soft