INTI-CMNB/KiBot #185: pdf_pcb_print title append accumulates

richardbarlow, 2022-04-14:
If there is more than one `pdf_pcb_print` output and they append something to the title, it looks as if the appended values accumulate (so that latter output has both its and the previous outputs text appended). Example:

```yaml
  - name: "assy_front_pdf"
    type: pdf_pcb_print
    layers:
      - Edge.Cuts
      - F.Courtyard
      - F.Fab
    options:
      title: "+ - Front"
      output: "%f-assy-front_%sr.%x"

  - name: "assy_back_pdf"
    type: pdf_pcb_print
    layers:
      - Edge.Cuts
      - B.Courtyard
      - B.Fab
    options:
      title: "+ - Back"
      mirror: true
      output: "%f-assy-back_%sr.%x"

```

The PDF produced by `assy_front_pdf` has the title "Original - Front", the PDF produced by `ass_back_pdf` has the title "Original - Front - Back".

set-soft, 2022-04-14:
Thanks for reporting, the code to restore the title had a line in the wrong place.
Now should work

closed 2022-04-14 by set-soft

richardbarlow, 2022-04-14:
Awesome. Thanks for fixing it so quickly.

I intend on moving our workflow over to `pcb_print` as soon as we have a bit of spare time.

commit 0e8e697d (2022-04-14): Fixed title not restored in print functions

Fixes #185