INTI-CMNB/KiBot #186: STEP generation does not work with project-relative 3d model paths

richardbarlow, 2022-04-14:
Using KiCAD v6 and a project that has relative paths to 3d models, the STEP generation doesn't work properly. It looks like KiBot is taking a copy of the `.kicad_pcb` file and this is breaking the relative paths:

```
Could not add 3D model to C50.
File not found: ${KIPRJMOD}/../lib/gaitpace.pretty/C_0402_1005Metric.step
Searched paths:
    /tmp/lib/gaitpace.pretty/C_0402_1005Metric.step
```

set-soft, 2022-04-14:
Hi @richardbarlow !
Are you sure you are using the current code? Please post the debug log.

STEPOptions inherits from Base3DOptions, which defines filter_components. This function stores the PCB using save_tmp_board, unlike save_tmp_dir_board this function creates a temporal .kicad_pcb file in the same place as the original .kicad_pcb

richardbarlow, 2022-04-14:
Ah yes, I can actually see in the debug output that KiBot appears to be replacing the relative paths. However, as you can then see a bit later in the log, the STEP export still sees the relative paths and breaks.

[kibot_step_log.txt](https://github.com/INTI-CMNB/KiBot/files/8489738/kibot_step_log.txt)

set-soft, 2022-04-14:
This message is from `kicad2step`, not from KiBot. I'll take a look at this latter, but the problem is in KiCad.

closed 2022-04-18 by set-soft

set-soft, 2022-04-18:
Hi @richardbarlow !

I found the problem, lamentably your log is incomplete. I think you first generated another output that saved the PCB to `/tmp`. Then KiBot recycled the PCB (just removing any changes) and went ahead to generate the new output (the problematic STEP).

But what I didn't know is that:
1. KiCad's Python API changes `KIPRJMOD` when we call `BOARD.Save()`. So saving to `/tmp` makes KiCad define `KIPRJMOD` to `/tmp`
2. `kicad2step` can be fooled defining `KIPRJMOD` in the environment
3. The way KiCad's Python API does it (directly calling the OS, not using Python) makes this change unnotizable, if you ask Python it will say nothing changed.

Now KiBot forces `KIPRJMOD` to point to the dir where the PCB is, no matter what Python says about `KIPRJMOD`.

commit 36987cfc (2022-04-18): Fixed kicad2step not loading 3D models (bizarre case)

- If we saved the PCB to /tmp and then we reverted the changes to
  use the PCB the KIPRJMOD environment var pointed to the temporal.
- KiCad defines it during Save and Python becomes unsync, so we
  can't even notice it!

Fixes #186