INTI-CMNB/KiBot #187: pcbdraw shows a weird artifact

KyleHunter, 2022-04-15:
When using the following:
`  - name: pcb_top
    comment: "Renders - top"
    type: pcbdraw
    dir: "Renders"
    options:
      format: png
      show_components: all`

I get this:

[
[image]
](url)

Changing the format to jpg, show_components: none does not help. The STEP export and pdf_pcb_print on this PCB work perfectly as well

nerdyscout, 2022-04-15:
I had some weird things using pcbdraw when a circular PCB got not working, but a PCB of 4x 1/4 arcs did. As you do have rounded corners I assume it is something similar. therefor I assume it is not an issue of kibot, but pcbdraw.

KyleHunter, 2022-04-15:
> 

Yes, you're right. I just tried with a rectangular PCB and it worked perfectly.

set-soft, 2022-04-18:
Hi @KyleHunter and @nerdyscout !

It would be nice to have a small test case that produces it. I tried a very naive one: [test.zip](https://github.com/INTI-CMNB/KiBot/files/8506930/test.zip)

And I got correct drawings:

[image]
[image]

KyleHunter, 2022-04-21:
@set-soft Got it. I'll see if I can't work one up to show it. Appreciate it!

set-soft, 2022-04-21:
Hi @KyleHunter !
A silly question: Are you sure your outline is perfectly closed? I mean: that each segment perfectly joins the next. Note that a small difference will make PcbDraw fail to construct an SVG patch that contains the board. In your image looks like 4 joints aren't perfect and you get a rectangular hole inside the board.

KyleHunter, 2022-04-28:
@set-soft Yeah, 100% sure. Really odd, I tried simplifying the design so I could share, but was struggling to reproduce it. It's still there on the actual design though. I'll see what I can manage to do.

set-soft, 2022-05-23:
Without further information I'll assume this is a problem when the PCB contour isn't closed.
This error could be fixed in PcbDraw, but it should be catched by KiCad.
I don't know if @yaqwsx has some comment about it.

closed 2022-05-23 by set-soft

set-soft, 2022-05-23:
Hi @yaqwsx !
What about some kind of *tolerance*, lets say 0.1 mm (configurable). If you don't find a matching segment end you look for one that is 0.1 mm (or less) closer.

set-soft, 2022-06-24:
Hi @KyleHunter !
This problem is fixed in my fork of PcbDraw: https://github.com/INTI-CMNB/PcbDraw/releases/tag/v0.9.0-2
Is a rounding issue in KiCad SVG generation.