INTI-CMNB/KiBot #189: setsoft/kicad_auto image - Wrong version for `pcbnew_do` (1.5.15), must be 1.6.7 or newer

mdeweerd, 2022-04-18:
Possibly a very recent update that results in "Wrong version for `pcbnew_do` (1.5.15), must be 1.6.7 or newer"
```plaintext
dev: Pulling from setsoft/kicad_auto
6ec8c9369e08: Already exists
62793550d258: Already exists
b3f8e4c43512: Pulling fs layer
b3f8e4c43512: Verifying Checksum
b3f8e4c43512: Download complete
b3f8e4c43512: Pull complete
Digest: sha256:45e82c3bf9c683626ab42456d1c3c000f973ecf1ebb90a10ce2de6c8d132d7bf
Status: Downloaded newer image for setsoft/kicad_auto:dev
docker.io/setsoft/kicad_auto:dev
[...]
- 'Export all production layers to PDF.' (pdf_pcb) [pdf_pcb_print]
ERROR:Wrong version for `pcbnew_do` (1.5.15), must be 1.6.7 or newer. (kibot - kiplot.py:108)
```

mdeweerd, 2022-04-18:
I tried  'latest' as well, getting other error.

```plaintext
latest: Pulling from setsoft/kicad_auto
Digest: sha256:6915d439e3e1d55b241d7cd107cd0fac6d53e0c97a6d9415c785653fbb2edc75
Status: Image is up to date for setsoft/kicad_auto:latest
docker.io/setsoft/kicad_auto:latest
ERROR:`import` items must be strings ({'file': 'LIBS/kibot/common.kibot.yaml'}) (kibot.kibot.error - error.py:40)
```

I tried quoting the file name, but that still results in the same error message.  I guess that this issue is actually fixed in dev.
```yaml
import:
  - file: LIBS/kibot/common.kibot.yaml
```

mdeweerd, 2022-04-18:
Tries 10.4-5.19, also issue with "items must be strings":

```plaintext
10.4-5.1.9: Pulling from setsoft/kicad_auto
c229119241af: Pulling fs layer
3f6708efe00c: Pulling fs layer
bf38200db55b: Pulling fs layer
c229119241af: Download complete
c229119241af: Pull complete
bf38200db55b: Verifying Checksum
bf38200db55b: Download complete
3f6708efe00c: Verifying Checksum
3f6708efe00c: Download complete
3f6708efe00c: Pull complete
bf38200db55b: Pull complete
Digest: sha256:8e73bd87bf13b460b568a3580ddbaf30031cbce953ebaf5a78b91401884660d7
Status: Downloaded newer image for setsoft/kicad_auto:10.4-5.1.9
docker.io/setsoft/kicad_auto:10.4-5.1.9
ERROR:`import` items must be strings ({'file': 'LIBS/kibot/common.kibot.yaml'}) (kibot.kibot.error - error.py:40)
```

set-soft, 2022-04-18:
Today setsoft/kicad_auto:latest (also ki6, 10.4-5.1.9 and ki6.0.4_Debian) were updated to KiAuto 1.6.10.

How do you get this error?

The following workflow passed all tests using latest and ki6: https://github.com/INTI-CMNB/KiBot/actions/runs/2186344949
It was less than half an hour ago.

set-soft, 2022-04-18:
I pulled `dev` and it contains 1.6.10:
```
$ ./run_shell_same_user_dev.sh
salvador@5a641d34e038:~$ eeschema_do --version
eeschema_do 1.6.10 - Copyright 2018-2022, INTI/Productize SPRL - License:
Apache 2.0
salvador@5a641d34e038:~$ exit
exit
```

set-soft, 2022-04-18:
> ERROR:`import` items must be strings ({'file': 'LIBS/kibot/common.kibot.yaml'})

What is this? Not part of KiBot repo, some action?

mdeweerd, 2022-04-18:
Same with 'setsoft/kicad_auto:nightly' (items must be strings).

I found the cause for the issue with 'setsoft/kicad_auto:dev', I was overriding KiAuto with my local git clone - removing the override fixed the initial issue, but I now get the trace below, which is related to kicost (disabling the kicost does not generate this trace=

```plaintext
- '' (kicost) [kicost]
Traceback (most recent call last):
  File "/usr/local/bin/kibot", line 16, in <module>
    main()
  File "/usr/local/lib/python3.7/dist-packages/kibot/__main__.py", line 398, in main
    generate_outputs(outputs, args.target, args.invert_sel, args.skip_pre, args.cli_order)
  File "/usr/local/lib/python3.7/dist-packages/kibot/kiplot.py", line 417, in generate_outputs
    run_output(out)
  File "/usr/local/lib/python3.7/dist-packages/kibot/kiplot.py", line 373, in run_output
    out.run(get_output_dir(out.dir, out))
  File "/usr/local/lib/python3.7/dist-packages/kibot/out_base.py", line 133, in run
    self.options.run(self.expand_filename(output_dir, self.options.output))
  File "/usr/local/lib/python3.7/dist-packages/kibot/out_kicost.py", line 156, in run
    GS.sch.save_netlist(f, self._comps, no_field=var_fields)
  File "/usr/local/lib/python3.7/dist-packages/kibot/kicad/v5_sch.py", line 2029, in save_netlist
    reparsed = minidom.parseString(rough_string)
  File "/usr/lib/python3.7/xml/dom/minidom.py", line 1968, in parseString
    return expatbuilder.parseString(string)
  File "/usr/lib/python3.7/xml/dom/expatbuilder.py", line 925, in parseString
    return builder.parseString(string)
  File "/usr/lib/python3.7/xml/dom/expatbuilder.py", line 223, in parseString
    parser.Parse(string, True)
xml.parsers.expat.ExpatError: not well-formed (invalid token): line 2, column 240
```

mdeweerd, 2022-04-18:
> > ERROR:`import` items must be strings ({'file': 'LIBS/kibot/common.kibot.yaml'})
> 
> What is this? Not part of KiBot repo, some action?

This is in my project's yaml file to include a yaml file I reuse accross projects:
```yaml
import:
  - file: LIBS/kibot/common.kibot.yaml
  - file: LIBS/kibot/JLCPCB_stencil.kibot.yaml
```

It's ok with docker image 'dev', but not in 'latest', 'nightly', ... .

mdeweerd, 2022-04-19:
Closing this because the original issue is fixed, and the latest, etc. images are older.

closed 2022-04-19 by mdeweerd

set-soft, 2022-04-19:
Also note that nightly is no longer updated. It will be updated when we approach 7.0 (in more than two years?)