INTI-CMNB/KiBot #191: 3D Model selection (continuation from #103)

mdeweerd, 2022-04-19:
I am trying to disable a model (case for the PCB).

I ran into 3 limitations:
1. The variant specified in the '%variant:<indexes>%' expression in PCBnew as a footprint property seems to be matched against the variant provided on the command line.  
   (I may be wrong, apparently none of my rules was matched, I am going to debug that).
   I expected this to follow the variant expression, specified in the filter:
```yaml
  - name: 5V_3btn
    comment: 5V power, 3 buttons per screen
    type: kicost
    file_id: _5V_3btn
    variant: ^(button3|screen5v|conn3|AD1)$
    pre_transform: kicost_rename
```
3. I ran into an exception when providing an empty list.
    With '%5V_3btn:100%' -> Exception.  With '%5V_3btn:100%' -> no exception.
   So I (try to) use the latter as a workaround.
4. If I never want to include a model unless the variant is matched, there is no simple method for it.  so '%5V_3btn:100%' excludes only for 5V_3btn, not for any other variant.
5. I did not find a working combination yet -> I will instrument the kibot code a bit to follow the flow, check what matching is tested.

```
Using command line value `5V_3btn` for global option `variant`
- '' (render_2d_top_black) [render_3d]
Traceback (most recent call last):
  File "/usr/local/bin/kibot", line 16, in <module>
    main()
  File "/usr/local/lib/python3.7/dist-packages/kibot/__main__.py", line 398, in main
    generate_outputs(outputs, args.target, args.invert_sel, args.skip_pre, args.cli_order)
  File "/usr/local/lib/python3.7/dist-packages/kibot/kiplot.py", line 417, in generate_outputs
    run_output(out)
  File "/usr/local/lib/python3.7/dist-packages/kibot/kiplot.py", line 373, in run_output
    out.run(get_output_dir(out.dir, out))
  File "/usr/local/lib/python3.7/dist-packages/kibot/out_base.py", line 133, in run
    self.options.run(self.expand_filename(output_dir, self.options.output))
  File "/usr/local/lib/python3.7/dist-packages/kibot/out_render_3d.py", line 174, in run
    board_name = self.filter_components()
  File "/usr/local/lib/python3.7/dist-packages/kibot/out_base_3d.py", line 240, in filter_components
    self.apply_variant_aspect()
  File "/usr/local/lib/python3.7/dist-packages/kibot/out_base_3d.py", line 207, in apply_variant_aspect
    slots = [int(v) for v in slots]
  File "/usr/local/lib/python3.7/dist-packages/kibot/out_base_3d.py", line 207, in <listcomp>
    slots = [int(v) for v in slots]
ValueError: invalid literal for int() with base 10: ''
```

set-soft, 2022-04-19:
> 1. The variant specified in the '%variant:%' expression in PCBnew as a footprint property ~seems to be~ is matched against the variant provided on the command line.
>    (I may be wrong, apparently none of my rules was matched, I am going to debug that).
>    I expected this to follow the variant expression, specified in the filter (models are mainly selected based on a function, not based on kibot's variant name):

This is now documented, the syntax is %VARIANT_NAME:SLOTS%. This isn't KiCost specific.

> I ran into an exception when providing an empty list.
With '%5V_3btn:100%' -> Exception. With '%5V_3btn:100%' -> no exception.
So I used the latter as a workaround.

I'm fixing it

> If I never want to include a model unless the variant is matched, there is no simple method for it.

I already told you to use a `default` variant. This *is* simple and forces KiBot to use the variants mechanism. Then  you can define what to do by default, i.e. don't use any model.

closed 2022-04-19 by set-soft

mdeweerd, 2022-04-19:
@set-soft

1. Thanks - I didn't catch the info about "default" in the past.
I just read the new documentation, thanks.  I suggest so write "a variant named 'default'" rather than "a default variant".

2. PCB filtering is of course not "kicost" specific, but the "variant" selection mecanism should IMHO be the same.  A kibot variant named "X" corresponds to several functionnalities/component selections which follow a certain selection method - in my case the KiCost variant filter.  
If I select the "left" LCD in variants X and Z, but not in variant Y, then I specify "left" in the expression for variant X and Z.  
The way it is implemented now, requires to add all related "KiBot variants" to the PCB footprint, and change that whenever a function is added or removed from a variant expression.

This is a "simple" example with 3 "kibot variants", and two component variants "only".

```
variants:
  - name: 5V_3btn
    type: kicost
    file_id: _5V_3btn
    variant: ^(button3|screen5_vleft|conn3|AD1)$
    pre_transform: kicost_rename
  - name: 
    type: kicost
    file_id: _5V_4btn
    variant: ^(button3|button4|screen5v_left|conn1|conn3|AD1)$
    pre_transform: kicost_rename
  - name: 
    type: kicost
    file_id: _5V_4btn
    variant: ^(button3|button4|screen3v_top|conn1|conn3|AD1)$
    pre_transform: kicost_rename
```

If somebody would be using "KiBOM", then the 'fit_field' mechanism would have to apply IMHO.

I probably could convert the kicost variant expression to a KiBOM specification and specify a comma-separated liste for KiBOM's `-r` or `-variant` option, but I do not think that this would change a lot with regards to the PCB model selection method currently proposed.

set-soft, 2022-04-20:
> * Thanks - I didn't catch the info about "default" in the past.
>   I just read the new documentation, thanks.  I suggest so write "a variant named 'default'" rather than "a default variant".

Not sure about it. You must create a *default* variant, is a convenience to name it `default`, but not mandatory. You can choose any name.

> * PCB filtering is of course not "kicost" specific, but the "variant" selection mecanism should IMHO be the same.  A kibot variant named "X" corresponds to several functionnalities/component [...]

You must take a look at how other variants work. I.e. the iBoM variants are quite simple and doesn't have anything to match.

>   The way it is implemented now, requires to add all related "KiBot variants" to the PCB footprint,

Most probably. 

> and change that whenever a function is added or removed from a variant expression.

Not really, unless you don't plan your variants.

We can add new mechanisms, but changing the current one to make a "match" will make it only usable for KiCost variants.
I could add something like `%_kicost.VARIANT:SLOTS%`

mdeweerd, 2022-04-20:
So the option:
```
global:
  variant: '5V_3btn'
```
So it is not just the 'default' variant, it is also applied when the variant is provided on the CLI?  I expected that the CLI option overrides that global option.
BTW, I had the above in the project's yaml, which for me was the default variant when not providing any CLI argument for the variant.

> Not really, unless you don't plan your variants.
Even when planning variants, options change over time due to different choices, debug, etc - other components for the options, or option removed/added from/to a variant.
So when a variant filter changes, one has to update the filters as well as well.

> will make it only usable for KiCost variants

It seems to me that the BaseVariant.filter child implementations all have some notion of variant name filtering, but do need some refactoring to make the variant test reusable, and providea new common method "filter_variant".  The default implementation (BaseVariant) could be to check for the Kibot variant label.  
What is valid in terms of unified filtering for the kicost filter is also valid for the kibom and ibom filters.

set-soft, 2022-04-20:
> It seems to me that the BaseVariant.filter child implementations all have some notion of variant name filtering, but do need some refactoring to make the variant test reusable, and providea new [...]

You assume that all people wants to use the KiCost regex to match the models, but I don't think so. And it makes things more obscure, for KiCost you get a regex match and the variant name is ignored, and for the rest the variant name is used. This is even more confusing. In many cases the variant name is more than enough. We could need more options, and this is what we have to figure out.

set-soft, 2022-04-20:
I added the `_kicost.TEXT` thing and also added `_default_`, this can be used to match when no other text matched. Note that you sitll need to use variants, is just to avoid repetitions when a lot of variants selects the same slots.

mdeweerd, 2022-04-20:
I "assume" that a user would like to have a unified selection mecanism.

KiBOM's "--variant" option allows you to do something similar "--variant button3,screen5_vleft,conn3,AD1" which uses all of the 'variants' listed in a single output file.  I'ld expect that to work the same way as for kicost's '^(button3|screen5_vleft|conn3|AD1)$' regex match.
The 'fit_field' (with KiBOM) can be "-production,+test"' but could very well be "+button3,+conn3" in which case I would find it acceptable to have '%+button3,+conn3:2,4%" as well .

Several targets already have an individual 'target' option that overrides the default variant.

Regarding the "default" model selection, I'ld propose to use the empty name - '%:SLOTS%'  - to be used if there is no matching '%VARIANT:SLOTS%'.

mdeweerd, 2022-04-20:
`%_kicost...%` and `%_kibom...%` offer a solution, thanks for that.  I would have gone with an extra method in the BaseVariant and children.  But let's not discuss this further.

> > Several targets already have an individual 'target' option that overrides the default variant.
> 
> I don't fully understand what you say.

For generating a bom, it is possible to add the 'variant: ' option to override the variant that was defined through the global option or the command line.  
For targets that allow it, that is a method to allow the user select a variant according to the target, and move between one filtering method to another.

> > Regarding the "default" model selection, I'ld propose to use the empty name - '%:SLOTS%' - to be used if there is no matching '%VARIANT:SLOTS%'.
> 
> I can add a case for this,
As far as I understand, you have added '%_default_:SLOTS%' for this.

set-soft, 2022-04-20:
> For generating a bom, it is possible to add the 'variant: ' option to override the variant that was defined through the global option or the command line.
> For targets that allow it, that is a method [...]

If you define `variant: ''` you disable the variants mechanism, unless you also have filters (which enables the filtering mechanism) you won't get any 3D model replacement because nothing is processed. 

I think I found a solution that will fit both ideas, the last patch implements:

- %VARIANT_NAME:SLOTS%
- $VARIANT_TEXT:SLOTS$

`VARIANT_TEXT` is what you put in the *variant field*. So now you can choose any of the mechanism without extra typing, just different delimiter.

commit bf73c118 (2022-04-19): Fixed problems when using %VARIANT:% in the 3D models variants

Fixes #191