INTI-CMNB/KiBot #192: Create no subfolder using the 'compress command'

ZOUDIO, 2022-04-20:
Currently the option 'dest' of command 'compress' specifies:
- dest: [string=''] Destination directory inside the archive, empty means the same of the file.

I would like to zip all files in my subfolder 'Gerbers', but now it generates a 'Gerbers' subfolder in my archive. Is there a way to avoid a subfolder, and have the files directly in the root of the zip?

Also, once the files are zipped up, the unpackaged files are redundant. Is there a command to clean this up, or should I script this externally?

set-soft, 2022-04-20:
@ZOUDIO please try the `move_files` new option.

closed 2022-04-20 by set-soft

ZOUDIO, 2022-04-21:
Thanks for the quick response. I should just be able to rebuild the container dev_k6 and have this new code right?

My current YAML looks like:
  - name: gerbers
    comment: Compress gerber files
    type: compress
    dir: Gerbers
    options:
      move_files: true
      files:
        - source: 'Gerbers/*'
        - dest: '/'

So I want to move all the files in the Gerber subfolder to a zip in the same subfolder, with the files in the root.
However, now with the move_files option I get a IsADirectoryError
Also, the files are still in a subdirectory. What could be the problem?

set-soft, 2022-04-21:
> Thanks for the quick response. I should just be able to rebuild the container dev_k6 and have this new code right?

No need to rebuild, just make sure you pull the last release (nothing to do if using CI/CD, docker pull if not)
 
> My current YAML looks like:
>
> * name: gerbers
>   comment: Compress gerber files
>   type: compress
>   dir: Gerbers
>   options:
>   move_files: true
>   files:
>   - source: 'Gerbers/*'
>   - dest: '/'
>
> So I want [...]

Please provide the whole config, I don't know what's inside `Gerbers`. I'll try to reproduce it.

set-soft, 2022-04-21:
> * name: gerbers
>   comment: Compress gerber files
>   type: compress
>   dir: Gerbers
>   options:
>   move_files: true
>   files:
>   - source: 'Gerbers/*'
>   - dest: '/'

Next time please paste the text inside a preformated block, like this:

```
- name: gerbers
  comment: Compress gerber files
  type: compress
  dir: Gerbers
  options:
    move_files: true
    files:
      - source: 'Gerbers/*'
        dest: '/'
```

The error in your config is that you are listing 2 different group of files, the first is for `Gerbers/*` (without destination) and the secons is for `*` (implicit) with `/` as destination. The correct way is the one used above.

set-soft, 2022-04-21:
Now you won't have problems to remove dirs.

commit 5d9b047d (2022-04-20): Added an option to remove the files we compressed.

Closes #192
commit 10b67f48 (2022-04-21): Fixed problems with compress.move_files

- When trying to remove directories

Fixes #192