INTI-CMNB/KiBot #193: KiBot fails on KiCad 6 with a symbol with a pin having an alternative assignment

KyleHunter, 2022-04-21:
Pretty much the title.

A schematic with this symbol:

[image]

Will fail like this:

`ERROR:Unknown pin attribute `[Symbol('alternate'), '/WR', Symbol('input'), Symbol('line')]` (kibot - kiplot.py:262)`

This is on the initial running on KiBot, not when generating any specific output.

set-soft, 2022-04-21:
Thanks @KyleHunter !

Too much undocumented features ... we must find them with the use.

closed 2022-04-21 by set-soft

KyleHunter, 2022-04-21:
@set-soft Awesome, thanks! Trust me, I'm going to have a lotttt more feedback coming up. Switching over my company to using KiBot to handle our output generations.

commit 7ac65452 (2022-04-21): Added support for KiCad 6 pin alternates

- Another undocumented feature

Fixes #193