INTI-CMNB/KiBot #196: ZIP command fails if target directory includes date formatter

Cushychicken, 2022-05-11:
I used the stock PCBway example (the one at [docs/samples/PCBWay.kibot.yaml](https://github.com/INTI-CMNB/KiBot/blob/master/docs/samples/PCBWay.kibot.yaml)) to create a kibot.yaml config file, which worked with no problems.  

I was able to modify the output directory for `PCBWay_ by modifying `dir: fab/%D/PCBway/gerbers` like the following to match our internal fabrication output directory structure:
```yaml
  - name: PCBWay_gerbers
    comment: Gerbers compatible with PCBWay
    type: gerber
    dir: "fab/%D/PCBWay/gerbers"
```

That also worked with no issues.

However, modifying the PCBWay_zip command to include a date stamp proves problematic. The following YAML produces a failure to ZIP:

```yaml
  - name: PCBWay_gerbers
    comment: Gerbers compatible with PCBWay
    type: gerber
    dir: "fab/PCBWay/gerbers"
```

...with the following error output:

```shell
Using config file: config.kibot.yaml
- 'Print schematic (PDF)' (print_sch) [pdf_sch_print]
- 'Interactive Bill of Materials (HTML)' (interactive_bom) [ibom]
- 'Print F.Cu+F.SilkS+Dwgs.User' (print_front) [pdf_pcb_print]
- 'Print B.Cu+B.SilkS+Dwgs.User' (print_bottom) [pdf_pcb_print]
- 'Gerbers compatible with PCBWay' (PCBWay_gerbers) [gerber]
- 'Drill files compatible with PCBWay' (PCBWay_drill) [excellon]
- 'ZIP file for PCBWay' (PCBWay_zip) [compress]
ERROR:Unable to generate `/home/user/share/quad-uart/fab/%D/PCBWay/gerbers/quad_uart.gtl` from 'Gerbers compatible with PCBWay' (PCBWay_gerbers) [gerber] (kibot.kibot.out_compress - out_compress.py:142)
```
Backing out the `%D` formatter from the file path allows this to run to completion. 

Are we allowed to use date and time formatters? We try to include these by default in our zip directories for ease of determining which version was sent to fabrication. It's not a huge issue if we have to omit it or add it manually.

Cushychicken, 2022-05-11:
My goal is to create a file tree exactly like the one you showed in your last comment. (I find it easier to keep track of versions submitted to vendor when you add a date.)

I copied the yaml you provided and got the same error. I suspect this might be a dumb YAML formatting issue - maybe some tabs that should be spaces on my end. Here's my full YAML file:

```yaml
# Gerber and drill files for PCBWay, with stencil (solder paste)
# URL: https://www.pcbway.com
# Based on setting used by Gerber Zipper (https://github.com/g200kg/kicad-gerberzipper)
kibot:
  version: 1

preflight:
  run_erc: true
  run_drc: true
  update_xml: true
  check_zone_fills: false
  ignore_unconnected: false

outputs:
  - name: 'print_sch'
    comment: "Print schematic (PDF)"
    type: pdf_sch_print
    dir: pdf
    options:
      output: "Schematic_%f.pdf"
  
  - name: 'print_pcb_top'
    comment: "Print F.Cu+F.SilkS+Dwgs.User"
    type: pdf_pcb_print
    dir: pdf
    options:
      output_name: "PCB_Top_%f.pdf"
    layers:
      - layer: F.Cu
      - layer: F.SilkS
      - layer: Edge.Cuts
      - layer: Dwgs.User

  - name: 'print_pcb_bottom'
    comment: "Print B.Cu+B.SilkS+Dwgs.User"
    type: pdf_pcb_print
    dir: pdf
    options:
      output_name: "PCB_Bottom_%f.pdf"
    layers:
      - layer: B.Cu
      - layer: B.SilkS
      - layer: Edge.Cuts
      - layer: Dwgs.User

  - name: 'bom_csv'
    comment: "Bill of Materials in CSV format"
    type: bom
    dir: bom 
    options:
      format: CSV  # HTML or CSV

  - name: 'interactive_bom'
    comment: "Interactive Bill of Materials (HTML)"
    type: ibom
    dir: bom
    options:
      blacklist: 'DNF*'
      name_format: '%f_%r_iBoM'

  - name: PCBWay_gerbers
    comment: "Gerbers compatible with PCBWay"
    type: gerber
    dir: "fab/%D/PCBWay/gerbers"
    options: &gerber_options
      exclude_edge_layer: true
      exclude_pads_from_silkscreen: true
      plot_sheet_reference: false
      plot_footprint_refs: true
      plot_footprint_values: true
      force_plot_invisible_refs_vals: false
      tent_vias: true
      use_protel_extensions: true
      create_gerber_job_file: false
      output: "%f.%x"
      gerber_precision: 4.6
      use_gerber_x2_attributes: false
      use_gerber_net_attributes: false
      disable_aperture_macros: true
      line_width: 0.1
      subtract_mask_from_silk: false
      inner_extension_pattern: '.gl%N'
    layers:
      - copper
      - F.SilkS
      - B.SilkS
      - F.Mask
      - B.Mask
      - F.Paste
      - B.Paste
      - Edge.Cuts

  - name: PCBWay_drill
    comment: "Drill files compatible with PCBWay"
    type: excellon
    dir: "fab/%D/PCBWay/drill"
    options:
      metric_units: false
      minimal_header: true
      zeros_format: SUPPRESS_LEADING
      # left_digits: 3
      # right_digits: 3
      # See https://github.com/INTI-CMNB/kicad-ci-test-spora/issues/1
      # and https://docs.oshpark.com/design-tools/gerbv/fix-drill-format/
      left_digits: 2
      right_digits: 4
      pth_and_npth_single_file: false
      pth_id: ''
      npth_id: '-NPTH'
      output: "%f%i.drl"

  - name: 'PCBWay_pnp'
    comment: "Pick and place file for PCBWay"
    type: position
    dir: "fab/PCBWay/pnp"
    options:
      format: ASCII   # CSV or ASCII format
      units: millimeters  # millimeters or inches
      separate_files_for_front_and_back: true
      only_smd: true

  - name: PCBWay_zip
    comment: "ZIP file for PCBWay"
    type: compress
    dir: fab
    options:
      output: gerbers-%D.%x
      format: ZIP
      files:
        - from_output: PCBWay_gerbers
          dest: /
        - from_output: PCBWay_drill
          dest: /
```

set-soft, 2022-05-12:
I tried your file using KiBot 1.0.0 and the light_control example and it worked.
Which KiBot are you using?

set-soft, 2022-05-14:
Hi @Cushychicken !  
Did you solve it?

Cushychicken, 2022-05-14:
No, I'm afraid not. Wondering what else I can try - I've copy/pasted the
YAML you provided here, done search and replace for tabs to spaces, and a
few other things, none of which have produced results.

Do you have any suggestions? Would any more info about my system be useful?

set-soft, 2022-05-14:
Hi @Cushychicken !

> No, I'm afraid not. Wondering what else I can try - I've copy/pasted the YAML you provided here, done search and replace for tabs to spaces, and a few other things, none of which have produced results.

> Do you have any suggestions? Would any more info about my system be useful?

Are you running KiBot locally (in a Linux machine) or in the GitHub servers?
If you are using a local copy please run:

```
kibot --version
```

If you are using a docker image you can add `-vvv` option to the line that invokes KiBot. I remmember you asked which docker images are good for KiCad 5 and I told you that `latest` and `dev` can be used. Are you using one of them?

This is the example I used for KiCad 5, try running KiBot on the directory you get after uncompressing this file: [kicad_5.zip](https://github.com/INTI-CMNB/KiBot/files/8693660/kicad_5.zip)

If you already migrated to KiCad 6 you can use: 
[kicad_6.zip](https://github.com/INTI-CMNB/KiBot/files/8693661/kicad_6.zip)

set-soft, 2022-08-12:
Hi @Cushychicken !
Without more information I can't reproduce the problem. I'm closing this issue. If you find this problem again, and have some time to find the origin, please open a new issue.

closed 2022-08-12 by set-soft