INTI-CMNB/KiBot #201: [BUG] Component with reverse mount LED footprint excluded from position file by `_mechanical` built in filter.

petejohanson, 2022-05-25:
Fill as much as you can, remove the rest

**Describe the bug**

While generating a position file for a project, one component, using footprint `LED_SMD:LED_1206_3216Metric_ReverseMount_Hole1.8x2.4mm` is being excluded from the position file because the default `_mechanical` filter includes a filter for `mount.*hole` which matches the part name.

As a result, the position file is missing this component (even though the BOM properly lists it)

**To Reproduce**

Ran kibot for my OSH repo at https://github.com/petejohanson/zmk-uno/

**Expected behavior**

A better regex is used to exclude mounting holes, but not reverse mount LEDs that have a hole.

**Screenshots**
N/A

**Environment (please complete the following information):**
Where are you running KiBot:

Run locally from docker image, or using the GH Actions Ci I linked.

**Additional context**
N/A

petejohanson, 2022-05-25:
I see this is actually fixed in `master`, just not released yet. I'll keep my own custom exclude filter or now, and remove it once that's released.

Thanks for the great project.

set-soft, 2022-05-26:
Hi @petejohanson !
The patch was included in the last release, but fixed only the 'Part' name, not the footprint. Now should be fixed.
Thanks for reporting!

petejohanson, 2022-05-26:
Ah, yes, I see that. Thanks!

Side note: this project is dangerous, it removes way too much of the friction from ordering follow up prototypes as you iterate on a design.

Should come with a giant warning label.