INTI-CMNB/KiBot #202: [BUG] Missing dependency: psutil

dbriscoe511, 2022-05-25:
Fill as much as you can, remove the rest

**Describe the bug**
A clear and concise description of what the bug is.

Quering the version of pcbnew_do requires module psutil, which is not a dependency. once it is intalled with pip the script works as expected

**To Reproduce**
Command line options passed to KiBot, configuration files (.kibot.yaml), PCB/Schematic example to reproduce it (Ex: a link to a repo where the files are)
- unistall psutil
- create a IPC netlist file in verbose mode
```
- 'IPC-D-356 netlist file' (Supportboard_d356) [netlist]
DEBUG:Output destination: ************* (kibot - kiplot.py:358)
DEBUG:Running: ['pcbnew_do', '--version'] (kibot - kiplot.py:105)
ERROR:Unable to determine pcbnew_do version:
 (kibot - kiplot.py:111)
>pcbnew_do --version                      4 ✘  3s  
Traceback (most recent call last):
  File "/usr/bin/pcbnew_do", line 31, in <module>
    import psutil
ModuleNotFoundError: No module named 'psutil'
 >pip install psutil                              1 ✘ 
Defaulting to user installation because normal site-packages is not writeable
Collecting psutil
  Downloading psutil-5.9.1-cp310-cp310-manylinux_2_12_x86_64.manylinux2010_x86_64.manylinux_2_17_x86_64.manylinux2014_x86_64.whl (282 kB)
     |████████████████████████████████| 282 kB 3.7 MB/s            
Installing collected packages: psutil
Successfully installed psutil-5.9.1
>pcbnew_do --version                               ✔ 
pcbnew_do 1.6.10 - Copyright 2018-2022, INTI/Productize SPRL - License: Apache 2.0

```

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
Kibot running in Manjaro stable 5.15, kicad version 6.0.5

**Additional context**
Add any other context about the problem here.

dbriscoe511, 2022-05-25:
Also, you git page is missing a donation link :+1:

set-soft, 2022-05-26:
Hi @dbriscoe511 !

I updated the KiAuto README.md to include the Python modules. They are installed by `pip`, but I forgot to mention them in the installation instructions. Thanks for pointing this.

Currently the only donation mechanism is using PayPal, the button close to the project stats: [image]
Lamentably GitHub mechanism is not supported for my country.

closed 2022-05-26 by set-soft