INTI-CMNB/KiBot #203: [BUG] Fields incorrectly grouped together when a '-' is in the field name

dbriscoe511, 2022-05-25:
Fill as much as you can, remove the rest

**Describe the bug**
A clear and concise description of what the bug is.

Parts are grouped together on the BOM sometimes that have different vales, seemingly when there is a dash in the name.

The dash almost seems like it is act as an escape character during grouping. 

Parts J7,J8 were grouped together with values M12A-08PMMP-SF8002 and M12A-08PFFP-SF8001
When the "group feilds:" line was not present kibot gave a warning that there was a field conflict of their MPNs. when MPN was added as a filter option, the warning went away, but the two parts were still grouped together.

Changing the value of J8 to aM12A-08PMMP-SF8002 allowed the parts to group together. 

**To Reproduce**
Command line options passed to KiBot, configuration files (.kibot.yaml), PCB/Schematic example to reproduce it (Ex: a link to a repo where the files are)
```
  - name: *****_bom
    comment: BoM for ********8
    type: bom
    dir: Manufacturing/Assembly
    options:
      output: '%f_%r_MFG_BOM.%x'
      ref_separator: ','
      group_fields: ['Part', 'Part Lib', 'Value', 'Footprint', 'Footprint Lib', 'Voltage', 'Tolerance', 'Current', 'Power','MPN']
      #ignore_dnf: false
      columns:
      - field: References
        name: Ref
      - field: Quantity Per PCB
        name: Qnty
      - MPN
      - field: ALT_PART_NUMBERS
        name: Alternate_MPN(s)
      - field: MAUNFACTURER
        name: Manufacturer
      - Value
      - Footprint
      - Description
      xlsx: 
        generate_dnf: true
        logo: false
        title: '%f Bill of Materials'
        extra_info: '********'
```

**Environment (please complete the following information):**
Where are you running KiBot:
```

KiBot installation checker

Core:
Linux: 5.15.41.1 (Linux danie-alienwareaurorar6 5.15.41-1-MANJARO #1 SMP PREEMPT Wed May 18 09:20:01 UTC 2022 x86_64 GNU/Linux)
Python: 3.10.4 (main, Mar 23 2022, 23:05:40) [GCC 11.2.0]
KiCad: (6.0.5)
Kibot: 1.1.0

Modules:
Colorama: 0.4.4
Distutils: 3.10.4
LXML: 4.8.0
PyYAML: 6.0.0 (6.0)
QRCodeGen: Ok
Requests: 2.26.0
XLSXWriter: 3.0.3

Tools:
Ghostscript: Not available
Git: 2.36.1
ImageMagick: 7.1.0.35 (Version: ImageMagick 7.1.0-35 Q16-HDRI x86_64 20062 https://imagemagick.org)
Interactive HTML BoM: Not available
KiBoM: 1.8.0 (KiBOM Version: 1.8.0)
KiCad Automation tools: 1.6.11 (1.6.11 - Copyright 2018-2022, INTI/Productize SPRL - License: Apache)
KiCost: 1.1.8 (KiCost v1.1.8)
Pandoc: Not available
PcbDraw: 0.6.1 (PcbDraw 0.6.1)
RAR: Not available
RSVG tools: 2.54.3
```

set-soft, 2022-05-26:
Hi @dbriscoe511 !
I'm not sure about the problem, I failed to reproduce it, here is what I tried: [dash_test.zip](https://github.com/INTI-CMNB/KiBot/files/8779544/dash_test.zip)

Running kibot on the `dash_test` folder I get:

```csv
Row,Description,Part,References,Value,Footprint,Quantity Per PCB,Status,Datasheet,MPN
1,8-pin DIN connector,DIN-8,J7,,,1, ,http://www.mouser.com/ds/2/18/40_c091_abd_e-75918.pdf,M12A-08PMMP-SF8001
2,8-pin DIN connector,DIN-8,J8,,,1, ,http://www.mouser.com/ds/2/18/40_c091_abd_e-75918.pdf,M12A-08PMMP-SF8002
```

What do you get? Can you adapt this small test to get the same issue?

set-soft, 2022-08-12:
Hi @dbriscoe511 !
I'm closing this issue because I can reproduce it. If you have a small test example to reproduce it please open a new issue.