INTI-CMNB/KiBot #204: [HOW] Add multiple 3d-packages search paths

slimcdk, 2022-05-28:
**What do you want to achieve?**
To configure multiple search paths for 3d-packages - like you do through the KiCAD GUI. I have a couple of libraries that do not place 3d-packages in the common KiCAD 3d-package directory.

**Do you have some PCB/Schematic to use as example?**
This is my project repo https://github.com/slimcdk/esp-stepper-servo
This is the libraries I'm including https://github.com/slimcdk/slimc-kicad-libraries

**Do you have some configuration file (.kibot.yaml) that you are using?**
My kibot config is composed by multiple files, but this sums a complete config for exporting a step file and a single image
```yaml
kibot:
  version: 1
  
preflight:
  run_erc: false
  run_drc: false
  check_zone_fills: true
  ignore_unconnected: true
  set_text_variables:
    - name: date
      command: git log -1 --format='%as'
    - name: commit
      command: git log -1 --pretty=format:%h

global:
  output: "%f-%i.%x"

outputs:
  - name: step
    type: step
    dir: exports
    options:
      metric_units: true
      
  - name: render top
    type: render_3d
    dir: exports/renders
    options:
      height: 1000
      width: 1600
      zoom: 2
      download: true
      ray_tracing: true
      view: top
```

**Environment (please complete the following information):**
Where are you running KiBot:
Locally on Ubuntu 22.04 with KiCAD 6 and KiBot 1.1.0 and on GHA with latest Ubuntu and KiBot v2_k6 Action

set-soft, 2022-05-29:
Hi @slimcdk !

On the KiCad side I think the best you can do is to use relative paths. The `KIPRJMOD` variable always contains the directory where the KiCad project is located. So you can use it as a base, then you can indicate a model as `${KIPRJMOD}/../lib/3D/R_0603_1608Metric.wrl`.

On the git repo side you can use submodules, as you are doing in https://github.com/slimcdk/slimc-kicad-libraries 
Just add them to https://github.com/slimcdk/esp-stepper-servo then you can add models like `${KIPRJMOD}/../gekkio-kicad-libs/Gekkio_Button_Switch_THT.3dshapes/E-Switch_RA12131100.wrl`
Or put the submodules in `/libs` and then use something like `${KIPRJMOD}/../libs/gekkio-kicad-libs/Gekkio_Button_Switch_THT.3dshapes/E-Switch_RA12131100.wrl`

Another mechanism: You can copy all the models you need to one folder. Take them from KiCad libs and from the libs you use. Then make `KICAD6_3DMODEL_DIR` point there. I know it looks rigid, but this will ensure your project is reproducible. You usually want to get consistent results, even if KiCad changes the models. Of course, fixes to the models needs updates to the models you collected. But this also allows you to customize the models with details specific for your project. At work we do it for symbols, footprints and 3D models. In this way projects becomes self-contained, you can send them to clients (we always provide the source design to our clients) and they have all the needed stuff. KiCad libs seems to be stable now, but they changed really often creating very bad surprises.

closed 2022-05-29 by set-soft

slimcdk, 2022-06-06:
Hi! 

I'm already including my library as a submodule in the directory `dependencies` as you suggest. I've tried including the `KIPRJMOD` env variable as part of my env variable to the 3d-packages, but that does not seem to yield anything. If `SLIMC_3DMODELS` contains an absolutte path, everything works fine. This is probably an issues with KiCAD directly and not KiBot. I want my library to be standalone as much as possible so I can't add relative paths between footprints and 3d packages that take the project structure into account :/ 

I guess my fix will be an step in the action to temporarily change the absolute path for the env variables to match the action environment

[image]

[image]

set-soft, 2022-06-07:
> I want my library to be standalone as much as possible so I can't add relative paths between footprints and 3d packages that take the project structure into account :/

I'm not sure what you want is possible, take a look at this KiCad issue: https://gitlab.com/kicad/code/kicad/-/issues/3792
Using KiCad 6 I wasn't able to even uses the aliases from the GUI.

One possible solution is to create a library in the content manager format
1. Ask people to install `slimc` in the 3rd party folder. As other libs managed by the content manager. Take a look at libs managed by the content manager.
2. Use `${KICAD6_3RD_PARTY}/3dmodels/....` for the models in the lib
3. footprint and symbol aliases aren't an issue because they can use `${KIPRJMOD}` in the `*-lib-table` files.
4. On the repo workflows define an environment variable called KICAD6_3RD_PARTY pointing to `dependencies`. You can use symbolic links to create a suitable structure.

Of course: this solution isn't simple for somebody doing a git clone, the 3D models must be installed, even when they are also in the repo

set-soft, 2022-06-07:
> Thanks. I'll play around with the `KICAD6` env to see if I can make that work. Will it be feasible to place a `kicad_common.json` in the workflow container somehow? I can generate the necessary JSON to have the environment variables there, but I fail to make it available in the gh action kibot step.

Take a look at the new `environment` global option. You'll need to use the _dev actions.
 
> In the forth step I create the file and place it in `/github/home/.config/kicad/6.0` but KiBot won't recognize it https://github.com/slimcdk/esp-stepper-servo/runs/6765512933?check_suite_focus=true#step:6:16

You are placing the file in a container and running KiBot in another container. This won't work.

cdwilson, 2022-08-23:
> I've tried including the KIPRJMOD env variable as part of my env variable to the 3d-packages, but that does not seem to yield anything.

@slimcdk just FYI, you can get portable libraries working on the KiCad side of things by defining a 3D search path alias instead of an environment variable. Take a look at the screenshots I posted in https://github.com/INTI-CMNB/KiBot/discussions/261#discussion-4328338  (note that while this works in KiCad, I'm still not sure how to get it working in KiBot, hence the discussion post)