INTI-CMNB/KiBot #205: [BUG]Some K6 variablesare not passed for STEP file generation

mmutlu-any, 2022-05-29:
STEP file is not generated correctly if component 3D model uses `${KICAD_USER_LIBRARY_DIR}` or `${KICAD6_3RD_PARTY}` (K6 native variables). However, works well if `${KISYS3DMOD}` (K5 native variable) is used. All 3 variables point to the identical path. 
Tested on Kibot v1.1 using Kicad 6.0.5 on desktop Ubuntu 20.04. (Not CI/CD)

When step file is exported directly from K6, all variables work fine. It would be convenient if Kibot would work the same way too.  

```
Could not add 3D model to D1.
File not found: ${KICAD_USER_LIBRARY_DIR}/my-kicad-library/my-packages3d/Bar_Graph_5_Segment.step
Searched paths:
    /home/mmu/Desktop/my_project/ 
```

mmutlu-any, 2022-05-31:
Thanks a lot for immediate response @set-soft  !

I tested it. `KICAD_USER_LIBRARY_DIR` is found correctly but now `KISYS3DMOD` is lost. It is funny, because I am testing it with a simple PCB having only 2 components. At least the terminal is showing a warning where it was searched: 
```WARNING:(W047) Missing 3D model for J1: `/usr/share/kicad/3dmodels/ ....   ```
`KISYS3DMOD` is a K5 parameter. I think above path would have been the default after installation, but, in my case it is set to `/<my_library_path>/`.

Setting K5 parameters happened a couple of years ago. But, probably we were inspired from [https://forum.kicad.info/t/library-management-in-kicad-version-5/14636](https://forum.kicad.info/t/library-management-in-kicad-version-5/14636). It suggests not to touch `KISYS3DMOD`. At some point, KiCAD had some library restructuring (github vs gitlab etc.) and probably we ended up touching it. Most of my libraries generated in K5 are using `KISYS3DMOD`. It will probably change with K6. But changing the history retroactively takes longer time. 

Pictures from the link as quick reference: 
[image]
[image]

set-soft, 2022-06-01:
Hi @mmutlu-any !
Ok, so you are redefining `KISYS3DMOD` in the KiCad GUI? Right? (Not in the OS environment)
Try the last patch. It should work.
If it doesn't run KiBot with debug information (-vvv) and look for lines like this:

```
DEBUG:Using VARIABLE_NAME="VALUE" (FROM WHERE) (kibot - config.py:LINE)
```

And post them here.
I'm not really sure why the previous failed to export `KISYS3DMOD` to `kicad2step`. I tested the last patch with https://github.com/INTI-CMNB/KiBot/blob/master/tests/board_samples/kicad_6/bom.kicad_pcb and defining `KISYS3DMOD` in the `kicad_common.json` and it worked as spected.

mmutlu-any, 2022-06-01:
Yes, I modify KiCad variables only through KiCad GUI (Preferences -> Configure Paths).
`#24d81f4` worked! Thank you very much!

mmutlu-any, 2022-06-08:
I am pretty sure that it was working. Now, I tried again on another project and it did not work. I went back to the previous project and it does not work now. I am trying to understand why.  
Maybe I am doing something wrong to test. I installed kibot from debian package. I edited `/usr/lib/python3/dist-packages/kibot/kicad/config.py` file. First according to `#d93d705`, then changed to `#24d81f4`. 
Should I do anything else? 

Kicad GUI always displays 3D models fine. Kibot finds `KICAD6_3RD_PARTY` and `KICAD_USER_LIBRARY_DIR` variables correctly, but somehow cannot find the step files from my libraries. 
Terminal output from failing one (only user library components are not added correctly): 
```
DEBUG:Executing: ['kicad2step_do', '-r', '-vvv', '-o', '/<project_folder>/kibot_output/step/<output_name>.step', '-f', '-d', '/<project_folder>/kibot_output/step', '-vv', '--grid-origin', '/<project_folder>/<layout_file>.kicad_pcb'] (kibot - out_step.py:86)
DEBUG:Output from command:
DEBUG:Detected KiCad v6.0.5 (6.0.5-a6ca702e91~116~ubuntu20.04.1 6000005) (kicad2step_do - misc.py:143)
DEBUG:Config path /home/<user>/.config/kicad/6.0 (kicad2step_do - misc.py:159)
DEBUG:KiCad environment: {'KICAD6_3DMODEL_DIR': '/usr/share/kicad/3dmodels', 'KICAD6_3RD_PARTY': '/home/<user>/<path_to_lib>/library', 'KICAD6_FOOTPRINT_DIR': '/usr/share/kicad/footprints', 'KICAD6_SYMBOL_DIR': '/usr/share/kicad/symbols', 'KICAD_PTEMPLATES': '/home/<user>/<path_to_lib>/template', 'KICAD_USER_LIBRARY_DIR': '/home/<user>/<path_to_lib>/library', 'KICAD_USER_TEMPLATE_DIR': '/home/<user>/<path_to_lib>/template', 'KISYS3DMOD': '/usr/share/kicad/3dmodels'} (kicad2step_do - misc.py:245)
DEBUG:Output dir: /<project_folder>/kibot_output/step (kicad2step_do - kicad2step_do:116)
INFO:KiCad to STEP wrapper
DEBUG:Command: ['kicad2step', '-o', '/<project_folder>/kibot_output/step/<output_file>.step', '-f', '--grid-origin', '/<project_folder>/<layout_file>.kicad_pcb'] (kicad2step_do - kicad2step_do:143)
DEBUG:Redirecting kicad2step output to kicad2step*.log (kicad2step_do.kiauto.file_util - file_util.py:337)
DEBUG:Waiting for virtual X server ... (kicad2step_do.kiauto.ui_automation - ui_automation.py:88)
```
Lib aliases are added correctly.
```
DEBUG:Running: ['kicad2step_do', '--version'] (kibot - kiplot.py:105)
DEBUG:Reading KiCad config from `/home/<user>/.config/kicad/6.0/kicad_common.json` (kibot - config.py:260)
DEBUG:- KiCad var: KICAD6_3DMODEL_DIR="/usr/share/kicad/3dmodels" (kibot - config.py:251)
DEBUG:- KiCad var: KICAD6_3RD_PARTY="/home/<user>/<path_to_lib>/library" (kibot - config.py:251)
DEBUG:- KiCad var: KICAD6_FOOTPRINT_DIR="/usr/share/kicad/footprints" (kibot - config.py:251)
DEBUG:- KiCad var: KICAD6_SYMBOL_DIR="/usr/share/kicad/symbols" (kibot - config.py:251)
DEBUG:- KiCad var: KICAD_PTEMPLATES="/home/<user>/<path_to_lib>/template" (kibot - config.py:251)
DEBUG:- KiCad var: KICAD_USER_LIBRARY_DIR="/home/<user>/<path_to_lib>/library" (kibot - config.py:251)
DEBUG:- KiCad var: KICAD_USER_TEMPLATE_DIR="/home/<user>/<path_to_lib>/template" (kibot - config.py:251)
DEBUG:- KiCad var: KISYS3DMOD="/usr/share/kicad/3dmodels" (kibot - config.py:251)
DEBUG:Detected KICAD_SYMBOL_DIR="/usr/share/kicad/symbols" (kibot - config.py:273)
DEBUG:Detected KICAD_TEMPLATE_DIR="/usr/share/kicad/template" (kibot - config.py:283)
DEBUG:Exporting KISYS3DMOD="/usr/share/kicad/3dmodels" (kibot - config.py:298)
DEBUG:Exporting KICAD6_SYMBOL_DIR="/usr/share/kicad/symbols" (kibot - config.py:309)
DEBUG:Exporting KICAD6_3DMODEL_DIR="/usr/share/kicad/3dmodels" (kibot - config.py:317)
DEBUG:Exporting KICAD6_TEMPLATE_DIR="/usr/share/kicad/template" (kibot - config.py:325)
```
Later on the same terminal:
```
Add component R79.
Could not add 3D model to J1.
File not found: ${KICAD6_3RD_PARTY}/My_Connector.3dshapes/JST_PA_B13B-PA-SK_1x13_P2.00mm_Vertical.step
Searched paths:
    /<project_path>/
Add component R78.
Could not add 3D model to J2.
File not found: ${KICAD_USER_LIBRARY_DIR}/My_Connector.3dshapes/JST_PA_B13B-PA-SK_1x13_P2.00mm_Vertical.step
Searched paths:
    /<project_path>/
```
This line does not make sense to me because .step file is actually there: 
`File not found: ${KICAD_USER_LIBRARY_DIR}/My_Connector.3dshapes/JST_PA_B13B-PA-SK_1x13_P2.00mm_Vertical.step`

set-soft, 2022-06-09:
Hi @mmutlu-any !

> Maybe I am doing something wrong to test. I installed kibot from debian package. I edited /usr/lib/python3/dist-packages/kibot/kicad/config.py file. First according to #d93d705, then changed to #24d81f4.
Should I do anything else?

Please try using the whole config.py file, in fact I'll suggest copying all the files from the last commit that passed all regression tests. The line numbers of the above log messages aren't usable.

> This line does not make sense to me because .step file is actually there:
> ```File not found: ${KICAD_USER_LIBRARY_DIR}/My_Connector.3dshapes/JST_PA_B13B-PA-SK_1x13_P2.00mm_Vertical.step```

The problem here is that `kicad2step` isn't expanding `KICAD_USER_LIBRARY_DIR`, this is because the variable doesn't look to be defined in the environment. With the current code:

```python
        # Export the rest. KiCad2Step needs them
        for k, v in KiConf.kicad_env.items():
            if k not in os.environ:
                os.environ[k] = v
                logger.debug('Exporting {}="{}"'.format(k, v))
```

And the log you show:

> DEBUG:- KiCad var: KICAD_USER_LIBRARY_DIR="/home/<user>/<path_to_lib>/library" (kibot - config.py:251)

This sounds impossible, but as the line numbers are different I think you are missing some patches.

commit d93d7054 (2022-05-30): Better KiCad v6 environment variables handling

- more variables detected (KICAD6_3RDPARTY, KICAD_USER_TEMPLATE_DIR)
- native KiCad 6 names (KICAD6_* instead of KICAD_*)
- all exported to the environment, not just the core

Fixes #205