INTI-CMNB/KiBot #207: pdf_sch_print and pdf_pcb_print text_variables replaced by its name instead its value

erichstuder, 2022-06-11:
Hello

I try to create pdf outputs of the schematic and the pcb.
But it looks as if the text_variables are replaced by its name instead of its value.
For example for the pcb [mainboard-F_Silkscreen.pdf](https://github.com/INTI-CMNB/KiBot/files/8883510/mainboard-F_Silkscreen.pdf) the default value for the commit is "noHash". On the Silkscreen the replacement works fine. But in the bottom right corner it becomes the value "GIT-COMMIT-HASH" which is the name of the text_variable itself.
Locally on my PC the replacement with the default value works fine as far as I can see.
I don't understand why it works on the Silkscreen but not on the Sheet itself.
Could it be a problem because the commit has no tag yet?
What else could be the problem?

On the schematic pdf the problem is the same:
[mainboard-schematic.pdf](https://github.com/INTI-CMNB/KiBot/files/8883511/mainboard-schematic.pdf)

Repository and branch:
https://github.com/erichstuder/RemoteControl/tree/pcb_fabrication_outputs

yml file:
https://github.com/erichstuder/RemoteControl/blob/pcb_fabrication_outputs/.github/workflows/pcb_fabrication_outputs.yml

yaml file:
https://github.com/erichstuder/RemoteControl/blob/pcb_fabrication_outputs/electronics/mainboard/mainboard.kibot.yaml

Thank you in advance
erichstuder

set-soft, 2022-06-11:
> Locally on my PC the replacement with the default value works fine as far as I can see.

Really?
I changed the directory to where the files are and executed `eeschema mainboard.kicad_sch`, and variables aren't replaced:

[image]

I think you must use the variables that KiCad uses, i.e. the ones that are expanded from the `Page settings` dialog. Note you have 9 comments fields that you can use. And note that those variables are suggested while editing the WKS file.

Then you can assign things like `${GIT-COMMIT-HASH}` to the comment X in the `Page settings` dialog.

This patch fixes your problem: https://github.com/set-soft/RemoteControl/commit/ade483f44a7e1ba37470f8683ed474f58a357e90

Take a look at the ouputs:

[mainboard-schematic.pdf](https://github.com/INTI-CMNB/KiBot/files/8883997/mainboard-schematic.pdf)
[mainboard-F_Silkscreen.pdf](https://github.com/INTI-CMNB/KiBot/files/8883998/mainboard-F_Silkscreen.pdf)

I'm  not sure why the `git tag` output is empty when using the workflows.

A couple of comments:
1. Try using the `pcb_print` output, is faster and more flexible
2. Avoid committing files like `mainboard.step`, it can be generated by KiBot and is changes all the time, artificially enlarging the repo.

closed 2022-06-11 by set-soft

erichstuder, 2022-06-11:
I did as you adviced and it works now!
Thanks for your hints and advice.

> I'm not sure why the git tag output is empty when using the workflows.
This is because the commit has no tag. I tried it out with a tagged commit and in worked. I'm fine with this behaviour.