INTI-CMNB/KiBot #214: [HOW] Ignoring ignored warnings

flaviut, 2022-06-29:
**What do you want to achieve?**

I see a bunch of DRC warnings like:

```
WARNING:(W058) Ignoring (items_not_allowed) Items not allowed (keepout area 'Antenna')
    Rule: keepout area 'Antenna'; Severity: error
    @(131.0532 mm, 23.9083 mm): Pad 33 [unconnected-(U5-Pad33)] of U5 on F.Cu (pcbnew_do.kiauto.file_util - file_util.py:112) (kibot - kiplot.py:164)
```

And also ERC warnings like:

```
WARNING:(W058) (multiple_net_names) Both +12V and CS- are attached to the same items; +12V will be used in the netlist
     (excluded)
    @(21.59 mm, 175.26 mm): Symbol #PWR0126 [+12V] Pin 1 [+12V, Power input, Line]
    @(120.65 mm, 36.83 mm): Label 'CS-' (eeschema_do.kiauto.file_util - file_util.py:138) (kibot - kiplot.py:164)
```

I don't want these warnings to show up, since I've filtered them in preflight, as well as the kicad GUI:

```
  filters:
    - filter: only one of the RF modules is populated so the overlap is fine
      error: items_not_allowed
      regex: "keepout area 'Antenna'"
```

Filtering by `58` doesn't seem to accomplish what I want:

```
  filters:
    - error: '58'
```

Any suggestion on how to exclude these warnings for both ERC and DRC?

**Environment (please complete the following information):**
Where are you running KiBot:

Arch Linux, pip, `KiBot 1.2.0 - Copyright 2018-2022, Salvador E. Tropea/INTI/John Beard - License: GPL v3+`

set-soft, 2022-06-29:
Hi @flaviut !

I see the docs weren't clear enough, now they explain it better.

When you ignore an **error** it is reported as a **warning**. Completely ignoring DRC and ERC errors isn't a good idea. 

The `W058` messages you see are warnings collected from KiAuto tool. They can be ignored using the mechanism to filter KiBot warnings:

#### Filtering KiBot warnings

KiBot warnings are marked with `(Wn)` where *n* is the warning id.

Some warnings are just recommendations and you could want to avoid them to focus on details that are more relevant to your project.
In this case you can define filters in a similar way used to [filter DRC/ERC errors](#filtering-drc-and-erc-errors).

As an example, if you have the following warning:

```
WARNING:(W43) Missing component `l1:FooBar`
```

You can create the following filter to remove it:

```yaml
global:
  filters:
    - number: 43
      regex:  'FooBar'
```

closed 2022-06-29 by set-soft

flaviut, 2022-06-30:
I see, that makes a lot of sense!

I guess a couple things I'm struggling with are:

- there's no super obvious indication of where each warning is coming from
- it took me a while to figure out that filtering kibot warnings is done by turning `W010` -> `10` for the yaml.

I understand it all now, but if you don't mind, I have a couple suggestions that might help others out too:

Log warnings/errors as:

```
KIBOT WARNING:(W058) ...
PREFLIGHT ERROR:(unconnected_items) ...
```

Be more accepting of kicad filter values, i.e. accept `number: W058` and process it internally to whatever.

Anyway, thank you for all your hard work on this project!

set-soft, 2022-10-12:
Hi @flaviut !
The 49b21c8e93d45fa26485f0b67d2270b832896bd4 commit fixes some confusing details.
I won't add prefixes like you suggest.

flaviut, 2022-10-12:
That's very helpful—definitely goes a long way towards avoiding confusion here.

Thank you!

commit cdc9d573 (2022-06-29): [DOCs] Added a note about what "filtering" DRC errors means

Closes #214