INTI-CMNB/KiBot #215: [HOW] to make render_3d output deterministic for use in git

dariuskt, 2022-07-02:
Hey!

I want that render_3d output image would be the same every time i render without changing anything in the project. Is there a way to achieve that?

I have created pre-commit hook in git to run kibot every time so that pdf, bom, images and gerbers would be kept up to date with a project.

Problem is that every time render_3d runs it will produce slightly different file and git will treat that as a change and in git history i have no way to know if there was a real change.
I had similar problem with pcbdraw, but solved that by using `format: jpg`
As render_3d produces png I suspect that png has some kind of timestamp or date inside that creates this problem.

Is there a way to convince render_3d produce jpg image instead of png? or maybe there is some option to exclude timestamps from png?

```yaml
  - name: pcb_image_3d_front
    type: render_3d
    options:
      output: 'docs/PCB_front.jpg'
      orthographic: true
      download: true
      solder_mask: '#104020'
      ray_tracing: true
      width: 1334
      height: 958
      view: top
      zoom: 5
```

- Using a docker image `setsoft/kicad_auto:ki6.0.5_Debian`

set-soft, 2022-07-04:
Hi @dariuskt !
I don't think this is possible. The 3D render isn't 100% deterministic. 
Why do you need a 100% deterministic result? 
In my regression tests I tolerate some "fuzziness". I compare the images using ImagMagic `compare` tool.
In general: images generated by KiCad can't be trusted, they slightly change from version to version, and some times from run to run.

dariuskt, 2022-07-05:
As I attempted to explain above in my version control situation, non-deterministic output hides changes from version control and user who wants to now is something changed or not. Additionally, every commit with no real change in the pcb layout just bloats repository size rapidly.

Well.. I did some testing and you are totally right, KiCad itself fails to reproduce same rendering between runs with no changes. I guess I would have to figure out some custom way how to automatically recognize real changes in constantly changing files.

Thanks for your time. I should've done more testing before opening an issue.

closed 2022-07-05 by dariuskt

set-soft, 2022-07-06:
Hi @dariuskt !
My recommendation: keep generated files out of the git repo.
You can have them as "artifacts" when using GitHub or GitLab CI/CD. You can use GitHub/GitLab sites or a local site (GitLab is fully free software).
I suggest including generated files only for important tags of your project, i.e. the files you sent to the manufaturer, the renders of the version you manufactured, etc.
But keeping the intermediate stuff as volatile data, not in the repo. The artifacts are usually removed after some time, lets say a month.