INTI-CMNB/KiBot #216: [BUG] Time-out waiting for 3d viewer to be responsive

Regressor, 2022-07-10:
KiBot can't render 3d image of board complaining about non-responsive 3d viewer

How to reproduce:
Create a board with a component having link to 3d model of rpi cm4 module (https://datasheets.raspberrypi.com/cm4/CM4-step.zip) and try to get 3d image of it.

Expected behavior: KiBot renders 3d image of board. 
Real behavior: KiBot quickly (about a minute) throws time-out error

I'm running KiBot in gitlab ci environment using setsoft/kicad_auto:10.4-5.1.6 image
I tried to change resolution, view and other parameters but result is same.

I started render_3d action with -vv key and then examined .ogv video file. The pcbnew 3d viewer screen shows clock cursor and a status text in bottom left angle "Loading cm4.step"

set-soft, 2022-07-11:
Hi @Regressor !
This is a very particular 3D model, I tried it using an old i5 (i5-2310) and it took 47 seconds to be rendered (the loading time was the problem because the OpenGL render took 65 ms).

Did you try using a global `kiauto_time_out_scale`?

Even in this slow machine I didn't get a time-out for a naive board using the module:

[image]
[image]

Using:

```yaml
# Example KiBot config file
kibot:
  version: 1

outputs:
  - name: "base"
    type: render_3d
    run_by_default: false
    options:
      width: 2000
      height: 1000

  - name: "base2"
    type: step
    run_by_default: false
    options:
      metric_units: false

  - name: "top"
    type: render_3d
    extends: "base"
    options:
      rotate_x: 3
      rotate_z: -2

  - name: "bottom"
    type: render_3d
    extends: "top"
    options:
      width: 1920
      view: "bottom"
```

But if the motherboard is even more complex you could easily get a time-out.

Regressor, 2022-07-12:
I tried kiauto_time_out_scale: 2.0 and got same result. Then I tried kiauto_time_out_scale: 4.0 and this helped, but bring another problem: now all images are "top" view :(

My config:

```
kiplot:
  version: 1

global:
  kiauto_time_out_scale: 4.0
  kiauto_wait_start: 120

preflight:
  run_erc: true
  update_xml: true
  run_drc: true
  check_zone_fills: false
  ignore_unconnected: false

  - name: 'render_3d_top'
    comment: "Render 3D image of board's top"
    type: render_3d
    dir: ../_fabrication/pcb_mainboard/_assemble/picture
    options:
      output: '%f_top.%x'
      view: 'top'
      background1: '#FFFFFF'
      background2: '#C2C2C2'
      solder_mask: '#11184c'
      zoom: 5
      width: 3840
      height: 2160

  - name: 'render_3d_bottom'
    comment: "Render 3D image of board's bottom"
    type: render_3d
    dir: ../_fabrication/pcb_mainboard/_assemble/picture
    options:
      output: '%f_bottom.%x'
      view: 'bottom'
      background1: '#FFFFFF'
      background2: '#C2C2C2'
      solder_mask: '#11184c'
      zoom: 5
      width: 3840
      height: 2160

```

Also it looks like zoom is ignored too:

[image]
[image]

I removed cm4.step, removed global section from yaml and pcb top / bottom was rendered as expected.
Then I returned cm4.step and got error:

ERROR:Time-out waiting for 3D viewer to be responsive (pcbnew_do - pcbnew_do:508) (kibot - kiplot.py:159)

Commands to render (from .gitlab-ci.yml):

```
    - cd pcb_mainboard
    - kibot -s all render_3d_top
    - cd ..
    - mv _library/buk_library.3dshapes/case _library/buk_library.3dshapes/case2 || true
    - cd pcb_mainboard
    - kibot -s all render_3d_bottom
    - cd ..
```

set-soft, 2022-07-14:
Hi @Regressor !

I have an idea to solve the 3D render issues, but it will be implemented for KiCad 6 first (not sure if I'll implement it for KiCad 5). Are you interested on testing it?

Note that the tests I mentioned all were produced without acceleration, KiBot can't use acceleration because KiCad runs in a virtual server.

BTW: Do you have any of these models publicly available?

The main problem is that KiCad losses events when busy, and you don't get information about what KiCad is doing.

Regressor, 2022-07-14:
Sorry. This is my own project, but part of it was taken from private device from my work so I can't post schematic and pcb files public :(
Currently I setup it to work without cm4.step file and all is ok. Thanks for such great software!

Regressor, 2022-07-15:
P.S. I had to change docker image a little because there were problems with letsencrypt certificate (https://stackoverflow.com/questions/69408776/how-to-force-older-debian-to-forget-about-dst-root-ca-x3-expiration-and-use-isrg) and I have own bom generator:

```
from setsoft/kicad_auto:10.4-5.1.9

RUN apt -y update && \
 apt -y install python3-pip && pip3 install kicad_netlist_reader && apt remove -y python3-pip && apt autoremove -y && apt clean -y && \
 apt install apt-transport-https ca-certificates -y && update-ca-certificates && rm /usr/share/ca-certificates/mozilla/DST_Root_CA_X3.crt && \
 update-ca-certificates

```

set-soft, 2022-07-15:
> Sorry. This is my own project, but part of it was taken from private device from my work so I can't post schematic and pcb files public :(

Is ok, I'll try to create some artificially complex stuff using various cm4.step instances.

set-soft, 2022-07-15:
For the records: The KiAuto branch with experimental KiCad 6 support is [kicad_interposer](https://github.com/INTI-CMNB/KiAuto/tree/kicad_interposer).

It seems to work for KiCad 6.0.5 and 6.0.6

The code detects the OpenGL buffer swaps to know when KiCad finished a render, it currently has some glitches (for some reason KiCad seems to skip 1 render from time to time), but is really fast and we know exactly when the render is finished, no speculative delays.

Regressor, 2022-07-16:
My main workstation has outdated win7 so I can't use KiCad6 but I'll try it on another one next week.

set-soft, 2022-07-26:
> My main workstation has outdated win7 so I can't use KiCad6 but I'll try it on another one next week.

I added more code and now the interposer trick seems to be working for KiCad 5. The new code is also trying to detect when KiCad is sleeping waiting for new events. I tried it with big files (24 seconds PCB load and components with 47 seconds 3D model load), but the rendering time without raytracing never went over 5 seconds. 

One important thing that I noticed is that KiCad has some animations when you move the camera (not for zoom). I found how to disable it for KiCad 6, but not for KiCad 5. This can generate huge delays because what KiCad 6 draws twice (don't ask me why twice) KiCad 5 uses 12 drawnings!  But the interposer code should be able to wait for the 12 drawings.

I also noticed that the memory needed for 4k renders is huge, it made the computer I'm using right now to become so busy that I simply rebooted it. In another try I waited for 10 minutes and after it I went to launch ;-) The computer was recovered, but I don't understand why it became that loaded, I have the swap disabled, older Linux boxes just killed the hungry process, don't know what is doing the current setup.

The mechanism needs more adjusts for sure, but looks like it will solve the time-out issues. 

Also: I was afraid that we could need tons of interposer binaries (one for each Linux system), but I tested that the one I put on git works for Debian 10 and Debian 11, using KiCad 5 and KiCad 6.

Regressor, 2022-07-27:
Can I test kicad5 version using docker image ? I tried to migrate to kicad6 and it seems it needs too much time (this project has ~20 boards and ~2 pcb panels) so I decide to do this later after beta-release :)

set-soft, 2022-07-27:
Hi @Regressor !

> Can I test kicad5 version using docker image ? 

Not with a pre-built image, I'll see if I can include it in the development images.

set-soft, 2022-07-30:
Hi @Regressor !

> Can I test kicad5 version using docker image ? I tried to migrate to kicad6 and it seems it needs too much time (this project has ~20 boards and ~2 pcb panels) so I decide to do this later after beta-release :)

Now we have a pre-built image: `setsoft/kicad_auto:dev` and `setsoft/kicad_auto:dev_k6` are using the experimental KiAuto.

Regressor, 2022-08-01:
Thanks! I'll try to run my CI jobs on this image this week

Regressor, 2022-08-03:
Something wrong with image setsoft/kicad_auto:dev   kibot-check complains about colorama and kicad auto

set-soft, 2022-08-03:
Hi @Regressor !

Are you sure you aren't doing some change on it (i.e. doing some auto remove operation on the packages).
I tried with a fresh image:

```
$ docker pull setsoft/kicad_auto:dev
dev: Pulling from setsoft/kicad_auto
6ec8c9369e08: Already exists 
62793550d258: Already exists 
9e950e636578: Already exists 
13d8992d4738: Pull complete 
Digest: sha256:760100838e384d63453f0be7633dd8a47d22cfe3e14444a96b93027d757afea0
Status: Downloaded newer image for setsoft/kicad_auto:dev
docker.io/setsoft/kicad_auto:dev
$ ./run_shell.sh
root@973eed0120f9:/# kibot-check 
KiBot installation checker

Core:
Linux: 5.10.0.16 (Linux 973eed0120f9 5.10.0-16-amd64 #1 SMP Debian 5.10.127-2 (2022-07-23) x86_64 GNU/Linux)
Python: 3.7.3 (default, Jan 22 2021, 20:04:44)  [GCC 8.3.0]
KiCad: 5.1.9+dfsg1-1~bpo10+1
Kibot: 1.2.0

Modules:
Colorama: 0.3.7
Distutils: 3.7.3
LXML: Ok
PyYAML: 3.13.0 (3.13)
QRCodeGen: Ok
Requests: 2.21.0
XLSXWriter: 1.1.2

Tools:
Ghostscript: 9.27.0 (9.27)
Git: 2.20.1
ImageMagick: 6.9.10.23 (Version: ImageMagick 6.9.10-23 Q16 x86_64 20190101 https://imagemagick.org)
Failed to run python3, error 1
Interactive HTML BoM: *UNKNOWN*
KiBoM: 1.8.0 (KiBOM Version: 1.8.0)
KiCad Automation tools: 1.6.15 (1.6.15 - Copyright 2018-2022, INTI/Productize SPRL - License: Apache)
KiCost: 1.1.10 (KiCost v1.1.10)
Pandoc: Not available
PcbDraw: 0.9.0.3 (PcbDraw 0.9.0-3)
RAR: 5.50.0 (RAR 5.50   Copyright (c) 1993-2017 Alexander Roshal   11 Aug 2017)
RSVG tools: 2.44.10

* Interactive HTML BoM not installed or too old
  Visit: https://github.com/INTI-CMNB/InteractiveHtmlBom
  Download it from: https://github.com/INTI-CMNB/InteractiveHtmlBom/releases
  - Mandatory for `ibom`

* Pandoc not installed or too old
  Install the `pandoc` package, i.e.: `sudo apt-get install pandoc`
  You should also install the following packages: texlive-latex-base, texlive-latex-recommended
  - Optional to create PDF/ODF/DOCX files for `report`

Color reference: ok, optional for an output, optional for general use, mandatory for an output, mandatory for general use

Did this help? Please consider commenting it on https://github.com/INTI-CMNB/KiBot/discussions/categories/kibot-check
root@973eed0120f9:/# dpkg -l *colorama
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name             Version      Architecture Description
+++-================-============-============-===========================================================
ii  python3-colorama 0.3.7-1      all          Cross-platform colored terminal text in Python - Python 3.x
```

If you run `apt-get autoremove` it will remove the packages you mention because they are installed but kiauto isn't installed anymore.

Regressor, 2022-08-03:
Yes. I need kicad_netlist_reader so I install pip3 then kicad_netlist_reader and after that remove pip3 & autoremove.
I got it working with this Dockerfile:

from setsoft/kicad_auto:dev
#from setsoft/kicad_auto:10.4-5.1.9

RUN apt -y update && \
 apt install -y python3-pip && pip3 install kicad_netlist_reader && \
 apt install python3-xvfbwrapper && apt remove python3-pip -y  && \
 apt install python3-colorama -y && apt install python3-psutil -y && apt install pandoc -y && \
 apt install recordmydesktop xdotool -y && \
 apt install apt-transport-https ca-certificates -y && update-ca-certificates && rm /usr/share/ca-certificates/mozilla/DST_Root_CA_X3.crt && \
 update-ca-certificates && apt autoremove -y && apt clean -y

So now it works and zoom is fine! I'll try to start it with CM4 step enabled

Regressor, 2022-08-03:
Yes! Its works! Zoom as expected, timeouts - default

[image]

set-soft, 2022-08-03:
Ok, I'll close this issue, even when the KiAuto needed still in beta, but now is documented how to try using it.

The new mechanism is faster and knows much better what KiCad is doing. The cost: is platform dependent, needs Linux and 64 bits x86, but this is the case for people using the docker images. The old mechanism is available as a fallback.

closed 2022-08-03 by set-soft

set-soft, 2022-08-12:
Hi @Regressor !

> Yes. I need kicad_netlist_reader so I install pip3 

An important note: the `setsoft/kicad_auto_test` images contains pip. They don't have a `dev` version, but when you switch to the stable images you'll be able to use them.
KiAuto 2.0.0 is now available, it has the interposer stuff and is included in the new docker images.
Important: installing pip recommended package isn't a good idea, they are a lot of heavy dependencies. You can use:

```
	apt-get -y install --no-install-recommends python3-pip unzip curl python3-setuptools python3-wheel 
```

> then kicad_netlist_reader and after that remove pip3 & autoremove.

Just avoid running the autoremove part when using the dev images, they'll be a little bit bigger, that's all.

> I got it working with this Dockerfile:
> 
> from setsoft/kicad_auto:dev 
> #from setsoft/kicad_auto:10.4-5.1.9

Now you can try using `11.4-5.1.9` or `latest`

> RUN apt -y update && 
> apt install -y python3-pip && 

Here you should avoid installing all recommended packages.

> apt install python3-xvfbwrapper && 
> apt install python3-colorama -y && 
> apt install python3-psutil -y && 
> apt install recordmydesktop xdotool -y &&

I think all of them were needed just because you used autoremove

> apt install pandoc -y && 

This is also included in the `kicad_auto_test` images.

> apt install apt-transport-https ca-certificates -y && update-ca-certificates && rm /usr/share/ca-certificates/mozilla/DST_Root_CA_X3.crt && update-ca-certificates && 

I think this isn't needed for the new `11.4-5.1.9`/`latest` images, they use Debian 11, instead of Debian 10.

You should try using `setsoft/kicad_auto_test:latest`, unless you need some fresh patch on KiBot. This image includes KiAuto 2.0.0, pip3, Pandoc and Debian 11

Regressor, 2022-08-13:
Thank you! I'll try this