INTI-CMNB/KiBot #219: [BUG] Variant: red cross on multi unit symbol only on unit A

myorbit, 2022-07-12:
**Describe the bug**

In a multi unit symbol, a red cross for a component which should not be included in the variant is only applied to unit 'A'. Unit 'B' has no red cross on it. For example this happens to an OpAmp which separates the power supply pins to its own unit.

**To Reproduce**
Schematic with a multi unit symbol in KiCad 6.

**Expected behavior**
A red cross on other units that are part of the same component for clear visibility on the schematic.

**Screenshots**
[image]
[image]

**Environment (please complete the following information):**
- In a CI/CD workflow with GitLab
- docker image:  kicad_auto:ki6

set-soft, 2022-07-12:
Hi @myorbit !

I'm not sure about it. Your component's unit B doesn't have any body to cross. Should we create a transparent body? Which size? 

Here is a more representative example: 
[test_cross_units.zip](https://github.com/INTI-CMNB/KiBot/files/9091743/test_cross_units.zip)

Using the above schematic and config I get:

[image]

So all units are indeed crossed. In fact: what you call UNIT B is just some pins moved away to make the drawing cleaner, but not a real UNIT

set-soft, 2022-07-12:
Going deeper I found a real bug:

[image]

The problem here was because 74LS00 has 2 different drawings for A-D units, and only one was crossed. The crossed one wasn't the visible one.

[image]

set-soft, 2022-07-12:
Ok, I added a global option to cross them using the box containing the pins. Can be enabled using:

```yaml
globals:
  cross_no_body: true
```

And you get:

[image]

commit 42503a90 (2022-07-12): [SCH][Print] A global option to cross components without a body.

- Only for KiCad 6

Closes #219