INTI-CMNB/KiBot #224: [BUG] V6 manually panelized board component positions are not unique

SukuWc, 2022-07-14:
Fill as much as you can, remove the rest

**Describe the bug**
I created schematic for the designed than created PCB for it. At this point kibot was working properly both locally and on the github action to produce manufacturing files. Then I selected the whole layout and used the Special Tools/Create Array command to create a panel layout, making sure to select _Keep existing reference designators_ to make sure that the references are the same, so they can be found in the BOM file by the assembly house.

Now the generated .pos position file now contains all of the components of every board instance on the panel, however all of the same components are exported with the same x y coordinates. For example C1 capacitor is now present in .pos file 16 times (since the panel is 16x1) however all of the instances have the same x y coordinates.

Not that if I use the build in position file exporter _File/Fabrication Outputs/Component Placement_  then I get a good result, with all of the position coordinates unique!

**To Reproduce**
I created a public repository only  contains files that are necessary to reproduce the issue. Clone the repository to run Kibot locally in the kicad project folder or check the github actions output on github.

https://github.com/SukuWc/interface-CICD-test

**Expected behavior**
All component instances with the same designator should be exported with their unique positions

**Screenshots**
[image]

**Environment (please complete the following information):**
Where are you running KiBot:
- Locally using latest release kibot (Xubuntu 22.04)
- Also on GitHub Actions using kicad_auto:dev_k6

set-soft, 2022-07-14:
Hi @SukuWc !
Ok, I'll take a look at this, the code doesn't expect to find repeated references.

set-soft, 2022-07-14:
Hi @SukuWc !
Ok, now should work.

commit c4b1d6b8 (2022-07-14): [Position] Fixed problems with repeated references

- Happens when doing manual panelization

Closes #224
closed 2022-07-14 by set-soft