INTI-CMNB/KiBot #226: [BUG] Gerber output only shows F.Cu filled

myorbit, 2022-07-14:
**Describe the bug**
I have a 6-layer board and all zones are filled and re-filled ('B') in the KiCad project.
1) I gernerate the Gerber files directly from the KiCad project. All copper layers are generated as intended.
2) After that I push changes to GitLab and there only the top layer (F.Cu ) is filled. All other layers just show the tracks.
(I also checked with other Gerber viewers.)

I made a minimal example where I noticed that it has to do something with my 'print_asm_notes' in the yaml config which generates a PDF with top and bottom assembly plots.

It would be great if someone could tell me if I am doing something wrong with the assembly prints? I am a bit lost right now.

**To Reproduce**
Here are my yaml files:

`.gitlab-ci.yml`
```
### Gitlab CI/CD for KiCad

stages:
  - run_erc
  - run_drc
  - gen_fab

image:
  name: setsoft/kicad_auto:dev_k6

erc:
  stage: run_erc
  script:
    - "kibot -d fabrication -s update_xml,run_drc -i"
  rules:
    - changes:
      - "**.kicad_sch"
      - ".gitlab-ci.yml"
      - "**.kibot.yaml"
      when: always
  artifacts:
    when: always
    paths:
      - fabrication/

drc:
  stage: run_drc
  script:
    - "kibot -d fabrication -s update_xml,run_erc -i"
  rules:
    - changes:
      - "**.kicad_pcb"
      - ".gitlab-ci.yml"
      - "**.kibot.yaml"
      when: always
  artifacts:
    when: always
    paths:
      - fabrication/

sch_outputs:
  stage: gen_fab
  script:
    - "kibot -d fabrication -s run_drc,run_erc"
  rules:
    - changes:
      - "**.kicad_sch"
      - ".gitlab-ci.yml"
      - "**.kibot.yaml"
      when: on_success
  artifacts:
    when: always
    paths:
      - fabrication/

pcb_outputs:
  stage: gen_fab
  script:
    - "kibot -d fabrication -s all gerbers excellon_drill zip_gerber_drill"
  rules:
    - changes:
      - "**.kicad_pcb"
      - ".gitlab-ci.yml"
      - "**.kibot.yaml"
      when: on_success
  artifacts:
    when: always
    paths:
      - fabrication/
```

`kicad_ci.kibot.yaml`
```
# KiBot config file
kibot:
  version: 1

preflight:
  run_erc: true
  update_xml: true
  run_drc: true
  filters:
    - filter: 'Filter to ignore courtyards_overlap'
      error: 'courtyards_overlap'
      regex: 'Footprint'
  check_zone_fills: true
  ignore_unconnected: false

global:
#   kiauto_wait_start: 60
#   kiauto_time_out_scale: 2
  # We want the revision added to the names for this project
  output: '%f_v%r-%i.%x'

outputs:
 
  - name: 'print_asm_notes'
    comment: "Assembly print as PDF"
    type: pcb_print
    dir: .
    options:
      format: 'PDF'
      output: '%f_v%r-asm_notes.%x'
      force_edge_cuts: true
      plot_sheet_reference: true
      drill_marks: none
      pages:
        - sheet: 'Assembly Top Layer'
          colored_holes: true
          holes_color: '#f5e7e3'
          layers:
          - layer: F.Cu
            color: '#f5e7e3'
          - layer: F.Fab
            color: '#840000'
            plot_footprint_refs: true
            plot_footprint_values: false
          - layer: Edge.Cuts
            color: '#000000'
        - sheet: 'Assembly Bottom Layer'
          colored_holes: true
          holes_color: '#def5f1'
          mirror: true
          layers:
          - layer: B.Cu
            color: '#def5f1'
          - layer: B.Fab
            plot_footprint_refs: true
            plot_footprint_values: false
          - layer: Edge.Cuts
            color: '#000000'

  
  - name: 'gerbers'
    comment: "Gerbers for the board house"
    type: gerber
    dir: gerbers
    options:
      # generic layer options
      exclude_edge_layer: true
      exclude_pads_from_silkscreen: false
      use_aux_axis_as_origin: false
      plot_sheet_reference: false
      plot_footprint_refs: true
      plot_footprint_values: false
      force_plot_invisible_refs_vals: false
      tent_vias: true

      # gerber options
      line_width: 0.1
      subtract_mask_from_silk: false
      use_protel_extensions: false
      gerber_precision: 4.6
      create_gerber_job_file: true
      use_gerber_x2_attributes: true
      use_gerber_net_attributes: true

    # 6-layer
    layers:
      - layer: F.Cu
      - layer: In1.Cu
      - layer: In2.Cu
      - layer: In3.Cu
      - layer: In4.Cu
      - layer: B.Cu
      - layer: F.Paste
      - layer: B.Paste
      - layer: F.Mask
      - layer: B.Mask
      - layer: F.Fab
      - layer: B.Fab
      - layer: User.Drawings
      - layer: User.3
      - layer: Edge.Cuts
        suffix: Edge_Cuts

  - name: 'excellon_drill'
    comment: "Excellon drill files (no map)"
    type: excellon
    dir: drill
    options:
      metric_units: false
      pth_and_npth_single_file: false
      use_aux_axis_as_origin: false
      minimal_header: false
      mirror_y_axis: false
      report:
       filename: '%f_v%r-%i.%x'

  - name: 'zip_gerber_drill'
    comment: ZIP file of Gerber with NC Drills
    type: compress
    dir: export
    options:
      files:
        - from_output: gerbers
          dest: /
        - from_output: excellon_drill
          dest: /
```

Here is what happens when I add 'print_asm_notes' before or after 'gerbers' in the following section:

```
pcb_outputs:
  stage: gen_fab
  script:
    - "kibot -d fabrication -s all gerbers excellon_drill zip_gerber_drill"
```

1) Do not use 'print_asm_notes': Generates my gerber files just fine with all zones filled in all layers. But apparently no assembly print is generated.
2) 'print_asm_notes' before 'gerbers': Does generates gerber files with zone filled only on top layer. Zones with higher priorities on other layers are filled but not the GND zone (zone priority 0) on other layers. Assembly print on top is filled. Assembly print on bottom layer shows all zones filled with higher priorities but not the GND zone with priority 0.
3) 'print_asm_notes' after 'gerbers': Generates my gerber files just fine with all zones filled in all layers. Assembly print on top is filled. Assembly print on bottom layer shows all zones filled with higher priorities but not the GND zone with priority 0.

**Expected behavior**
Gerbers with KiBot should generate same output as with direct Fabrication Outputs -> Gerbers in KiCad.
Assembly prints should print correct with all zones filled.

**Screenshots**
Gerbers with all zones filled correctly:
[image]

Gerbers with only zones of higher priority filled:
[image]

Assembly print top layer with all zones filled:
[image]

Assembly print bottom with only higher priority zones filled:
[image]

**Environment (please complete the following information):**
* In a CI/CD workflow with GitLab
* docker image: kicad_auto:dev_k6

set-soft, 2022-07-14:
Hi @myorbit !
Do you have a test PCB, I thought this problem was fixed.
In order to print the through hole pads KiBot remove all other stuff and then restores the removed stuff, including refill. Looks like in your case not all is restored. But I remmber it worked for the light_control.kicad_pcb test file.

set-soft, 2022-07-15:
Hi @myorbit !
I need a test case with these priorities in order to reproduce the problem, I checked the zone filler works for `light_control.kicad_pcb` PCB test case

myorbit, 2022-07-15:
I understand. Trying to prepare a minimal example for this case.
Please be patient as I have to release a PCB for fabrication just right now. I will try as fast as possible!
Thank you very much for being so dedicated! This is really awesome.

set-soft, 2022-07-15:
Hi @myorbit !
Don't worry, do it when you have the time.

myorbit, 2022-07-17:
Hi @set-soft!
I have prepared a minimal example including a KiCad project and the yaml configs. You can find it here: [https://github.com/myorbit/test-zone-fills](https://github.com/myorbit/test-zone-fills)
Hope you can reproduce the bug.

Here are some screenshots from KiCad and my KiBot outputs. KiCad top/bottom layer zones are all clearly shown. KiBot outputs on bottom layer only shows the 'shield' zone filled.

KiCad View (showing top/bottom):
[image]

KiBot generated Gerber (showing top/bottom):
[image]

KiBot generated Assembly (showing top/bottom):
[image]
[image]

set-soft, 2022-07-18:
Hi @myorbit !
I found the source of the problem: The API has a GetLayer() member, it returns **1** layer, but KiCad 6 supports zones in multiple layers. Now I must see how to handle it.

closed 2022-07-18 by set-soft

myorbit, 2022-07-20:
Thank you very much for your immediate fix!

I can confirm that zone filling now works but I still got an issue with the location of the 'print'asm_notes': I noticed that I get a segmentation fault on:
```
pcb_outputs:
  stage: gen_fab
```

For a workaround and my summary please browse to the end of this message! (I hope you can follow my elaborations.)

I don't know if this is related to the recent fix. I have never experienced it before this fix. Sometimes, if I retry the failed job (in GitLab) it completes without a segmentation fault in rare cases depending on the location of the 'print_asm_notes_variant'.

```
- 'Assembly print as PDF' (print_asm_notes) [pcb_print]
- 'Assembly print as PDF' (print_asm_notes_variant) [pcb_print]
WARNING:(W045) Pad with solder paste, but no copper or solder mask aperture in U2 (kibot - out_base.py:337)
/bin/bash: line 117:    12 Segmentation fault      (core dumped) kibot -d fabrication -s all print_asm_notes print_asm_notes_variant gerbers excellon_drill zip_gerber_drill
Uploading artifacts for failed job
```

I can imagine two possible issues:

1) This could happen because I do run two 'pcb_print' in a row? The second 'print_asm_notes_variant' is for a variant which uses a simple variant filter.
Maybe there is still an issue because I get a warning (W045) (see above code) which do not appear on the first 'print_asm_notes' but on the the second 'print_asm_notes_variant'.
The only difference between 'print_asm_notes' and 'print_asm_notes_variant' is that on the variant I apply the simple 'dnf_filter' option. The part mentioned (U2) is marked in the DNF filter, amongst other parts. And U2 definitely got copper and a solder mask aperture. I checked again in footprint and PCB.
2) It could happen because I concatenate the title in the variant assembly print with a colon. In a previous commit with this option I could retry the failed job and got not segmentation fault. But the ouptut in the title was different. Instead of the colon ':' it was replaced by '_'.

      pages:
        - sheet: 'Assembly Top Layer'
          title: '+ (Variant: Calif)'

To test these two possbile issues...
* I first removed the colon: Segmentation faults with the above W045 warning.
* And then removed the second 'print_asm_notes_variant': No segmentation faults but still got the warning W045. (Because I have a gerber output with a 'dnf_filter'.)

**Workaround:**
What does fix this issue is again re-order both 'print_asm_notes' and 'print_asm_notes_variant' to the end of the file `kicad_ci.kibot.yaml`. But only if I introduce a `type: gerber` with the dnf_filter option after the gerbers without a variant. Then the 'print_asm_notes_variant' comes much later in the file.
I saw that the warning W045 pops up where the `dnf_filter` ist first applied, either at the gerber_variant 'print_asm_notes_variant'.

**Summary:**
* The warning pops up only with the 'dnf_filter' options applied to 'gerber' and/or 'pcb_print'.
* So maybe the 'pcb_print' type handle this in a different way than 'gerber' type which leads to a segmentation fault?
* How can I get more debug logs for this segmentation faults within GitLab CI?

set-soft, 2022-07-20:
Hi @myorbit !

Do you have copy of the YAML file that triggers this crash? (SIGSEGV can be only from KiCad, and yes this is most probably something removed by the pcb_print output that isn't 100% restored)

myorbit, 2022-07-25:
I currently cannot reproduce the segfault issue on the simple `test-zone-fills` project, only on my much more complicated board.

set-soft, 2022-07-25:
> I saw that the warning W045 pops up where the `dnf_filter` ist first applied, either at the gerber_variant 'print_asm_notes_variant'.

Warnings that are exactly the same are reported once.

> * How can I get more debug logs for this segmentation faults within GitLab CI?

You can't easily do it. You need to debug KiCad. 

> WARNING:(W045) Pad with solder paste, but no copper or solder mask aperture in U2 (kibot - out_base.py:337)

Can you create a test case where this warning is wrongly generated?

commit 513b6526 (2022-07-18): [PCB Print] Fixed problems with zones on multiple layers

- Now we remove them from the layer set
Closes #226