INTI-CMNB/KiBot #227: [BUG] Eurocircuits technology classification in design report does not match actual Eurocircuits classification

hkleen, 2022-07-15:
**Describe the bug**
For designs using vias with a very small OAR (less than .125 mm) the technology classification reported in the design report does not match the classification by Eurocircuits.

**To Reproduce**
For reference take the light_control test case. The minimum vias used are 20 mil (.508 mm) pads and 10 mil (.254 mm) hole. Taking into account plating will result in a .35 mm drill being used, thus the OAR would be 3 mil (.08 mm). The classification would be 10C as is reported.
By default Eurocircuits will reduce the finished hole diameter for small holes (can be configured, default is anything smaller than .45 mm) in order to increase the OAR value. So using the default settings the classification will be 6E since they reduced the requested finished hole size down to .1 mm (which is the minimum they are offering).
The via holes will be shrunk to .15 mm resulting in a .25 mm drill being used and an OAR of .125 mm. So using a smallest requested finished hole size of .15 mm would also work, resulting in a classification of 6D.

**Expected behavior**
Classification should take into account the drill size reduction applied by Eurocircuits. For the light_control the reported classification should be 6D.

set-soft, 2022-07-15:
Hi @hkleen !

I'm marking it as enhancement because from what you say the computed value is correct. We just need to add support for a special feature they offer. 

Do you know how they call it?  If I had to choose a name it will be something like "eurocircuits_adjust_small_drills"

I understand this is a default, but is an option.

Now about how it works:
- Do you have the URL for the document explaining it?
- Does it apply to all plated holes?
- How is applied? You mention the minimum is .1 mm, but then you say they will use .15 mm, which one is default? Is this something you select? Perhaps we should add a couple of options: 1) The size value where we start applying the rule (.45 mm) and 2) The adjusted size (.15 mm) both combined with one option to enable the mechanism (a total of 3 options)
-

hkleen, 2022-07-15:
Hi @set-soft !

I did some experimentation on the Eurocircuits order page.

I don't think the feature has an official name, at least I wasn't able to find one. The option in the order form is called "Holes <= may be reduced" and by default .45 mm is selected.  The tooltip for this option is: Final holes sizes smaller or equal to this given diameter can be reduced to accomodate the correct annular ring values."

As for it works (It's a little complicated):
- There is some documentation of this feature in the section on "defining Via holes" [here](https://www.eurocircuits.com/pcb-design-guidelines/drilled-holes/).
- result of my experiments: the fix is applied to any plated holes having an OAR of less than .125 mm (which is the minimum for their default, no additional charge,  technology) if the hole diameter is equal or smaller than the value selected for "Holes <= may be reduced". 
- The adjustment is performed, even if other structures on the PCB force a higher technology classification as long as the resulting finished hole is has a diameter > .25 mm.  For example if there are traces .1mm wide (resulting in class 8, so .1 mm OAR would be OK) the fix is still applied to plated holes with an OAR < .125 mm as long as the resulting finished hole is > .25 mm.  
- If a valid OAR can not be achieved  without changing the drill classification Eurocircuits will change the order settings to drill class "E" (.1 mm hole). The adjustments of the vias themselves is done to the extend necessary, for example .5 / .25 mm vias (OAR: 0.075 mm) will be changed to .5 / .15 mm (OAR: 0.125 mm). This would normally result in a class "D" (.15 mm hole) for the PCB, but Eurocircuits chooses class "E" by default (with an additional charge), but since the actual holes are all .15 mm the user can manually select class "D" (.15 mm holes) and this seems to work.

I hope this clears it up a little. Personally I think this Eurocircuits "feature" is kind of a mess as  it is not really obvious what will be changed for a given design and I don't really like the mfg changing my design data. But I guess it is what it is.

If you need any further information I could do some more experiments.

closed 2022-09-08 by set-soft

set-soft, 2022-09-08:
Hi @hkleen !

After releasing 1.3.0 I revisited it. Now KiBot reports 6D for the light_control example.
I'm quite sure a lot of cases aren't correctly supported, so feel free to report any case where it doesn't work.
The patch is quite large because I'm computing a lot of values adjusted to Eurocircuits criteria, so now we have tons of variants for the same value. I know some of them can be removed, but including them helped to make things symetric.
In the example case KiBot correctly identifies vias with 0.508 mm of pad and a drill size of 0.35 mm. As this is less than 0.45 mm (configurable), and they produce an OAR of less than 0.125 mm, KiBot is reducing the hole to 0.25 mm (0.15 mm finished). This changes things to 6D.

hkleen, 2022-09-12:
Hi @set-soft!

I did some checks on our designs and for all of them the technology class reported by KiBot matches the one reported by Eurocircuits. If I run into any issues in the future I will report them. For now everything seems to work just fine, thanks again!