INTI-CMNB/KiBot #229: [HOW]

dommae123, 2022-07-18:
**What do I want to achieve?**
I want to add a custom variable, that links to our custom-3D-model-lib-folder, in my case called BOS_LIB, where all our 3D Models of our company are located in. This lib is already stored in our repository and I want Kibot to get all the custom-3D-models from this repository folder.

**Do I have some configuration file (.kibot.yaml) that I'm using?**
I'm using the following yaml- config-files for the pipeline in azure:
```yaml
trigger: none

pool:
  vmImage: 'ubuntu-latest'

container: setsoft/kicad_auto:ki6

steps:
- script:
    kibot -d Generated/48V -g variant=48V -s all

- script: 
    kibot -d Generated/24V -g variant=24V -s all

- task: CopyFiles@2
  inputs:
    SourceFolder: '$(Pipeline.Workspace)/s'
    Contents: '**/.artifactignore'
    TargetFolder: '$(Pipeline.Workspace)'

- task: PublishPipelineArtifact@1
  inputs:
    targetPath: '$(Pipeline.Workspace)/s/Generated'
    publishLocation: 'pipeline'
    artifactName: Balancer
```
and the yaml config file for kibot:
```yaml
# This is a working example.
kibot:
  version: 1

#global:
#  environment: 
#    BOS_LIB: $/{BOS_LIB/}
preflight:
  run_erc: false
  run_drc: false
  check_zone_fills: true
  ignore_unconnected: true
  set_text_variables:
    - name: BOS_LIB
      command: "echo '../lib/3d_models'"

filters:
  - name: 'Variant rename'
    type: var_rename
    separator: ':'
    variant_to_value: true

variants:
  - name: 'default'
    comment: 'default'
    type: kibom

  - name: 48V
    comment: '48V Variant'
    type: kibom
    file_id: _48V
    variant: 48V
    pre_transform: 'Variant rename'

  - name: 24V
    comment: '24V Variant'
    type: kibom
    file_id: _24V
    variant: 24V
    pre_transform: 'Variant rename'

outputs:
- name: download_datasheet
  comment: downloads the datasheets
  type: download_datasheets
  dir: input-from-production/datasheets

- name: basic_pdf_sch_print
  comment: Schematic in PDF format
  type: pdf_sch_print
  dir: output-to-production/schematic_pdf

- name: gerber_modern
  comment: Gerbers in modern format, recommended by the standard
  type: gerber
  dir: output-to-production/gerbers
  layers:
  - layer: F.Cu
    suffix: F_Cu
    description: Front copper
  - layer: In1.Cu
    suffix: In1_Cu
    description: Inner layer 1
  - layer: In2.Cu
    suffix: In2_Cu
    description: Inner layer 2
  - layer: B.Cu
    suffix: B_Cu
    description: Bottom copper
  - layer: B.Paste
    suffix: B_Paste
    description: Bottom solder paste
  - layer: F.Paste
    suffix: F_Paste
    description: Front solder paste
  - layer: B.Silkscreen
    suffix: B_Silkscreen
    description: Bottom silkscreen (artwork)
  - layer: F.Silkscreen
    suffix: F_Silkscreen
    description: Front silkscreen (artwork)
  - layer: B.Mask
    suffix: B_Mask
    description: Bottom soldermask (negative)
  - layer: F.Mask
    suffix: F_Mask
    description: Front soldermask (negative)
  - layer: User.Drawings
    suffix: User_Drawings
    description: User drawings
  - layer: User.Eco1
    suffix: User_Eco1
    description: For user usage 1
  - layer: User.Eco2
    suffix: User_Eco2
    description: For user usage 2
  - layer: Edge.Cuts
    suffix: Edge_Cuts
    description: Board shape
  - layer: B.Courtyard
    suffix: B_Courtyard
    description: Bottom courtyard area
  - layer: F.Courtyard
    suffix: F_Courtyard
    description: Front courtyard area
  - layer: B.Fab
    suffix: B_Fab
    description: Bottom documentation
  - layer: F.Fab
    suffix: F_Fab
    description: Front documentation

- name: basic_position_CSV
  comment: Components position for Pick & Place
  type: position
  dir: output-to-production/pick-n-place
  options:
    format: CSV
    only_smd: false
    output: '%f_pos_%D.%x'

- name: gerber_drills
  comment: Gerber drill files
  type: gerb_drill
  dir: output-to-production/gerbers
  options:
    use_aux_axis_as_origin: false

- name: print_pdf
  comment: Assembly drawing PDF for the PCB
  type: pcb_print
  dir: output-to-production/PCB_assembly_drawing
  options:
    force_edge_cuts: false
    keep_temporal_files: false
    title: "Assembly Drawing"
    scaling: 1.0
    pages:
      - layers: [Edge.Cuts, F.Courtyard, F.Fab ]
        sheet: 'Fabrication layers front'
      - layers: [Edge.Cuts, B.Courtyard, B.Fab ]
        sheet: 'Fabrication layers back'
        mirror: true

- name: basic_render_3d_30deg_top
  comment: 3D view from 30 degrees top
  type: render_3d
  dir: output-to-production/PCB_3D_view
  output_id: top_view
  options:
    ray_tracing: true
    rotate_x: 3
    rotate_z: -1
    download: true
    zoom: 6
    height: 1000
    width: 1600

- name: basic_render_3d_30deg_bottom
  comment: 3D view from 30 degrees bottom
  type: render_3d
  dir: output-to-production/PCB_3D_view
  output_id: bottom_view
  options:
    ray_tracing: true
    rotate_x: 3
    rotate_z: -1
    download: true
    zoom: 6
    view: bottom
    height: 1000
    width: 1600

- name: iBoM
  comment: Interactive HTMl BoM
  type: ibom
  dir: output-to-production/BoM/iBoM
  options:
    highlight_pin1: false

- name: bom_html
  comment: Simple HTML BoM
  type: bom
  dir: output-to-production/BoM/simple_BoM
  options:
    format: HTML
    html:
      datasheet_as_link: MFP
      title: 'KiBot variants example'
      highlight_empty: false

- name: step
  comment: Creates step files from the PCB
  type: step
  dir: Step_temp
  options:
    download: true
    no_virtual: true

- name: 2d_render_pcb_top
  comment: Beautiful 2D-Rendering of the PCB
  type: pcbdraw
  dir: output-to-production/PCB_2D_view
  options:
    bottom: false
    format: png

- name: 2d_render_pcb_botoom
  comment: Beautiful 2D-Rendering of the PCB
  type: pcbdraw
  dir: output-to-production/PCB_2D_view
  options:
    bottom: true
    format: png

- name: Zip Gerbers and Drills
  comment: ZIP file
  type: compress
  dir: output-to-production/gerbers
  options:
    files:
      - from_output: gerber_drills
      - from_output: gerber_modern
    move_files: true

- name: Zip Step File
  comment: step files can become huge -> zip
  type: compress
  dir: output-to-cad
  options:
    files:
      - from_output: step
    move_files: true
```

**My Environment:**
I'm running Kibot using the docker-image "setsoft/kicad_auto:ki6"

**Example of thrown Error Messages while running Kibot**
`ERROR:Unable to expand 'KICAD6_3DMODEL_DIR' in '${KICAD6_3DMODEL_DIR}/Package_TO_SOT_SMD.3dshapes/SOT-23.wrl' (kibot - config.py:71)
WARNING:(W047) Missing 3D model for Q30: '/__w/1/s/${KICAD6_3DMODEL_DIR}/Package_TO_SOT_SMD.3dshapes/SOT-23.wrl' (kibot - out_base_3d.py:131)
`
and
`
ERROR:Unable to expand 'BOS_LIB' in '${BOS_LIB}/BOS_LIB.3dshapes/INGUN_KS-100_WL_0.STEP' (kibot - config.py:71)
WARNING:(W047) Missing 3D model for TP96: '/__w/1/s/${BOS_LIB}/BOS_LIB.3dshapes/INGUN_KS-100_WL_0.STEP' (kibot - out_base_3d.py:131)
`

So there seems to be an issue with the `${KICAD6_3DMODEL_DIR}` and my custom `${BOS_LIB}`. Do you guys have any ideas that can help me?

set-soft, 2022-07-18:
Hi @dommae123 !

Did you try doing what [kibot_variants_arduprog](https://github.com/INTI-CMNB/kibot_variants_arduprog) does?
The models are stored in the `3D` dir of the repo and then the footprints say `${KIPRJMOD}/3D/CP_Elec_8x10.wrl`. The `KIPRJMOD` is defined by KiCad and points to where the project file is located.

In the config you can change:

```yaml
  set_text_variables:
    - name: BOS_LIB
      command: "echo '../lib/3d_models'"
```

By:

```yaml
  set_text_variables:
    - name: BOS_LIB
      text: '../lib/3d_models'
```

To make it simpler. But this won't work if you use `-s all`, because you'll be skipping `set_text_variables`. Just remove the `-s all`, you don't have any slow thing to skip anyways. Text variables should work for what you want.

About `KICAD6_3DMODEL_DIR`: you can run KiBot with debug info (`-vvv`) to get more information about what KiCad environment variables are detected. You can also try running the development image (`setsoft/kicad_auto:dev_ki6`) It has some differences in the KiCad 6 environment stuff. But: the docker image doesn't contain the 3D models, they are downloaded on demand.

I'm closing this as issue, if you find that the above mentioned mechanisms fail to work please open an specific issue to fix it.

closed 2022-07-18 by set-soft

dommae123, 2022-07-20:
Hi Salvador,
thanks for your replay.

Your are right, I use kibot mainly to handle the variants in our project, and it works like a charm! Thanks for your work!

The 3D Model problem is not vital for us, but it would be nice to see the variants also in the 3D-rendered view of the PCB. 

Regarding your questions/quotes:
(1) Our 3d models are stored in a folder in our repository, and the footprints says:`${BOS_LIB}/BOS_LIB.3dshapes/SJ121381_JAE_4-POLIG.STEP`. I can't change the ${BOS_LIB} variable because we use it in every project. 
(2) if I remove the `-s all` part and add your `set-text-variables`-section, then the following error appears:
`ERROR:In preflight set_text_variables: Trying to define KiCad 6 variables but the project is missing (None) (kibot.pre_base - pre_base.py:82)`

So what can I do about that? any idea?

PS: Don't know which specific issue that problem is related to, so I'm just answering here. If you want, you can rename it and open a certain new issue.

set-soft, 2022-07-20:
Hi @dommae123 !

The `Trying to define KiCad 6 variables but the project is missing` means that KiBot couldn't find a the KiCad project file (`PCB_NAME.kicad_pro`). The text variables are stored there.

dommae123, 2022-07-28:
This problem is solved for me. The "set_text_variables" did work out well in the end. I could include our custom lib with that and we are very happy using Kibot now. Thanks for your excellent work. Good kharma to you =)