INTI-CMNB/KiBot #242: [BUG]

dommae123, 2022-07-28:
Hi there,

I have found a bug when creating variants with Kibot (used in Docker-image kicad-auto:ki6). There seems to be a failure output in the i(nteractive)Bom, when having different values for each variant. (For example two variants: "12V" and "48V" and the different values are passed by the "12V:Value" & "48V:value" fields). In the iBom, there seems to be no change for the two variants in the value field. Whereas in the simple BOM, there is a change in the values for each variant. 

Here is my configuration in the schematic-editor:
[image]

For the iBom in the yaml file, I just used:
```
- name: iBoM
  comment: Interactive HTMl BoM
  type: ibom
  dir: output-to-production/BoM/iBoM
  options:
    highlight_pin1: false
```

The output for the 12V-Variant in the iBom is as follows:
[image]

The output  for the 12V-Variant in the normal bom is like this:
[image]

Hope that the problem is sufficently explained and someone can help me with that problem. It's more or less essential for us, because we use the iBOM everywhere in our processes. 

Thanks,
Thomas

set-soft, 2022-07-29:
I think the problem is that you need to apply a filter to your iBoM output. The iBoM tool knows nothing about "VARIANT:FIELD" mechanism. You have to tell KiBot to convert it. I don't know how the variant is applied in your configuration, but you must ensure that a `var_rename` is applied to the iBoM output.

dommae123, 2022-07-29:
Hi and thanks again for your rapid answer!

I think I already applied a var_rename-filter, which works for every output-type, but obviously not in the output of the iBoM. And after reading the Kibot manual, I can't really find out how to do that in iBoM :/ There seems to be a variant-option in the iBoM, but that is probably not what you wanted to propose.

At the Moment, my kibot-yaml-file looks like this:

```
##################################
####### HEADER SECTION ###########
##################################
kibot:
  version: 1

preflight:
  run_erc: false
  run_drc: false
  check_zone_fills: false
  ignore_unconnected: true
  set_text_variables:
    - name: COX_LIB
      text: '/__w/1/s/KiCad/libraries'

global:
  environment:
    models_3d: https:////gitlab.com/kicad/libraries/kicad-packages3D/-/raw/master

# If there are variants in your PCB, add them here under variants. Like this example:
#  - name: 48V
#    comment: '48V Variant'
#    type: kibom
#    file_id: _48V
#    variant: 48V
#    pre_transform: 'Variant rename'
#
# if there is no variant (means only 1 variant), then leave the default variant 

filters:
  - name: 'Variant rename'
    type: var_rename
    separator: ':'
    variant_to_value: true

variants:
  - name: 48V
    comment: '48V Variant'
    type: kibom
    file_id: _48V
    variant: 48V
    pre_transform: 'Variant rename'
  - name: 12V
    comment: '12V Variant'
    type: kibom
    file_id: _12V
    variant: 12V
    pre_transform: 'Variant rename'

##################################
####### OUTPUT SECTION ###########
##################################

outputs:

- name: basic_pdf_sch_print
  comment: Schematic in PDF format
  type: pdf_sch_print
  dir: output-to-production/schematic_pdf

- name: gerber_modern
  comment: Gerbers in modern format, recommended by the standard
  type: gerber
  dir: output-to-production/gerbers
  layers:
  - layer: F.Cu
    suffix: F_Cu
    description: Front copper
  - layer: In1.Cu
    suffix: In1_Cu
    description: Inner layer 1
  - layer: In2.Cu
    suffix: In2_Cu
    description: Inner layer 2
  - layer: B.Cu
    suffix: B_Cu
    description: Bottom copper
  - layer: B.Paste
    suffix: B_Paste
    description: Bottom solder paste
  - layer: F.Paste
    suffix: F_Paste
    description: Front solder paste
  - layer: B.Silkscreen
    suffix: B_Silkscreen
    description: Bottom silkscreen (artwork)
  - layer: F.Silkscreen
    suffix: F_Silkscreen
    description: Front silkscreen (artwork)
  - layer: B.Mask
    suffix: B_Mask
    description: Bottom soldermask (negative)
  - layer: F.Mask
    suffix: F_Mask
    description: Front soldermask (negative)
  - layer: User.Eco1
    suffix: User_Eco1
    description: For user usage 1
  - layer: User.Eco2
    suffix: User_Eco2
    description: For user usage 2
  - layer: Edge.Cuts
    suffix: Edge_Cuts
    description: Board shape

- name: basic_position_CSV
  comment: Components position for Pick & Place
  type: position
  dir: output-to-production/pick-n-place
  options:
    format: CSV
    only_smd: false
    output: '%f_pos_%D.%x'

- name: gerber_drills
  comment: Gerber drill files
  type: gerb_drill
  dir: output-to-production/gerbers
  options:
    use_aux_axis_as_origin: false

- name: print_pdf
  comment: Assembly drawing PDF for the PCB
  type: pcb_print
  dir: output-to-production/PCB_assembly_drawing
  options:
    force_edge_cuts: false
    keep_temporal_files: false
    title: "Assembly Drawing"
    scaling: 1.0
    pages:
      - layers: [Edge.Cuts, F.Courtyard, F.Fab ]
        sheet: 'Fabrication layers front'
      - layers: [Edge.Cuts, B.Courtyard, B.Fab ]
        sheet: 'Fabrication layers back'
        mirror: true

- name: basic_render_3d_30deg_top
  comment: 3D view from 30 degrees top
  type: render_3d
  dir: output-to-production/PCB_3D_view
  output_id: _view
  options:
    ray_tracing: true
    rotate_x: 3
    rotate_z: -1
    download: true
    zoom: 6
    height: 1440
    width: 2560
    no_virtual: true

- name: basic_render_3d_30deg_bottom
  comment: 3D view from 30 degrees bottom
  type: render_3d
  dir: output-to-production/PCB_3D_view
  output_id: _view
  options:
    ray_tracing: true
    rotate_x: 3
    rotate_z: -1
    download: true
    zoom: 6
    view: bottom
    height: 1440
    width: 2560
    no_virtual: true

- name: iBoM
  comment: Interactive HTMl BoM
  type: ibom
  dir: output-to-production/BoM/iBoM
  options:
    highlight_pin1: false

- name: bom_html
  comment: Simple HTML BoM
  type: bom
  dir: output-to-production/BoM/simple_BoM
  options:
    format: HTML
    html:
      datasheet_as_link: MFP
      title: 'KiBot variants example'
      highlight_empty: false

- name: step
  comment: Creates step files from the PCB
  type: step
  dir: Step_temp
  options:
    download: true
    no_virtual: true

- name: 2d_render_pcb_top
  comment: Beautiful 2D-Rendering of the PCB
  type: pcbdraw
  dir: output-to-production/PCB_2D_view
  options:
    bottom: false
    format: png
    show_components: all

- name: 2d_render_pcb_botoom
  comment: Beautiful 2D-Rendering of the PCB
  type: pcbdraw
  dir: output-to-production/PCB_2D_view
  options:
    bottom: true
    format: png
    show_components: all

- name: download_datasheet
  comment: downloads the datasheets
  type: download_datasheets
  dir: input-from-production/datasheets
  options:
    output: ${Manufacturer_Name}__${Manufacturer_Part_Number}.pdf

- name: Zip Gerbers and Drills
  comment: ZIP file
  type: compress
  dir: output-to-production/gerbers
  options:
    files:
      - from_output: gerber_drills
      - from_output: gerber_modern
    move_files: true

- name: Zip Step File
  comment: step files can become huge -> zip
  type: compress
  dir: output-to-cad
  options:
    files:
      - from_output: step
    move_files: true

# add more outputs here, if you want!
```

set-soft, 2022-07-29:
Hi @dommae123 !
I took a look and iBoM output has only limited support for variants, just supports removing components.
I'll take a look at how hard is to add support for other stuff.

closed 2022-07-30 by set-soft

set-soft, 2022-07-30:
Hi @dommae123 !

The above path should solve the problem. It stores the modified fields in a fake netlist and in KiCad 6's properties fields. 

This will be available in the `dev_k6` image as soon as the patches passes all the regression tests (won't be added until any detectable side effect is fixed).

Please let me know if it works for you, iBoM is a particular case where data from the Schematic needs to be passed to the PCB, which isn't normal in KiCad 5.

Note that I marked it as `bug` because it is surprising for the user that it doesn't work, but internally it wasn't implemented, there are a lot of outputs and a lot of things you can do with filters and variants. In cases like iBoM they need special handling and is implemented as people needs it.

dommae123, 2022-07-30:
Hi @set-soft ,

thanks for your help so far. 🥇  But I can't tell you if it works for me. Instead of using the docker-image, I installed kibot on my Azure_server manually by `sudo apt-get install -f kibot` (see code below). And with that, it didn't show any changes in the iBoM.

But I guess, this is not the way the implement your changes from the commit-path above. I guess I downloaded a kibot-version without your newly implemented changes. 
So I can't tell you at the moment unfortunately.

```
##################################################################################
########### START: CHANGE ONLY THIS PART #########################################
##################################################################################
variables: 
- name: 'PCB' 
  value: 'LUX_Battery_Power' # <-  Name of the folder of the PCB
- name: 'version' 
  value: 'V1.0_dev' # <- Name of the folder of the PCB Version

parameters: # if there are variants, list them with '- var1; - var2; - ...' under default-section   if not, then just write '- default'
  - name: variants
    type: object
    default:
      - 48V # <-
      - 12V # <-
##################################################################################
########### END: CHANGE ONLY THIS PART ###########################################
##################################################################################

trigger: none

#pool:
#  vmImage: 'ubuntu-latest'

#container: setsoft/kicad_auto:ki6

################

pool:
  vmImage: 'ubuntu-latest'
steps:
- script: |
    sudo add-apt-repository "deb http://de.archive.ubuntu.com/ubuntu jammy main restricted universe multiverse"
    sudo add-apt-repository "deb http://de.archive.ubuntu.com/ubuntu jammy-updates main restricted universe multiverse"
    sudo add-apt-repository "deb http://de.archive.ubuntu.com/ubuntu jammy-security main restricted universe multiverse"
    sudo add-apt-repository "deb http://de.archive.ubuntu.com/ubuntu jammy-backports main restricted universe multiverse"
    wget https://set-soft.github.io/debian/kibot.list
    sudo cp kibot.list /etc/apt/sources.list.d/
    wget https://set-soft.github.io/debian/kibot.gpg
    sudo cp kibot.gpg /etc/apt/
    sudo apt-get update
    sudo apt-get install -f kibot
    kibot-check

###############

#####
#steps:
- ${{ each variant in parameters.variants }}:
  - script:
      kibot -c ../$(PCB).kibot.yaml -d Generated/${{ variant }} -g variant=${{ variant }}
    displayName: 'KiBot Power ${{ variant }}'
    workingDirectory: './KiCad/projects/LUX/$(PCB)/$(VERSION)'

- task: CopyFiles@2
  inputs:
    SourceFolder: '$(Pipeline.Workspace)/s/KiCad/projects/LUX/'
    Contents: '*.artifactignore'
    TargetFolder: '$(Pipeline.Workspace)/s/KiCad/projects/LUX/$(PCB)/$(VERSION)/Generated'

- task: PublishPipelineArtifact@1
  inputs:
    targetPath: '$(Pipeline.Workspace)/s/KiCad/projects/LUX/$(PCB)/$(VERSION)/Generated'
    publishLocation: 'pipeline'
    artifactName: $(PCB)
```

set-soft, 2022-08-01:
Hi @dommae123 !

> But I guess, this is not the way the implement your changes from the commit-path above.

Right, it won't include the changes. The .deb files correspond to stable releases.

In order to test the last code you can add:

```
    sudo dpkg --remove kibot 
    curl https://codeload.github.com/INTI-CMNB/KiBot/zip/refs/heads/master --output pp.zip
    unzip pp.zip
    sudo pip3 install --no-compile KiBot-master/
    rm -rf KiBot-master pp.zip
```

Do it after the `sudo apt-get install -f kibot` and before the `kibot-check`
The first line removes the stable KiBot (but not it dependencies), then we download the current git code and install it using pip

dommae123, 2022-08-26:
Hi @set-soft ,
so I tested your implementation for the variants in iBom for all of my kicad-projects and most of my projects worked perfectly. However, for one project, there still seems to be a problem corresponding to the out_ibom.py-file in line 180.

Here is my console output for this problem:
```
Starting: KiBot Balancer 48V
==============================================================================
Task         : Command line
Description  : Run a command line script using Bash on Linux and macOS and cmd.exe on Windows
WARNING:(W047) Missing 3D model for FB2: `/__w/1/s/KiCad/projects/LUX/LUX_Battery_Balancer/V1.0_dev/https:/gitlab.com/kicad/libraries/kicad-packages3D/-/raw/master/Inductor_SMD.3dshapes/L_1206_3216Metric.wrl` (kibot - out_base_3d.py:131)
WARNING:(W047) Missing 3D model for RV3: `/__w/1/s/KiCad/projects/LUX/LUX_Battery_Balancer/V1.0_dev/https:/gitlab.com/kicad/libraries/kicad-packages3D/-/raw/master/Potentiometer_SMD.3dshapes/Potentiometer_Bourns_3314G_Vertical.wrl` (kibot - out_base_3d.py:131)
WARNING:(W058) Old file format detected, please convert it to KiCad 6 if experimenting problems (pcbnew_do - interposer.py:569) (kibot - kiplot.py:126)
- '3D view from 30 degrees bottom' (basic_render_3d_30deg_bottom) [render_3d]
- 'Interactive HTMl BoM' (iBoM) [ibom]
Traceback (most recent call last):
  File "/usr/local/bin/kibot", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.9/dist-packages/kibot/__main__.py", line 342, in main
    generate_outputs(outputs, args.target, args.invert_sel, args.skip_pre, args.cli_order, args.no_priority,
  File "/usr/local/lib/python3.9/dist-packages/kibot/kiplot.py", line 441, in generate_outputs
    _generate_outputs(outputs, targets, invert, skip_pre, cli_order, no_priority, dont_stop)
  File "/usr/local/lib/python3.9/dist-packages/kibot/kiplot.py", line 432, in _generate_outputs
    run_output(out, dont_stop)
  File "/usr/local/lib/python3.9/dist-packages/kibot/kiplot.py", line 359, in run_output
    out.run(get_output_dir(out.dir, out))
  File "/usr/local/lib/python3.9/dist-packages/kibot/out_base.py", line 162, in run
    self.options.run(self.expand_filename(output_dir, self.options.output))
  File "/usr/local/lib/python3.9/dist-packages/kibot/out_ibom.py", line 180, in run
    GS.sch.save_netlist(f, self._comps)
  File "/usr/local/lib/python3.9/dist-packages/kibot/kicad/v5_sch.py", line 2017, in save_netlist
    self.save_netlist_libparts(root)
  File "/usr/local/lib/python3.9/dist-packages/kibot/kicad/v5_sch.py", line 1954, in save_netlist_libparts
    libpart.set('part', res[1])
IndexError: list index out of range
##[error]Bash exited with code '1'.
Finishing: KiBot Balancer 48V
```

At the moment, I don't know what the difference of this project is in comparism to my other projects, where iBom works perfectly. Maybe you already have an idea?! If not, just tell me what you need from me in order to fix this together. Hoping for your support again. Thanks in advance...

Thomas

dommae123, 2022-08-26:
Hi @set-soft,
it seems, that I just found the reason for the problem in the project, that didn't work so far. There are 3 parts, in which there is no "lib" written in the netlist. Have a look at the netlist of the schematic:
[image]
When there is not "lib" given, then the split-command doesn't work and the list is out of range.

As soon as I deleted the corresponding parts with no lib written in the netlist, the project worked just fine and the iBom and all the rest of the kibot-outputs were generated without problem.

Just wanted to let you know, so that you can add something to your code, maybe some debug outputs for the users, that might get the same problem in the future.

Greetings from Germany,
Thomas

set-soft, 2022-08-29:
Hi @dommae123 !
Thanks for the info, I think this should work now.

commit 3b6c9e7d (2022-07-30): [iBoM] Added support for variants that change component fields

Closes #242