INTI-CMNB/KiBot #244: [BUG] No pcbdraw output

s-ol, 2022-07-29:
**Describe the bug**
Running kibot no error is produced, but the pcbdraw output is simply missing.

```
kibot -b pcb/pcb.kicad_pcb -e pcb/pcb.kicad_sch -c config.kibot.yaml
- '' (ibom) [ibom]
WARNING:(W037) Field Mouser# of component U1 (USBLC6-2SC6) contains extra spaces: ` 511-USBLC6-2SC6 ` removing them. (kibot - v5_sch.py:1029)
- '' (bom) [bom]
- '' (pnp) [position]
- '' (pcb) [pcbdraw]
- '' (schematic) [svg_sch_print]
WARNING:(W058) Missing KiCad main config file /root/.config/kicad/6.0/kicad_common.json (eeschema_do - misc.py:179) (kibot - kiplot.py:164)
Found 21 unique warning/s (40 total)
```

**To Reproduce**
Kibot is being run from the docker image. The `kibot` invocation above is a macro for this:

```
function kibot
  docker run --rm -it -v /tmp/.X11-unix:/tmp/.X11-unix -e DISPLAY=$DISPLAY \
      -v $(pwd):$HOME/workdir \
      --env NO_AT_BRIDGE=1 \
      --workdir="$HOME/workdir" \
      setsoft/kicad_auto:ki6 kibot "$@"
end
```

Here's the config file:

```
kibot:
  version: 1

preflight:
  check_zone_fills: true
  run_drc: false
  run_erc: false

  set_text_variables:
    - name: 'VCS_DATE'
      command: 'git log -1 --format="%as"'

    - name: 'VCS_REV'
      command: 'git rev-parse --short HEAD'

    - name: 'VCS_TAG'
      command: 'git describe --tags --exact-match HEAD || echo "r?"'

global:
  dir: 'dist/%d_0x33.board_%r'
  output: '%i.%x'
  solder_mask_color: 'black'
  silk_screen_color: 'white'

filters:
  - name: 'exclude_virtual'
    type: 'generic'
    exclude_any:
      - column: Footprint
        regex: 'Jumper'
      - column: Value
        regex: '^PMOD/GPIO$'
      - column: Value
        regex: 'Outline'

outputs:
  - type: 'ibom'
    name: 'ibom'
    options:
      layer_view: 'B'
      dnf_filter: exclude_virtual

  - type: 'bom'
    name: 'bom'
    options:
      exclude_filter: exclude_virtual

  - type: 'position'
    name: 'pnp'
    options:
      output: 'placement.%x'
      separate_files_for_front_and_back: false

  - type: 'pcbdraw'
    name: 'pcb'
    options:
      output: 'pcb.%x'
      format: 'svg'
      bottom: true

  - type: 'svg_sch_print'
    name: 'schematic'
```

The board repository can be found at https://git.s-ol.nu/kb/0x33.board/hardware/-/

**Expected behavior**
There should be a `pcb.svg` in `dist/...`

**Environment (please complete the following information):**
Where are you running KiBot:
- Using a docker image: `kicad_auto:ki6`

set-soft, 2022-07-29:
Hi @s-ol !

I can't access the above URL, I get a connection reset.

I tried using one of the test examples and it worked with the current code.

Try running KiBot with debug information (i.e. `-vvv`) and look for the pcbdraw output information to see if it helps, post the information here.

s-ol, 2022-07-29:
I just figured out that it seems to be the `pcb.svg` name that causes this, if I set it to `output: "stuff.%x"` for example it shows up.

set-soft, 2022-07-29:
I didn't need it with the current code, I got `pcb.svg`

s-ol, 2022-07-29:
Sorry, fixed the repo link above, it was supposed to be: https://git.s-ol.nu/hw/0x33.board/hardware/-/

Could it be that the issue is caused by the `pcb` part matching the name of the project (`pcb.kicad_pcb` in this case)?

set-soft, 2022-07-29:
> Could it be that the issue is caused by the `pcb` part matching the name of the project (`pcb.kicad_pcb` in this case)?

Now I see the problem: you are printing the schematic SVG to the same folder as the PcbDraw SVG. The problem is that the default name for the schematic is PROJECT.FORMAT, KiBot then renames it to what you want, but in this case you are generating the PcbDraw first, named `pcb.svg` and then the schematic, which temporally has the same name.

I took note about it. You should avoid this collision until the problem is fixed. Just choose another name for the PcbDraw output, or use different target dirs.

set-soft, 2022-08-11:
Hi @s-ol !
This problem should be solved now. The WIP KiAuto (2.0.0) now supports selecting the file name for the SVG. It takes care of moving any file with the name that KiCad uses by default and restores it after finishing the job. Is not a recommended usage, but you won't loose the file ;-)

closed 2022-08-11 by set-soft

s-ol, 2022-08-12:
awesome!

commit 5785fd3e (2022-08-11): [SCH Print][Netlist][Fixed] Problems with name collisions

- When the name chosen by KiCad was an alredy existing file (from
  another output) it was removed by KiAuto. So then KiCad created
  the file, we renamed it to what the user selected and ... well,
  the original file was lost. Now KiAuto takes care of this.
Fixes #244