INTI-CMNB/KiBot #248: [FEATURE] Add extra BOM lines in KiBot BOM output

cdwilson, 2022-08-12:
**Is your feature request related to a problem? Please describe.**
In my assembly BOM, I need to include the schematic and PCB part numbers.  Currently, I'm using symbols in the schematic to get these part numbers to show up in the output assembly BOM.

[image]

**Describe the solution you'd like**
Using schematic symbols just to add the schematic and PCB part numbers to the BOM feels like a hack.  I'd like to be able to include extra BOM lines in the KiBot `bom` output simply by adding some additional configuration in the `config.kibot.yaml` file (rather than having to use symbols inside of KiCad).

set-soft, 2022-08-12:
Hi @cdwilson !
DId you try the `extra_info` option of HTML and XLSX formats?

cdwilson, 2022-08-12:
Yes, but from what I can tell, this can't be used to add extra rows to the BOM.  I need these part numbers to be added as normal rows in the BOM, so that I can import the CSV into InvenTree and have the schematic & PCB P/N's included in the assembly-level BOMs.

[image]

FYI, this isn't a super high priority request since I can already achieve this using the schematic symbols, but I wanted to mention it as a nice-to-have and see if anybody else has a need for something like this (if not, feel free to close)

set-soft, 2022-08-12:
Hi @cdwilson !
Ok, let me see if I understand: you want to add fake components to the BoM?
At the same time you don't want to make them visible, right?
I think you can try creating a complement schematic with these components. Then use the aggregate feature to get the components from this complementary schematic.
I don't like the idea of adding extra lines because they will need special handling and they'll have a lot of associated complexity. Using components, as you are doing, provides a lot of control about which columns are used, where is the rows positioned (using the reference and values to force the position in the sorting), etc.
If the real problem is that you don't like them to be visible then using aggregate you'll get it solved.

cdwilson, 2022-08-12:
> At the same time you don't want to make them visible, right?

It's not so much about visibility (I don't mind that they are visible), just more about the flexibility to add these components without having to put them into the schematic at all. (I was trying to replicate a workflow I used at a previous job where we added these part numbers to the BOM using an external configuration file, rather than adding them into the schematic.)

> I don't like the idea of adding extra lines because they will need special handling and they'll have a lot of associated complexity.

If this adds extra complexity and special handling, then I agree, probably not a good idea.  I'll close this and just stick to using the schematic symbols.

closed 2022-08-12 by cdwilson

set-soft, 2022-08-12:
Hi @cdwilson !
A think that we could add in the future is some mechanism to import a CSV in the aggregate list, so you could import an arbitrary BoM. I'll take note about this.

set-soft, 2022-09-13:
Hi @cdwilson !
The above patch allows importing components from a simple CSV file.

cdwilson, 2022-09-14:
Hi @set-soft, I'm testing out the new CSV aggregate feature.

The documentation says:

> The Reference and Value are mandatory, in most cases Part is also needed.

However, I'm getting the following error unless I define an additional column named `Footprint`:

```
Traceback (most recent call last):
  File "/usr/local/bin/kibot", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.9/dist-packages/kibot/__main__.py", line 363, in main
    generate_outputs(outputs, args.target, args.invert_sel, args.skip_pre, args.cli_order, args.no_priority,
  File "/usr/local/lib/python3.9/dist-packages/kibot/kiplot.py", line 468, in generate_outputs
    _generate_outputs(outputs, targets, invert, skip_pre, cli_order, no_priority, dont_stop)
  File "/usr/local/lib/python3.9/dist-packages/kibot/kiplot.py", line 459, in _generate_outputs
    run_output(out, dont_stop)
  File "/usr/local/lib/python3.9/dist-packages/kibot/kiplot.py", line 386, in run_output
    out.run(get_output_dir(out.dir, out))
  File "/usr/local/lib/python3.9/dist-packages/kibot/out_base.py", line 163, in run
    self.options.run(self.expand_filename(output_dir, self.options.output))
  File "/usr/local/lib/python3.9/dist-packages/kibot/out_bom.py", line 854, in run
    do_bom(output, format, comps, self)
  File "/usr/local/lib/python3.9/dist-packages/kibot/bom/bom.py", line 546, in do_bom
    groups = group_components(cfg, comps)
  File "/usr/local/lib/python3.9/dist-packages/kibot/bom/bom.py", line 483, in group_components
    g.update_fields(cfg.conv_units, cfg.bottom_negative_x, x_origin, y_origin, cfg.angle_positive,
  File "/usr/local/lib/python3.9/dist-packages/kibot/bom/bom.py", line 354, in update_fields
    self.fields[ColumnList.COL_FP_LIB_L] = comp.footprint_lib
AttributeError: 'SchematicComponent' object has no attribute 'footprint_lib'
```

In my case, I'm using this feature to add a document part number to the BOM which does not have a footprint.

**Question 1:** Is it possible to make the footprint column optional?  

> in most cases `Part` is also needed

I found that I did not need to specify `Part` in my CSV and the import still worked OK.  The minimal CSV that worked for me was:

```csv
Reference,Quantity per PCB,Value,Footprint
PCB1,1,100037,foo
```

**Question 2:** In what specific cases is `Part` needed? (i.e. when will KiBot throw an error when `Part` is not specified?)

> The `Reference` and `Value` are mandatory

There are some cases where line items need to be included on the BOM, but they do not have associated reference designators (bare PCB fab, mechanical parts, adhesive, conformal coating material, etc).  When I worked in a PCB assembly shop, we saw this case frequently where customers provided BOMs where some of the BOM lines had no reference designators, but the assembly drawing called out where the part/material was intended to be included on the final assembly.

I think in the ideal case, only the `Quantity per PCB` and `Value` fields should be required, and all other fields should be optional.  (Manufacturing software I've used in the past, e.g. [CalcuQuote](https://www.calcuquote.com/), only _required_ `Quantity per PCB` and `Manufacturer Part Number`, but since KiCad doesn't have an explicitly defined `Manufacturer Part Number` field, I think it's OK to make `Value` required instead).

For example, I'd like to be able to aggregate a minimal CSV that looks like this:

```csv
Quantity per PCB,Value
1,100037
```

**Question 3:** Is it possible to make the `Reference` field optional for the CSV import case?

cdwilson, 2022-09-15:
@set-soft one more thing I noticed while testing this out:

When I generate the aggregated BOM, I now get multiple sections from the individual BOM inputs to the aggregate BOM:

[image]

I think this makes sense when you're aggregating two totally different schematics, but in my case I'm just using this to add one or two non-schematic parts to a single BOM.  It adds a lot of extra noise to the BOM output.

Would you be open to adding option to the `aggregate` BOMs that consolidates them into the parent BOM?

[image]

cdwilson, 2022-09-15:
BTW, let me know if you'd like me to put additional feedback on this topic somewhere else (discussion, other issue, etc).

set-soft, 2022-09-15:
> However, I'm getting the following error unless I define an additional column named `Footprint`:

Thanks, fixed in 76742bb69fcff68e7a04e14cf494e758c865ee41

set-soft, 2022-09-15:
> **Question 2:** In what specific cases is `Part` needed? (i.e. when will KiBot throw an error when `Part` is not specified?)

The main problem is when you use the default grouping criteria and you add passive components that are already in your BoM. Suppose you have 10 10k resistors, and now you add another one. If you don't provide a part name KiBot will assume using 10k is OK, but then it won't be grouped with the other 10 resistors because the have R as part, not 10k.

Note that using Part = Value usually works for I.C. because they have the same Part and Value in KiCad.

set-soft, 2022-09-15:
> **Question 3:** Is it possible to make the `Reference` field optional for the CSV import case?

Hmmm ... I don't think so. The reference is used to identify the component, is the most important value. It helps to link it with the PCB and to apply filters. Why should you need it? In the worst case provide a reference, just make it unique, like start numbering them XX1, XX2, etc. This will help you identify problems because KiBot will say the problem is with XX1, which can have all the other fields equal to XX2.

cdwilson, 2022-09-15:
> The reference is used to identify the component, is the most important value. It helps to link it with the PCB and to apply filters. Why should you need it?

You have a much better understanding of the KiBot filters than I do, so there may be internal reasons or something obvious why KiBot needs every BOM line item to have a reference designator.  I agree with you that for any component which has a reference designator on the PCB, there should always be a corresponding reference designator in the BOM.

However, my use case for this feature is to add parts to the BOM which don't have any reference designator on the PCB (anything that has a refdes on the PCB I already have in the schematic).  For example, the bare PCB itself has a part number (different from the PCB Assembly P/N), so it needs to be included in the PCB Assembly BOM.  However, the KiCad board file does not have any reference designator for the bare PCB itself.  So creating a reference designator like `PCB1` doesn't have any purpose except to make KiBot happy (the PLM system where the Assembly BOM is stored doesn't care if a line item has reference designators or not, it only cares about the quantity per board and part number).  Similarly, the schematic can have a unique P/N that needs to be added to the Assembly BOM, and there may be actual physical components like screws, or an RF shield cover, or conformal coating that need to be added to the BOM, but will not have reference designators on the PCB.

> In the worst case provide a reference, just make it unique, like start numbering them XX1, XX2, etc.

I don't want you to have to rework a bunch of KiBot stuff to make this work, so I can totally do this (it's what I'm doing now).  The only reason why I don't want to do this is because it requires some attention to detail to make sure these "fake" reference designators are unique (they are not automatically annotated, so there is a potential to screw it up by mistake).

Also, if my use case is abnormal, another option is that I could probably just post-process the BOM external to KiBot to add these additional components (shouldn't be too hard with a python script).

cdwilson, 2022-09-15:
Testing out the `csv_accept_no_ref` and getting the following error:

```
Traceback (most recent call last):
  File "/usr/local/bin/kibot", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.9/dist-packages/kibot/__main__.py", line 368, in main
    generate_outputs(outputs, args.target, args.invert_sel, args.skip_pre, args.cli_order, args.no_priority,
  File "/usr/local/lib/python3.9/dist-packages/kibot/kiplot.py", line 468, in generate_outputs
    _generate_outputs(outputs, targets, invert, skip_pre, cli_order, no_priority, dont_stop)
  File "/usr/local/lib/python3.9/dist-packages/kibot/kiplot.py", line 459, in _generate_outputs
    run_output(out, dont_stop)
  File "/usr/local/lib/python3.9/dist-packages/kibot/kiplot.py", line 386, in run_output
    out.run(get_output_dir(out.dir, out))
  File "/usr/local/lib/python3.9/dist-packages/kibot/out_base.py", line 163, in run
    self.options.run(self.expand_filename(output_dir, self.options.output))
  File "/usr/local/lib/python3.9/dist-packages/kibot/out_bom.py", line 846, in run
    comps = self.variant.filter(comps)
  File "/usr/local/lib/python3.9/dist-packages/kibot/var_kibom.py", line 93, in filter
    comps = super().filter(comps)
  File "/usr/local/lib/python3.9/dist-packages/kibot/var_base.py", line 53, in filter
    apply_exclude_filter(comps, self.exclude_filter)
  File "/usr/local/lib/python3.9/dist-packages/kibot/fil_base.py", line 157, in apply_exclude_filter
    c.included = filter.filter(c)
  File "/usr/local/lib/python3.9/dist-packages/kibot/fil_generic.py", line 174, in filter
    if self.exclude_all_hash_ref and comp.ref[0] == '#':
IndexError: string index out of range
```

Here's the CSV:

```csv
Quantity per PCB,Value,Part
1,100037,PCB
```

cdwilson, 2022-09-15:
FYI, this change fixes it for me:

```python
if self.exclude_all_hash_ref and comp.ref and comp.ref[0] == '#':
```

https://github.com/INTI-CMNB/KiBot/blob/4f54712a94e11a1a97b41dce8e4c850ee43679b5/kibot/fil_generic.py#L174

cdwilson, 2022-09-15:
Found another small thing:

When I merge a CSV BOM without any reference designators, I get the following warning:

```
WARNING:Missing `references` in aggregated file `merge_bom.csv` (kibot - out_bom.py:720)
```

This doesn't have a KiBot warning number typically associated with other KiBot warnings (used to filter out the warning).

set-soft, 2022-09-16:
Hi @cdwilson !
Both should be fixed now.

cdwilson, 2022-09-16:
Thanks, works now for me.