INTI-CMNB/KiBot #249: [HOW] how to setup render_3d correctly

computergeek1507, 2022-08-12:
**What do you want to achieve?**
When I run render_3d I get error for "Unable to expand `KICAD6_3DMODEL_DIR`" I assume its a folder structure issue or I have the working directory incorrect.

**Do you have some PCB/Schematic to use as example?**
https://github.com/computergeek1507/PB_16/

**Do you have some configuration file (.kibot.yaml) that you are using?**
You can attach it or paste the content in the following section:
```yaml
  - name: board_render
    comment: "render view"
    type: render_3d
    dir: .
```

Github action step
```yaml
name: DRC PB_16v1 Files
      run: |
        cd PB_16v1
        rm -rf PB_16v1/
        kibot -c ../.github/workflows/scripts/kibot/drc.config.kibot.yaml -e PB_16.kicad_sch -b PB_16.kicad_pcb -d PB_16v1
```

**Environment (please complete the following information):**
Where are you running KiBot:
- GitHub Action
- docker: container: setsoft/kicad_auto:dev_k6

- Using a desktop system? Which OS, distro and version?
Windows 10

Link to github action if it helps
https://github.com/computergeek1507/PB_16/actions/workflows/drc.yaml

[image]

set-soft, 2022-08-12:
Hi @computergeek1507 !
The docker images doesn't contain the 3D models, they are over than 5 GiB of data, much bigger that the OS, KiCad, KiBot and all the tools (aroung 1.6 GiB).

So KiBot is unable to detect where they are ;-)

If you run KiBot on a system with the 3D models installed you won't get these errors.

But the errors aren't really important in your case, if you enable debug information (i.e. using -vvv) you'll see things like:

```
ERROR:Unable to expand `KICAD6_3DMODEL_DIR` in `${KICAD6_3DMODEL_DIR}/Button_Switch_THT.3dshapes/SW_CuK_JS202011CQN_DPDT_Straight.wrl` (kibot - config.py:71)
WARNING:(W047) Missing 3D model for SW1: `/home/salvador/0Data/Eccosur/0Test/PB_16-master/PB_16v1/${KICAD6_3DMODEL_DIR}/Button_Switch_THT.3dshapes/SW_CuK_JS202011CQN_DPDT_Straight.wrl` (kibot - out_base_3d.py:131)
DEBUG:Downloading `https://gitlab.com/kicad/libraries/kicad-packages3D/-/raw/master/Button_Switch_THT.3dshapes/SW_CuK_JS202011CQN_DPDT_Straight.wrl` (kibot - out_base_3d.py:38)
DEBUG:Downloading `https://gitlab.com/kicad/libraries/kicad-packages3D/-/raw/master/Button_Switch_THT.3dshapes/SW_CuK_JS202011CQN_DPDT_Straight.step` (kibot - out_base_3d.py:38)
```

So KiBot is downloading them on demand ;-)

In fact I got the same render using the KiCad in my system (with 3D models installed) and using the docker image:

[image]

BTW, I suggest enabling ray tracing:

[image]

I also suggest including the 3D models in your repo and use them from the repo. This will avoid problems if KiCad people changes their names, or removes some of them.

closed 2022-08-12 by set-soft

computergeek1507, 2022-08-12:
thanks, how do I enable ray tracing?

set-soft, 2022-08-16:
Hi @computergeek1507 !

To enable ray tracing add the `ray_tracing: true` option to the `render_3d` options. Here is an example:

```yaml
- name: basic_render_3d_30deg
  comment: 3D view from 30 degrees
  type: render_3d
  dir: 3D
  output_id: 30deg
  options:
    ray_tracing: true
    rotate_x: 3
    rotate_z: -2
```