INTI-CMNB/KiBot #250: [BUG] drc_exclusions_workaround does not seem to be working

cdwilson, 2022-08-13:
**Describe the bug**
I'm trying to use the new `drc_exclusions_workaround` (see https://github.com/INTI-CMNB/KiAuto/issues/26), but I'm still getting errors when running KiBot.  I'm not sure if there is a bug, or if I've got something configured incorrectly.

In my KiCad PCB file, I have excluded the DRC errors via the DRC GUI:

[image]

In my KiBot YAML, I have the following setting:

```yaml
global:
  drc_exclusions_workaround: true
```

When I run KiBot, I get the following errors:

```
- Running the DRC
ERROR:Found 6 DRC errors and 7 unconnected pad/s or warnings (pcbnew_do - pcbnew_do:1191) (kibot - kiplot.py:123)
ERROR:(copper_edge_clearance) Board edge clearance violation (board setup constraints edge clearance 0.3048 mm; actual 0.1263 mm)
    @(144.0000 mm, 98.0000 mm): Arc on Edge.Cuts
    @(145.9500 mm, 99.1300 mm): Through hole pad MH [GND] of J1 (pcbnew_do.kiauto.file_util - file_util.py:140) (kibot - kiplot.py:123)
ERROR:(copper_edge_clearance) Board edge clearance violation (board setup constraints edge clearance 0.3048 mm; actual 0.1225 mm)
    @(144.0000 mm, 98.0000 mm): Arc on Edge.Cuts
    @(144.0000 mm, 100.2500 mm): Through hole pad MH [GND] of J1 (pcbnew_do.kiauto.file_util - file_util.py:140) (kibot - kiplot.py:123)
ERROR:(copper_edge_clearance) Board edge clearance violation (board setup constraints edge clearance 0.3048 mm; actual 0.1263 mm)
    @(144.0000 mm, 98.0000 mm): Arc on Edge.Cuts
    @(142.0500 mm, 99.1300 mm): Through hole pad MH [GND] of J1 (pcbnew_do.kiauto.file_util - file_util.py:140) (kibot - kiplot.py:123)
ERROR:(copper_edge_clearance) Board edge clearance violation (board setup constraints edge clearance 0.3048 mm; actual 0.0000 mm)
    @(129.0000 mm, 98.0000 mm): Line on Edge.Cuts
    @(144.0000 mm, 99.0000 mm): Pad MH [GND] of J1 on B.Cu (pcbnew_do.kiauto.file_util - file_util.py:140) (kibot - kiplot.py:123)
ERROR:(copper_edge_clearance) Board edge clearance violation (board setup constraints edge clearance 0.3048 mm; actual 0.0000 mm)
    @(129.0000 mm, 98.0000 mm): Line on Edge.Cuts
    @(144.0000 mm, 98.0000 mm): Through hole pad MH [GND] of J1 (pcbnew_do.kiauto.file_util - file_util.py:140) (kibot - kiplot.py:123)
ERROR:(copper_edge_clearance) Board edge clearance violation (board setup constraints edge clearance 0.3048 mm; actual 0.0000 mm)
    @(129.0000 mm, 98.0000 mm): Line on Edge.Cuts
    @(144.0000 mm, 98.5000 mm): Pad MH [GND] of J1 on B.Cu (pcbnew_do.kiauto.file_util - file_util.py:140) (kibot - kiplot.py:123)
WARNING:(W058) (missing_courtyard) Footprint has no courtyard defined
    @(174.0000 mm, 92.0000 mm): Footprint ~ (pcbnew_do.kiauto.file_util - file_util.py:149) (kibot - kiplot.py:126)
WARNING:(W058) (missing_courtyard) Footprint has no courtyard defined
    @(167.0000 mm, 26.0000 mm): Footprint ~ (pcbnew_do.kiauto.file_util - file_util.py:149) (kibot - kiplot.py:126)
WARNING:(W058) (missing_courtyard) Footprint has no courtyard defined
    @(83.7000 mm, 147.6000 mm): Footprint ~ (pcbnew_do.kiauto.file_util - file_util.py:149) (kibot - kiplot.py:126)
WARNING:(W058) (missing_courtyard) Footprint has no courtyard defined
    @(135.9000 mm, 107.6000 mm): Footprint ~ (pcbnew_do.kiauto.file_util - file_util.py:149) (kibot - kiplot.py:126)
WARNING:(W058) (missing_courtyard) Footprint has no courtyard defined
    @(140.0000 mm, 110.3000 mm): Footprint ~ (pcbnew_do.kiauto.file_util - file_util.py:149) (kibot - kiplot.py:126)
WARNING:(W058) (missing_courtyard) Footprint has no courtyard defined
    @(132.2500 mm, 107.7500 mm): Footprint ~ (pcbnew_do.kiauto.file_util - file_util.py:149) (kibot - kiplot.py:126)
WARNING:(W058) (missing_courtyard) Footprint has no courtyard defined
    @(149.0000 mm, 104.0000 mm): Footprint ~ (pcbnew_do.kiauto.file_util - file_util.py:149) (kibot - kiplot.py:126)
ERROR:DRC errors: 13 (kibot.pre_run_drc - pre_run_drc.py:69)
```

**Expected behavior**
My assumption is that these errors should be ignored by KiBot because they are excluded in the KiCad project file and the `drc_exclusions_workaround: true` is enabled.

**Environment (please complete the following information):**
Where are you running KiBot:
- Using a docker image? `setsoft/kicad_auto:dev_k6`

set-soft, 2022-08-13:
Hi @cdwilson !
What do you get in the DRC report file? What does KiBot say when using `-vvv`?
The exclusions are working for the following regression test: `test_drc_filter_2`.
You must check that KiAuto is actually using the GUI method. If KiAuto is using the GUI mechanism and the DRC report contains errors check that the project is there and that it contains the exclusions (`board.design_settings.drc_exclusions` JSON value)

cdwilson, 2022-08-13:
Hi @set-soft 

> What do you get in the DRC report file?

```
** Drc report for /home/root/workdir/ModST_Mouse.kicad_pcb **
** Created on 2022-08-13 22:15:12 **

** Found 13 DRC violations **
[missing_courtyard]: Footprint has no courtyard defined
    Local override; Severity: warning
    @(174.0000 mm, 92.0000 mm): Footprint ~
[missing_courtyard]: Footprint has no courtyard defined
    Local override; Severity: warning
    @(167.0000 mm, 26.0000 mm): Footprint ~
[missing_courtyard]: Footprint has no courtyard defined
    Local override; Severity: warning
    @(83.7000 mm, 147.6000 mm): Footprint ~
[missing_courtyard]: Footprint has no courtyard defined
    Local override; Severity: warning
    @(135.9000 mm, 107.6000 mm): Footprint ~
[missing_courtyard]: Footprint has no courtyard defined
    Local override; Severity: warning
    @(140.0000 mm, 110.3000 mm): Footprint ~
[missing_courtyard]: Footprint has no courtyard defined
    Local override; Severity: warning
    @(132.2500 mm, 107.7500 mm): Footprint ~
[missing_courtyard]: Footprint has no courtyard defined
    Local override; Severity: warning
    @(149.0000 mm, 104.0000 mm): Footprint ~
[copper_edge_clearance]: Board edge clearance violation (board setup constraints edge clearance 0.3048 mm; actual 0.1263 mm)
    Rule: board setup constraints edge; Severity: error
    @(144.0000 mm, 98.0000 mm): Arc on Edge.Cuts
    @(145.9500 mm, 99.1300 mm): Through hole pad MH [GND] of J1
[copper_edge_clearance]: Board edge clearance violation (board setup constraints edge clearance 0.3048 mm; actual 0.1225 mm)
    Rule: board setup constraints edge; Severity: error
    @(144.0000 mm, 98.0000 mm): Arc on Edge.Cuts
    @(144.0000 mm, 100.2500 mm): Through hole pad MH [GND] of J1
[copper_edge_clearance]: Board edge clearance violation (board setup constraints edge clearance 0.3048 mm; actual 0.1263 mm)
    Rule: board setup constraints edge; Severity: error
    @(144.0000 mm, 98.0000 mm): Arc on Edge.Cuts
    @(142.0500 mm, 99.1300 mm): Through hole pad MH [GND] of J1
[copper_edge_clearance]: Board edge clearance violation (board setup constraints edge clearance 0.3048 mm; actual 0.0000 mm)
    Rule: board setup constraints edge; Severity: error
    @(129.0000 mm, 98.0000 mm): Line on Edge.Cuts
    @(144.0000 mm, 99.0000 mm): Pad MH [GND] of J1 on B.Cu
[copper_edge_clearance]: Board edge clearance violation (board setup constraints edge clearance 0.3048 mm; actual 0.0000 mm)
    Rule: board setup constraints edge; Severity: error
    @(129.0000 mm, 98.0000 mm): Line on Edge.Cuts
    @(144.0000 mm, 98.0000 mm): Through hole pad MH [GND] of J1
[copper_edge_clearance]: Board edge clearance violation (board setup constraints edge clearance 0.3048 mm; actual 0.0000 mm)
    Rule: board setup constraints edge; Severity: error
    @(129.0000 mm, 98.0000 mm): Line on Edge.Cuts
    @(144.0000 mm, 98.5000 mm): Pad MH [GND] of J1 on B.Cu

** Found 0 unconnected pads **

** Found 0 Footprint errors **

** End of Report **
```

> What does KiBot say when using -vvv?

There is a ton of output (especially from the interposer), but here is the snippet from the DRC (let me know if there is something specific I should look for):

```
Running the DRC
DEBUG:Executing: /usr/bin/pcbnew_do -r -vvv run_drc -o /home/root/workdir/Release/ModST_Mouse-drc_FCD-000035.txt -f /home/root/workdir/Release/kibot_errors.filter -F /home/root/workdir/ModST_Mouse.kicad_pcb /home/root/workdir/Release (kibot - kiplot.py:142)
DEBUG:Command line: /usr/bin/pcbnew_do -r -vvv run_drc -o /home/root/workdir/Release/ModST_Mouse-drc_FCD-000035.txt -f /home/root/workdir/Release/kibot_errors.filter -F /home/root/workdir/ModST_Mouse.kicad_pcb /home/root/workdir/Release (kibot - kiplot.py:144)
[... 33 lines cut ...]
```

Here are the `drc_exclusions` from the project file:

```
      "drc_exclusions": [
        "copper_edge_clearance|142250000|98000000|00000000-0000-0000-0000-000060704a77|706694d8-88da-4aaf-96ae-5dc2fbf92bac",
        "copper_edge_clearance|142250000|98000000|00000000-0000-0000-0000-000060704a77|75ac3a70-1a3f-4a7f-af09-8eb124b0f528",
        "copper_edge_clearance|142250000|98000000|00000000-0000-0000-0000-000060704a77|f865b411-58fc-4062-a5c6-ee2b2cee7288",
        "copper_edge_clearance|142482331|98876227|00000000-0000-0000-0000-000060704a95|2b61d995-3b8c-49bd-9cae-b87331cb5a11",
        "copper_edge_clearance|144000000|99752454|00000000-0000-0000-0000-000060704a95|241dd19a-a7eb-4678-b519-408a54c6689e",
        "copper_edge_clearance|145517669|98876227|00000000-0000-0000-0000-000060704a95|0745e23b-9e00-4b57-b62c-8dfd0b6bf23c",
        "missing_courtyard|132250000|107750000|00000000-0000-0000-0000-000060b4fb10|00000000-0000-0000-0000-000000000000",
        "missing_courtyard|135900000|107600000|00000000-0000-0000-0000-0000608ddf63|00000000-0000-0000-0000-000000000000",
        "missing_courtyard|140000000|110300000|00000000-0000-0000-0000-0000609ea707|00000000-0000-0000-0000-000000000000",
        "missing_courtyard|149000000|104000000|00000000-0000-0000-0000-000060b692c4|00000000-0000-0000-0000-000000000000",
        "missing_courtyard|167000000|26000000|00000000-0000-0000-0000-000060aba781|00000000-0000-0000-0000-000000000000",
        "missing_courtyard|174000000|92000000|00000000-0000-0000-0000-000060ab9039|00000000-0000-0000-0000-000000000000",
        "missing_courtyard|83700000|147600000|00000000-0000-0000-0000-000060abb3f1|00000000-0000-0000-0000-000000000000"
      ],
```

> You must check that KiAuto is actually using the GUI method.

How do I confirm this?  Is there a setting I need to enable in the YAML file? (besides `drc_exclusions_workaround: true`)

set-soft, 2022-08-16:
> Hi @set-soft
> 
> > What do you get in the DRC report file?
> 
> ```
> ** Drc report for /home/root/workdir/ModST_Mouse.kicad_pcb **
> ** Created on 2022-08-13 22:15:12 **
> 
> ** Found 13 DRC violations **
> [missing_courtyard]: Footprint has no courtyard defined
>     Local override; Severity: warning
>     @(174.0000 mm, 92.0000 mm): Footprint ~
> ** End of Report **
> ```

They aren't excluded in the report.

> 
> > What does KiBot say when using -vvv?
> 
> There is a ton of output (especially from the interposer), but here is the snippet from the DRC (let me know if there is something specific I should look for):

Ok, if you get a lot of interposer stuff it means KiAuto is using KiCad GUI. 

> Here are the `drc_exclusions` from the project file:

Correct these are the `drc_exclusions`, they are in the project.

> > You must check that KiAuto is actually using the GUI method.
> 
> How do I confirm this?

The interposer dialog confirms this. All these messages comes from KiCad (well ... from the interposer lib loaded by KiCad)

> Is there a setting I need to enable in the YAML file? (besides `drc_exclusions_workaround: true`)

Nope, it looks like you are doing the right thing. KiBot is asking KiAuto to use KiCad (not the Python API) to run the DRC, and your project contains the exclusions. But, as you can see, KiCad still reports the problems as warnings/errors in the DRC report (created by KiCad).

Options: 
1) Insist trying to exclude the errors, if you run the DRC again: they are actually excluded? When I created the test I had to exclude things more than once.
2) Use KiBot filters, if they don't work I can fix them, but I can't fix KiCad ;-)

cdwilson, 2022-08-17:
Hi @set-soft 

> They aren't excluded in the report.

You're right, good catch, I missed this.  

Interestingly, when I generate the report manually in the GUI using the "Save" button, the DRC report shows the exclusions as `(excluded)`:

[image]

```
[missing_courtyard]: Footprint has no courtyard defined
    Local override; Severity: warning (excluded)
    @(149.0000 mm, 104.0000 mm): Footprint ~
[copper_edge_clearance]: Board edge clearance violation (board setup constraints edge clearance 0.3048 mm; actual 0.1263 mm)
    Rule: board setup constraints edge; Severity: error (excluded)
    @(144.0000 mm, 98.0000 mm): Arc on Edge.Cuts
    @(145.9500 mm, 99.1300 mm): Through hole pad MH [GND] of J1
```

> KiBot is asking KiAuto to use KiCad (not the Python API) to run the DRC

If this is using KiCad GUI (instead of the Python API), I'm not sure what could be the difference between running the DRC locally on my Mac vs. in the KiBot docker image. It should be the same right? The only thing I can think of is that I'm using KiCad 6.0.7 on Mac, but the KiBot image is 6.0.6.

> if you run the DRC again: they are actually excluded? When I created the test I had to exclude things more than once.

Yes, the DRC errors that I excluded in the KiCad GUI always stay excluded, even when I run DRC multiple times.  However, while testing this, I found a related issue (that I think will be fixed in KiCad 6.0.8), where I was unable to exclude a specific `Error: Unresolved text variable` violation coming from the worksheet (they kept reappearing whenever I ran DRC again): https://gitlab.com/kicad/code/kicad/-/issues/12221  However, even if I switch to the default worksheet (which makes these DRC errors go away), I'm still seeing the same issue described above with the other error types not being excluded in the KiBot DRC.

> Use KiBot filters, if they don't work I can fix them, but I can't fix KiCad ;-)

Yup, I've switched back to just using Kibot filters.  They work correctly without any issues!

set-soft, 2022-08-17:
Hi @cdwilson !

> If this is using KiCad GUI (instead of the Python API), I'm not sure what could be the difference between running the DRC locally on my Mac vs. in the KiBot docker image. It should be the same right? The only thing I can think of is that I'm using KiCad 6.0.7 on Mac, but the KiBot image is 6.0.6.

The different version could be related.

> > Use KiBot filters, if they don't work I can fix them, but I can't fix KiCad ;-)
> 
> Yup, I've switched back to just using Kibot filters. They work correctly without any issues!

Ok, I'll try to generate a KiCad 6.0.7 image (now is available in Debian stable), so we can see if this is the problem.

set-soft, 2022-08-17:
Hi @cdwilson !
Now `setsoft/kicad_auto:dev_k6` uses KiCad 6.0.7, can you test using it?

cdwilson, 2022-08-17:
Hi @set-soft thanks for updating the image!  

I'm still seeing the error, but I'm curious if it's something wrong with my specific setup.  Can you try a quick test for me?

I created an simple project from one of the examples shipped with KiCad.  There are two branches:
- [main](https://github.com/cdwilson/test-kibot/tree/main) ([f007466](https://github.com/cdwilson/test-kibot/commit/f007466fc55a9034d78e284fb4135f052f400d48)): Uses KiBom filters to exclude `silk_over_copper` DRC error
- [test-drc-exclusions-workaround](https://github.com/cdwilson/test-kibot/tree/test-drc-exclusions-workaround) ([14f0d45](https://github.com/cdwilson/test-kibot/commit/14f0d450a0f63c4e006591e0160115786313c085)): Uses KiCad exclusions added manually via the KiCad DRC GUI 

You should just be able to clone the repo, check out each branch, and run `make` to run kibot in the latest docker image:
```
git clone git@github.com:cdwilson/test-kibot.git
cd test-kibot/
make
<kibot will run in docker>
git checkout test-drc-exclusions-workaround
make clean
make
<kibot will run in docker>
```

When I run the `main` branch (using KiBom filters), I don't get any DRC errors:
```
❯ make
docker pull "setsoft/kicad_auto:dev_k6"
dev_k6: Pulling from setsoft/kicad_auto
Digest: sha256:c8943dbfab3a0b1a2aaf44d9f27bee70dfd790cdc472008f676d13505f56aae7
Status: Image is up to date for setsoft/kicad_auto:dev_k6
docker.io/setsoft/kicad_auto:dev_k6
docker run --rm -it --name="KiBot" --env NO_AT_BRIDGE=1 --env INTERACTIVE_HTML_BOM_NO_DISPLAY=True --volume="/Users/chris/Downloads/test-kibot:/home/root/workdir" --workdir="/home/root/workdir" "setsoft/kicad_auto:dev_k6" /bin/bash -c "kibot -e pic_programmer.kicad_sch -b pic_programmer.kicad_pcb -c release.kibot.yaml -d _build; sync"
WARNING: The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm64/v8) and no specific platform was requested
- Running the DRC
WARNING:(W058) Missing KiCad main config file /root/.config/kicad/6.0/kicad_common.json (pcbnew_do - misc.py:181) (kibot - kiplot.py:126)
- Running the ERC
- Updating BoM in XML format
- 'Export the BOM to CSV' (bom_csv) [bom]
WARNING:(W009) KiCad config without environment.vars section (kibot - config.py:293)
WARNING:(W010) Unable to find KiCad 3D models (kibot - config.py:367)
WARNING:(W020) Malformed value: `22uF/25V` (no match in C3) (kibot - units.py:144)
Found 4 unique warning/s (6 total, 2 filtered)
Restoring pic_programmer.kicad_pro from Git
Updated 0 paths from the index
```

However, when I run the `test-drc-exclusions-workaround` branch I get DRC errors:
```
❯ make
docker pull "setsoft/kicad_auto:dev_k6"
dev_k6: Pulling from setsoft/kicad_auto
Digest: sha256:c8943dbfab3a0b1a2aaf44d9f27bee70dfd790cdc472008f676d13505f56aae7
Status: Image is up to date for setsoft/kicad_auto:dev_k6
docker.io/setsoft/kicad_auto:dev_k6
docker run --rm -it --name="KiBot" --env NO_AT_BRIDGE=1 --env INTERACTIVE_HTML_BOM_NO_DISPLAY=True --volume="/Users/chris/Downloads/test-kibot:/home/root/workdir" --workdir="/home/root/workdir" "setsoft/kicad_auto:dev_k6" /bin/bash -c "kibot -e pic_programmer.kicad_sch -b pic_programmer.kicad_pcb -c release.kibot.yaml -d _build; sync"
WARNING: The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm64/v8) and no specific platform was requested
- Running the DRC
WARNING:(W058) Missing KiCad main config file /root/.config/kicad/6.0/kicad_common.json (pcbnew_do - misc.py:181) (kibot - kiplot.py:126)
ERROR:Found 2 DRC errors and 0 unconnected pad/s or warnings (pcbnew_do - pcbnew_do:1191) (kibot - kiplot.py:123)
ERROR:(silk_over_copper) Silkscreen clipped by solder mask
    @(73.6600 mm, 40.6400 mm): Line on Edge.Cuts
    @(171.4500 mm, 36.1950 mm): Line on F.Silkscreen (pcbnew_do.kiauto.file_util - file_util.py:140) (kibot - kiplot.py:123)
ERROR:(silk_over_copper) Silkscreen clipped by solder mask
    @(73.6600 mm, 40.6400 mm): Line on Edge.Cuts
    @(172.7200 mm, 36.1950 mm): Line on F.Silkscreen (pcbnew_do.kiauto.file_util - file_util.py:140) (kibot - kiplot.py:123)
ERROR:DRC errors: 2 (kibot.pre_run_drc - pre_run_drc.py:69)
Restoring pic_programmer.kicad_pro from Git
Updated 1 path from the index
```

Do you get the same errors?

set-soft, 2022-08-17:
Hi @cdwilson !
I'm starting to try it, but in a first attempt I'm getting the same, and looking at KiCad's UI I got:

[image]

This is what KiAuto gets, I'll try to find more.

cdwilson, 2022-08-17:
Which branch do you get that image from?

`main` should match the image you posted if you checkout the branch and run DRC (no KiCad exclusions from GUI):

[image]

The `test-drc-exclusions-workaround` contains the exclusions in the kicad project file, so when you checkout the branch and run DRC, you should get:

[image]

set-soft, 2022-08-17:
Hi @cdwilson !
This is what KiAuto sees when using the exclusions branch.
And I think I found why, the problem is with `fill_zones: true` again because of KiCad's Python API *limitations* ... I think the problem is that zones gets filled and the PCB saved, **but** KiCad is failing to keep the exclusions in the project, they aren't loaded (and this is why the Python API doesn't currently support the exclusions), and KiCad saves the project (without any need because we are asking to save the **PCB**, not the project) and then the exclusions are just removed. I'll add a workaround for this (back-up the project and restore it).

cdwilson, 2022-08-17:
I think this should also address the issue we discussed in https://github.com/INTI-CMNB/KiBot/discussions/230, right? (i.e. I won't need to [manually restore the project file](https://github.com/cdwilson/test-kibot/blob/f007466fc55a9034d78e284fb4135f052f400d48/Makefile#L31-L34) any longer)

set-soft, 2022-08-17:
> I think this should also address the issue we discussed in #230, right? (i.e. I won't need to [manually restore the project file](https://github.com/cdwilson/test-kibot/blob/f007466fc55a9034d78e284fb4135f052f400d48/Makefile#L31-L34) any longer)

Most probably, but I'm not 100% sure this is the only place where the project can be altered. 
The patched version should be available in a few minutes (unless it doesn't pass all the tests), so you can check if it solves the problem you mention in #230.

closed 2022-08-17 by set-soft

cdwilson, 2022-08-17:
Thx, I'll test it out once it finishes building.

cdwilson, 2022-08-17:
@set-soft looks like the latest image fixes the issue, thanks for the quick fix!

cdwilson, 2022-08-17:
>> (i.e. I won't need to [manually restore the project file](https://github.com/cdwilson/test-kibot/blob/f007466fc55a9034d78e284fb4135f052f400d48/Makefile#L31-L34) any longer)
>
> Most probably, but I'm not 100% sure this is the only place where the project can be altered.

@set-soft Just FYI, I just realized that after KiBot is run, the `text_variables` in the KiCad project is populated/modified.  So for now, I still need to restore the project file from git after KiBot is run.

set-soft, 2022-08-19:
Hi @cdwilson !
Now you have an option to restore it ;-)

cdwilson, 2022-08-19:
Awesome, thanks @set-soft!

commit 0180136f (2022-08-17): [DRC] Another workaround for KiCad lack of exclusions support in Python

- Saving the project can remove them, so `fill_zones` removed them

Fixes #250