INTI-CMNB/KiBot #255: [BUG] pre_run_erc error

KenwoodFox, 2022-08-16:
It looks like this popped up recently, i noticed all my ERCs began failing around the same time among a few projects, the error i get is this:

```
kibot -e Segment_Display_Board/Segment_Display_Board.kicad_sch -c config.kibot.yaml -d _output -b Segment_Display_Board/Segment_Display_Board.kicad_pcb
- Running the ERC
ERROR:ERC returned 1 (kibot.pre_run_erc - pre_run_erc.py:72)
```

And it seems to be the same error on github as well

```
...
*****************************************************************************************
*****************************************************************************************

KiBot GitHub Action v2

KiBot 1.2.0 - Copyright 2018-2022, Salvador E. Tropea/INTI/John Beard - License: GPL v3+
ii  kicad          6.0.6+dfsg-1~bpo11+1 amd64        Electronic schematic and PCB design software
Debian: 11.4
pcbnew_do 2.0.0 - Copyright 2018-2022, INTI/Productize SPRL - License: Apache
2.0
KiCost v1.1.10
PcbDraw 0.9.0-3
iBoM: v2.5.0

*****************************************************************************************
*****************************************************************************************
Options: -c 'Hardware/config.kibot.yaml' -d '_hardwareOutput' -b 'Hardware/Segment_Display_Board/Segment_Display_Board.kicad_pcb' -e 'Hardware/Segment_Display_Board/Segment_Display_Board.kicad_sch'
- Running the ERC
ERROR:ERC returned 1 (kibot.pre_erc - pre_erc.py:67)
```

Does anybody else notice this or is it just me?

set-soft, 2022-08-17:
Hi @KenwoodFox !
Please try using `-vvv` option to collect the debug information.
Error 1 is usually an internal error, so this could be a bug related to some combination of parameters not covered by the regression tests.

KenwoodFox, 2022-08-17:
Heres the output i collected! Thanks for responding so quick,

```
...
DEBUG: - Dep after applying from KiAuto: {'name': 'KiCad Automation tools', 'github': 'INTI-CMNB/KiAuto', 'command': 'pcbnew_do', 'pypi': 'kiauto', 'downloader': 'pytool', 'id': 'KiAuto', 'from': 'KiAuto', 'role': 'mandatory', 'version': '1.6.7'} (kibot - dep_downloader.py:876)
DEBUG:- Importing out_svg_sch_print (kibot - kiplot.py:85)
DEBUG:- Processing dependencies for `out_svg_sch_print` (kibot - dep_downloader.py:928)
DEBUG: - Data: {'Dependencies': [{'from': 'KiAuto', 'role': 'mandatory', 'command': 'eeschema_do', 'version': '2.0.0'}]} (kibot - dep_downloader.py:929)
DEBUG: - Registering dep {'from': 'KiAuto', 'role': 'mandatory', 'command': 'eeschema_do', 'version': '2.0.0'} (kibot - dep_downloader.py:940)
DEBUG: - Dep after applying from KiAuto: {'name': 'KiCad Automation tools', 'github': 'INTI-CMNB/KiAuto', 'command': 'eeschema_do', 'pypi': 'kiauto', 'downloader': 'pytool', 'id': 'KiAuto', 'from': 'KiAuto', 'role': 'mandatory', 'version': '2.0.0'} (kibot - dep_downloader.py:876)
DEBUG:- Importing pre_annotate_pcb (kibot - kiplot.py:85)
DEBUG:- Importing pre_annotate_power (kibot - kiplot.py:85)
DEBUG:- Importing pre_any_replace (kibot - kiplot.py:85)
DEBUG:- Importing pre_base (kibot - kiplot.py:85)
DEBUG:- Importing pre_check_zone_fills (kibot - kiplot.py:85)
DEBUG:- Importing pre_erc_warnings (kibot - kiplot.py:85)
DEBUG:- Importing pre_fill_zones (kibot - kiplot.py:85)
DEBUG:- Importing pre_filters (kibot - kiplot.py:85)
DEBUG:- Importing pre_ignore_unconnected (kibot - kiplot.py:85)
DEBUG:- Importing pre_pcb_replace (kibot - kiplot.py:85)
DEBUG:- Processing dependencies for `pre_pcb_replace` (kibot - dep_downloader.py:928)
DEBUG: - Data: {'Dependencies': [{'from': 'Git', 'role': 'Find commit hash and/or date'}]} (kibot - dep_downloader.py:929)
DEBUG: - Registering dep {'from': 'Git', 'role': 'Find commit hash and/or date'} (kibot - dep_downloader.py:940)
DEBUG: - Dep after applying from Git: {'name': 'Git', 'url': 'https://git-scm.com/', 'downloader': 'git', 'debian': 'git', 'from': 'Git', 'role': 'Find commit hash and/or date'} (kibot - dep_downloader.py:876)
DEBUG:- Importing pre_run_drc (kibot - kiplot.py:85)
DEBUG:- Processing dependencies for `pre_run_drc` (kibot - dep_downloader.py:928)
DEBUG: - Data: {'Dependencies': [{'from': 'KiAuto', 'role': 'mandatory', 'version': '2.0.0'}]} (kibot - dep_downloader.py:929)
DEBUG: - Registering dep {'from': 'KiAuto', 'role': 'mandatory', 'version': '2.0.0'} (kibot - dep_downloader.py:940)
DEBUG: - Dep after applying from KiAuto: {'name': 'KiCad Automation tools', 'github': 'INTI-CMNB/KiAuto', 'command': 'pcbnew_do', 'pypi': 'kiauto', 'downloader': 'pytool', 'id': 'KiAuto', 'from': 'KiAuto', 'role': 'mandatory', 'version': '2.0.0'} (kibot - dep_downloader.py:876)
DEBUG:- Importing pre_run_erc (kibot - kiplot.py:85)
DEBUG:- Processing dependencies for `pre_run_erc` (kibot - dep_downloader.py:928)
DEBUG: - Data: {'Dependencies': [{'from': 'KiAuto', 'role': 'mandatory', 'command': 'eeschema_do', 'version': '1.5.4'}]} (kibot - dep_downloader.py:929)
DEBUG: - Registering dep {'from': 'KiAuto', 'role': 'mandatory', 'command': 'eeschema_do', 'version': '1.5.4'} (kibot - dep_downloader.py:940)
DEBUG: - Dep after applying from KiAuto: {'name': 'KiCad Automation tools', 'github': 'INTI-CMNB/KiAuto', 'command': 'eeschema_do', 'pypi': 'kiauto', 'downloader': 'pytool', 'id': 'KiAuto', 'from': 'KiAuto', 'role': 'mandatory', 'version': '1.5.4'} (kibot - dep_downloader.py:876)
DEBUG:- Importing pre_sch_replace (kibot - kiplot.py:85)
DEBUG:- Processing dependencies for `pre_sch_replace` (kibot - dep_downloader.py:928)
DEBUG: - Data: {'Dependencies': [{'from': 'Git', 'role': 'Find commit hash and/or date'}]} (kibot - dep_downloader.py:929)
DEBUG: - Registering dep {'from': 'Git', 'role': 'Find commit hash and/or date'} (kibot - dep_downloader.py:940)
DEBUG: - Dep after applying from Git: {'name': 'Git', 'url': 'https://git-scm.com/', 'downloader': 'git', 'debian': 'git', 'from': 'Git', 'role': 'Find commit hash and/or date'} (kibot - dep_downloader.py:876)
DEBUG:- Importing pre_set_text_variables (kibot - kiplot.py:85)
DEBUG:- Processing dependencies for `pre_set_text_variables` (kibot - dep_downloader.py:928)
DEBUG: - Data: {'Dependencies': [{'from': 'Git', 'role': 'Find commit hash and/or date'}]} (kibot - dep_downloader.py:929)
DEBUG: - Registering dep {'from': 'Git', 'role': 'Find commit hash and/or date'} (kibot - dep_downloader.py:940)
DEBUG: - Dep after applying from Git: {'name': 'Git', 'url': 'https://git-scm.com/', 'downloader': 'git', 'debian': 'git', 'from': 'Git', 'role': 'Find commit hash and/or date'} (kibot - dep_downloader.py:876)
DEBUG:- Importing pre_update_qr (kibot - kiplot.py:85)
DEBUG:- Importing pre_update_xml (kibot - kiplot.py:85)
DEBUG:- Processing dependencies for `pre_update_xml` (kibot - dep_downloader.py:928)
DEBUG: - Data: {'Dependencies': [{'from': 'KiAuto', 'role': 'mandatory', 'command': 'eeschema_do', 'version': '1.5.4'}]} (kibot - dep_downloader.py:929)
DEBUG: - Registering dep {'from': 'KiAuto', 'role': 'mandatory', 'command': 'eeschema_do', 'version': '1.5.4'} (kibot - dep_downloader.py:940)
DEBUG: - Dep after applying from KiAuto: {'name': 'KiCad Automation tools', 'github': 'INTI-CMNB/KiAuto', 'command': 'eeschema_do', 'pypi': 'kiauto', 'downloader': 'pytool', 'id': 'KiAuto', 'from': 'KiAuto', 'role': 'mandatory', 'version': '1.5.4'} (kibot - dep_dow [... cut ...]
> DEBUG:Checking if /home/joe/git/kitsune_robotics/BuisnessCard/Hardware/Buisness Card/Buisness Card.kicad_pro was modified (eeschema_do.kiauto.file_util - file_util.py:333)
> DEBUG:Checking if /home/joe/git/kitsune_robotics/BuisnessCard/Hardware/Buisness Card/Buisness Card.kicad_prl was modified (eeschema_do.kiauto.file_util - file_util.py:333)
>  (kibot - kiplot.py:156)
WARNING:(W054) Time out detected, on slow machines or complex projects try: (kibot - kiplot.py:158)
WARNING:(W054) `kiauto_time_out_scale` and/or `kiauto_wait_start` global options (kibot - kiplot.py:159)
ERROR:Failed to update the BoM, error 1 (kibot.pre_update_xml - pre_update_xml.py:56)
```

set-soft, 2022-08-17:
This is really strange:

> INFO:Wait for BoM file creation
> DEBUG:Waiting for `['IO:open:/home/joe/git/kitsune_robotics/BuisnessCard/Hardware/Buisness Card/Buisness Card.xml']` starts=False times=1 (eeschema_do - [...]

The code is waiting for the open, but you only get the close!
Try running `pcbnew_do --info` and check that the copy displayed there is the last code from git. Also look for mixed stuff, like a main script from some part of the system using the module from another.

The log is quite strange, it also tries to wrap fclose three times! For a regular log, generated using the docker images take a look at https://github.com/INTI-CMNB/KiAuto/actions/runs/2843441743 download https://github.com/INTI-CMNB/KiAuto/suites/7780060376/artifacts/327474869 and see the file *ERC_Ok/error.txt* 
You'll see only one message saying we are wrapping fclose and you'll find the open and close calls together:

```
DEBUG:>>Interposer<<:IO:open:/__w/KiAuto/KiAuto/output/ERC_Ok/good-project.erc (@2826.846 D 0.321) (eeschema_do - interposer.py:150)
DEBUG:>>Interposer<<:IO:close:/__w/KiAuto/KiAuto/output/ERC_Ok/good-project.erc (@2826.855 D 0.01) (eeschema_do - interposer.py:150)
```

So I suspect your system has a very different libc (where something goes wrong when intercepting fclose) and something different in wxWidgets.

Do you get the same error (time out waiting for file creation) using the GitHub action? What about removing the spaces in the names?

KenwoodFox, 2022-08-18:
Here's pcbnew! 

```
❯ pcbnew_do --info
This is KiAuto v2.0.0
Installed at: /usr/bin/pcbnew_do
Using kiauto module from: /usr/lib/python3.10/site-packages/kiauto
Interpreted by Python: /usr/bin/python (v3.10.5 (main, Aug  1 2022, 07:53:20) [GCC 12.1.0])
Tools:
- kicad: /usr/bin/kicad (v(6.0.7))
- xdotool: /usr/bin/xdotool
- recordmydesktop: None
- xsltproc: /usr/bin/xsltproc
- xclip: /usr/bin/xclip
- convert: /usr/bin/convert
```

You mentioned a libc error, how would i go about maybe seeing what i have compared to you?

Ill see if i can enble verbose mode in the github action, is there a yaml keyword for that?

set-soft, 2022-08-18:
> Here's pcbnew!

This looks ok. Doble check the interposer is the last available from git. 

> You mentioned a libc error, how would i go about maybe seeing what i have compared to you?

Hmmm ... is complex. Can you try compiling the interposer? (just running `make` inside the interposer dir)
One possible problem is the use of some macro that is replacing fopen/fopen64 in your system.
We should also find why the wrapping for fclose is not going as expected.

> Ill see if i can enble verbose mode in the github action, is there a yaml keyword for that?

The key is `verbose` use a value of 3, `verbose: '3'` (use a string with a 3, not just 3, I'm not sure if the integer is the same)

set-soft, 2022-08-18:
Hi @KenwoodFox !
I uploaded a new interposer lib: https://github.com/INTI-CMNB/KiAuto/blob/master/interposer/libinterposer.so
It should print more information to help to detect what's going on.

set-soft, 2022-08-19:
Hi again @KenwoodFox !
I also added a patch that should fix the issue. But even if it works we should find the exact problem, to avoid logging such a huge ammount of data.
It could be related to the fact that Linux has fopen, fopen64, open and open64. And wxWidgets uses fopen, but on Debian this is magically converted to a call to fopen64.

set-soft, 2022-12-04:
I'll assume the KiAuto 2.0.1 solved this. If not feel free to open a new issue.

closed 2022-12-04 by set-soft