INTI-CMNB/KiBot #256: [BUG] `plot_sheet_reference` does not respect `page_layout_descr_file` project file setting

cdwilson, 2022-08-22:
**Describe the bug**
The `gerber` output option `plot_sheet_reference: true` does not respect the user's `page_layout_descr_file` project setting in the `*.kicad_pro` file.  Instead, it uses the default KiCad worksheet instead when generating the gerber file.  (I think the `gerber` output needs an option like `sheet_reference_layout` in the `pcb_print` output.)

In the following example, I have a custom worksheet defined for the project:

[image]

This is what it looks like in the PCB editor:

[image]

When I export the `User.Drawings` layer via KiBot with the option `plot_sheet_reference: true`, I get the following Gerber file:

[image]

**To Reproduce**
```
git clone --branch plot-sheet-reference git@github.com:cdwilson/test-kibot.git
cd test-kibot
make
# compare `_build/PDF/pic_programmer_assembly.pdf` vs. `_build/Gerber/pic_programmer_User_Drawings.gbr` (Gerber is using default title block)
```

**Expected behavior**
KiBot should use the worksheet defined in the KiCad project file.

**Environment (please complete the following information):**
Where are you running KiBot:
- Using a docker image? `kicad_auto:dev_k6`

**Additional context**
Add any other context about the problem here.

set-soft, 2022-08-22:
Hi @cdwilson !
This is a known limitation of KiCad. The Python API doesn't load the worksheet layout.
In KiCad 5 it was even worst because KiCad just crashed ;-) 
KiCad 6 "fixed" it loading the default one.
The documentation for this option clearly states:

             Include the frame and title block. Only available for KiCad 6 and you get a poor result
             The `pcb_print` output can do a better job for PDF, SVG, PS, EPS and PNG outputs

closed 2022-08-22 by set-soft

cdwilson, 2022-08-22:
> This is a known limitation of KiCad. The Python API doesn't load the worksheet layout.

Ahh, I didn't realize this was a KiCad limitation... thanks for explaining it.

> Include the frame and title block. Only available for KiCad 6 and you get a poor result

Just FYI, when I read the documentation, I assumed that "poor result" meant something like "low resolution" or "poor quality" (not that the drawing frame would be totally different).

> P.S. The pcb_print output is the one you want, gerbers with frames are useless.

I've started using `pcb_print` instead and it works as expected 👍

set-soft, 2022-09-14:
> Just FYI, when I read the documentation, I assumed that "poor result" meant something like "low resolution" or "poor quality" (not that the drawing frame would be totally different).

I added a clarification