INTI-CMNB/KiBot #257: [BUG] `dnf_filter` needed when variants are used

cdwilson, 2022-08-22:
When a variant is used, `dnf_filter: _kibom_dnf_BOM` must be specified in addition to `fit_field: BOM` and `ignore_dnf: true` in order for DNP to be correctly processed.  When a variant is not used, this is not needed.

Here's how to reproduce it:

```
git clone git@github.com:cdwilson/test-kibot.git
cd test-kibot/

git checkout dnf-filter-without-variant
make
open _build/BOM/bom_no_dnf_filter.html  # note that C4 is listed under "Optional components (DNF=Do Not Fit)"
open _build/BOM/bom_with_dnf_filter.html  # note that C4 is listed under "Optional components (DNF=Do Not Fit)"
open _build/BOM/kibom_no_dnf_filter.html  # note that C4 is listed under "Optional components (DNF=Do Not Fit)"
make clean

git checkout dnf-filter-with-variant
make
open _build/BOM/bom_no_dnf_filter.html  # BROKEN: C4 is NOT listed under "Optional components (DNF=Do Not Fit)"
open _build/BOM/bom_with_dnf_filter.html  # note that C4 is listed under "Optional components (DNF=Do Not Fit)"
open _build/BOM/kibom_no_dnf_filter.html  # note that C4 is listed under "Optional components (DNF=Do Not Fit)"
make clean
```

_Originally posted by @cdwilson in https://github.com/INTI-CMNB/KiBot/discussions/251#discussioncomment-3441905_

set-soft, 2022-08-22:
Hi @cdwilson !
When you use a variant the filters are from the variant.
The documentation for `fit_field` says: *Field name used for internal filters* I'm  adding  *(not for variants)* to make it more clear.
When you specify a variant you want to make it uniform for all the outputs. 
The local filters are there just to avoid needing to create a variant just because you need to apply a simple filter.
I'm adding *This option is for simple cases, consider using a full variant for complex cases*.

But not changing other stuff. If you use a `variant` all the information goes inside the variant, so you can apply it to any output that supports variants.

closed 2022-08-22 by set-soft

cdwilson, 2022-08-22:
Hi @set-soft, thanks for the clear explanation!

> The documentation for fit_field says: Field name used for internal filters [...] P.S. Note that these filters are used as default filters for the internal variant created to apply the filters, but not for any variant you specify, which must define its own filters.

This was the source of my confusion.  I didn't realize that internal filters were not used for variants I defined, and that `fit_field` was only supposed to be used when variants were not being used.

> I'm adding (not for variants) to make it more clear.

this is clear now 👍

> If you use a variant all the information goes inside the variant, so you can apply it to any output that supports variants.

I didn't realize that it was possible to define a `dnf_filter` inside the `variants` definition!  I was defining `dnf_filter` under each output separately... 🤦  Defining it in the variant makes SO much more sense!

Is there a place in the documentation where all possible fields/options for `variants` are listed? If not, I think that might help other new users know what is possible to define on the variant itself.

Thanks again, sorry for my confusion on this!