INTI-CMNB/KiBot #258: [BUG] error when using `hide_excluded: true`

cdwilson, 2022-08-22:
**Describe the bug**
When using `hide_excluded: true` option with the `pcb_print` output, I think there might be a bug if the PCB has footprints that don't have a corresponding schematic symbol linked to them.

For example, I have a PCB with some logos where the refdes of each logo is `REF**`, and there is no corresponding symbol in the schematic.  When I run KiBot on this PCB, I get the error:

```
Traceback (most recent call last):
  File "/usr/local/bin/kibot", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.9/dist-packages/kibot/__main__.py", line 342, in main
    generate_outputs(outputs, args.target, args.invert_sel, args.skip_pre, args.cli_order, args.no_priority,
  File "/usr/local/lib/python3.9/dist-packages/kibot/kiplot.py", line 441, in generate_outputs
    _generate_outputs(outputs, targets, invert, skip_pre, cli_order, no_priority, dont_stop)
  File "/usr/local/lib/python3.9/dist-packages/kibot/kiplot.py", line 432, in _generate_outputs
    run_output(out, dont_stop)
  File "/usr/local/lib/python3.9/dist-packages/kibot/kiplot.py", line 359, in run_output
    out.run(get_output_dir(out.dir, out))
  File "/usr/local/lib/python3.9/dist-packages/kibot/out_base.py", line 162, in run
    self.options.run(self.expand_filename(output_dir, self.options.output))
  File "/usr/local/lib/python3.9/dist-packages/kibot/out_pcb_print.py", line 1015, in run
    self.filter_components()
  File "/usr/local/lib/python3.9/dist-packages/kibot/out_pcb_print.py", line 329, in filter_components
    self.remove_fab(GS.board, comps_hash)
  File "/usr/local/lib/python3.9/dist-packages/kibot/out_base.py", line 387, in remove_fab
    if not c.included:
AttributeError: 'NoneType' object has no attribute 'included'
```

I think this line:

https://github.com/INTI-CMNB/KiBot/blob/396d701a0d7dcd590d7e4261b122d6b888ef3623/kibot/out_base.py#L387

might need to be something like:

```
if c and not c.included:
```

set-soft, 2022-08-23:
@cdwilson !
You are right, thanks!
BTW: we always use a logo counterpart in the schematic. Even for things like the warning about not disposing the PCB as regular waste. It helps to avoid unintentional logo removal from the PCB while doing massive components updates.

closed 2022-08-23 by set-soft

cdwilson, 2022-08-23:
> BTW: we always use a logo counterpart in the schematic. Even for things like the warning about not disposing the PCB as regular waste. It helps to avoid unintentional logo removal from the PCB while doing massive components updates.

That's a good suggestion.  I was simply locking the footprints in the PCB to prevent removal, but using a schematic symbol is probably the better practice.

commit bc001147 (2022-08-23): [Fixed][PCB_Print] Problems with `hide_excluded: true`

- For components not in the SCH
Fixes #258