INTI-CMNB/KiBot #266: [BUG] Generated kicost spreadhseet doesnt respect DNP from variants

dbriscoe511, 2022-08-29:
**Describe the bug**
parts that are DNP on the BOM because of a variant choice are not marked as DNP in kicost. this could lead to confusion when purchasing parts

**To Reproduce**
1. Create a kicad project
2. add a part with the config field set to -vartest, add an MPN to it
3. set up kibot to run with that varient, generate a BOM, and a kicost spreadsheet..
4. If kibot was successfully set up, the part will be in the DNP section of the BOM, but not of the kicost sheet.

set-soft, 2022-08-31:
Hi @dbriscoe511 !
Please submit an example. There are infinite ways to do what you describe: Is this using the `kicost` output or the `bom` output (which also generates a KiCost sheet)? Which variant style are you using? Are you using `kicost` variant mechanism or KiBot mechanism? Are you mixing both? etc.

dbriscoe511, 2022-08-31:
Sorry, I was not aware of all the diffirent ways this could be accomplished. Le me know if this is not a bug, but just requires a configuration change.
 - kibom varient style
 - kicost output
 
 ` - name: kicostgen
    comment: kicost output generator 
   type: kicost
    dir: Manufacturing/Assembly
    options:
      number: 1`

set-soft, 2022-09-01:
Hi @dbriscoe511 !
This should be supported. I need an example where it doesn't work.

set-soft, 2022-12-03:
Without an example after 3 months I'll close this issue. If somebody can provide one please open a new issue.

closed 2022-12-03 by set-soft