INTI-CMNB/KiBot #270: [FEATURE] Add DNPs to solderpaste layer in gerbers

dbriscoe511, 2022-08-31:
**Is your feature request related to a problem? Please describe.**
Some manufacturers prefer that if parts are considered DNP there is no solder paste on their pads. this saves on expensive solder past and makes it easy to populate the components later if needed. Some companies, like JLCPCB do this automatically when you upload a board for manufacturing, but not all. 

**Describe the solution you'd like**
If a part is marked DNP directly or through a varient, add an option to the gerber export to exclude solderpaste (f.paste/b.paste layers) on it's pads. 

**Describe alternatives you've considered**
Other than directly editing the gerber files or making a custom DNP footprint for each part, I'm not sure if this is possible

set-soft, 2022-09-01:
Hi @dbriscoe511 !
This should be already done, solder paste and glue are removed. Do you have an example where it doesn't work?

dbriscoe511, 2022-09-01:
Nope, this looks like it works. I didn't see it in the documentation and my manufacturer was complains, but it looks like they were wrong i guess.

closed 2022-09-01 by dbriscoe511

dommae123, 2022-09-13:
Hey @set-soft and @dbriscoe511 ,
this option seems to work. In my project, the paste in the gerber files is removed from parts, that are marked as "DNF". But I don't want that =) Is there a possibility to add solder paste to the pads of a not-fitted-part? I want to have solder paste on all my pads, regardless of wheather the parts are fitted or not.

Thanks for your answer,
Thomas

set-soft, 2022-09-13:
Hi @dommae123 !
The above patch adds the global `remove_solder_paste_for_dnp` option. Try setting it to `false`

set-soft, 2022-09-13:
Hi again @dommae123 !
You are right about the 3D render, the above patch should fix the problem.
Now we also apply the 2D processing to the PCBs used for the 3D targets.

dommae123, 2022-09-13:
Hi @set-soft,
thanks for your very fast response. All your changes are working wonderfully. The gerbers and the 3D views are perfect right now. Thanks again very very much.
Greetings from Germany,
Thomas