INTI-CMNB/KiBot #272: [KiDiff] Link from the generic name to the specific name

mdeweerd, 2022-09-01:
Currently KiBot creates a link from the specific name containing the version references to the generic name.  This has a few issues:

1. The link: in my case the link is useable from Cygwin, but not from the File Explorer (windows).  Maybe that is because I start docker from Cygwin but I would be surprised that this impact the symbolic links as other windows programs dont like them.

2. After multiple runs there would be multiple links pointing to the same file which changes, while the link name itself has the version information.  So I prefer the copy, or a fixed link from the generic name to the more specific name.  In case of the latter (link from the generic name to the specific name), `make`'s requirements on the timestamp should still be ok as the link itself is updated as well as the target of the link.

_Originally posted by @mdeweerd in https://github.com/INTI-CMNB/KiBot/discussions/265#discussioncomment-3521167_

closed 2022-09-02 by set-soft

mdeweerd, 2022-09-02:
@set-soft Thank you for adding the option to copy, however also consider making a diffent link: point to the specific name from the original name.

Otherwise a file  difference_VA_VB.pdf would point to difference.pdf on the first run where difference.pdf is for VA vs VB.
Then there could a difference_V0_VC.pdf which would point difference.pdf which is V0 vs VB, which results in difference_VA_VB.pdf  pointing to a V0 to VB comparisson.
Also, one might want to compare the curren master to multiple versions in the same run (different configs), so then this woujld also confligt.
So difference.pdf should point to difference_VA_VB.pdf  - which may require renaming/moving the original file before making the link.

set-soft, 2022-09-02:
@mdeweerd I understand what you say, but the link is just to add more information, not to create a history.
I.e. you can use it to archive the results and know which hashes were involved. 
The target is the PDF file, not a simple link.

mdeweerd, 2022-09-02:
My point is that the link can create misleading information.
Anyway, I'll activate the option to create a copy that you kindly created.

commit e9e4225b (2022-09-02): [Diff] Added option to copy instead of link

Closes #272