INTI-CMNB/KiBot #276: [FEATURE] KiDiff - Use current tag or current hash for link

mdeweerd, 2022-09-03:
The link or copied file shows "None" when there was no git reference specification for the relevant version.

To avoid changing the configuration file to compare between the release version and some previous version, I propose that KiBot checks if the the checkout corresponds to a tag, and use that tag .

Also, curerntly tha hash is shown in the "linked" difference file, even if the tag was specified in the yaml configuration.  That feature could be applied on the unspecified git branch (if it is not tagged), but I suggest that the specified tag/reference would be used for the name.

Finally, I suggest that the short git hash would always be used, but append a "M" or other symbol to it if there were local changes.  (`svn` appends an "M" to the revision number when there are local changes).

Possibly some of the above would only be applied if the 'old_type' or 'new_type' is set to 'git'.  
There is currently a side effect: when `new_type` is set, but not `new` (missing reference), then kibot tries to checkout an empty reference.  
I suggest that in case of a missing reference, kibot detects the best reference by itself.

Here is a oneline to discover applied tags:

```
 git describe --exact-match --tags $(git log -n1 --pretty='%h')
```

set-soft, 2022-09-05:
> The link or copied file shows "None" when there was no git reference specification for the relevant version.
>
> To avoid changing the configuration file to compare between the release version and some [...]

Please clarify. If the file says None is because you specified a file, not a git hash. We can't assume this file is related to any tag or hash. In fact the default is to generate a diff between the current file, which could be modified, and the HEAD. 
Are both sentences related?

> Also, curerntly tha hash is shown in the "linked" difference file, even if the tag was specified in the yaml configuration. That feature could be applied on the unspecified git branch (if it is not tagged), but I suggest that the specified tag/reference would be used for the name.

True, we can use the supplied name and the hash as second value.

> Finally, I suggest that the short git hash would always be used, but append a "M" or other symbol to it if there were local changes. (`svn` appends an "M" to the revision number when there are local changes).

"Modified files" isn't a concept handled by KiBot. Only files and git points.

> Possibly some of the above would only be applied if the 'old_type' or 'new_type' is set to 'git'. There is currently a side effect: when `new_type` is set, but not `new` (missing reference), then [...]

Not sure what "best reference" means in this context. If `new_type` is `file` then KiBot uses the current PCB/SCH. But if `new_type` is `git` ... what's the best? 

> Here is a oneline to discover applied tags:
> 
> ```
>  git describe --exact-match --tags $(git log -n1 --pretty='%h')
> ```

Is that for the *if the the checkout corresponds to a tag, and use that tag*?

set-soft, 2022-09-05:
> True, we can use the supplied name and the hash as second value.

Noe the test generates: `light_control-diff_pcb_KIBOT_LAST-1(b44d31d)-HEAD(9f7c17e).pdf`

mdeweerd, 2022-09-05:
Personally, if it is not a manually positionaed tag it is not really interesting, and HEAD is not either - the branch would be better.

I would avoid parentheses in a name - it can be a headache in filenames.

So if it's a "real tag", then just the tag or tag + short hash is useful, otherwise just the hash.  branch names, HEAD and KIBOT_LAST-1 are not of real use and jjust adds noise to the output filename.

set-soft, 2022-09-05:
I disagree that they are noise, they are what you selected.

mdeweerd, 2022-09-05:
> I disagree that they are noise, they are what you selected.

To me it's noise in the sense that it's a moving target, not useful to those that are going to receive the data and not very useful in a post-analysis and people might reference the version as "KIBOT_LAST-1".

mdeweerd, 2022-09-05:
> "Modified files" isn't a concept handled by KiBot. Only files and git points.
While investigating the pcb_replace option, I disovered that the "--dirty" option as in `git describe --tags --dirty` does the job.

> Please clarify. If the file says None is because you specified a file, not a git hash.
> [...] but if new_type is git ... what's the best?

I did not specify anything at that point, so I guess that from kibot's point of view it is the current file.
However, when I specify the `new` type as git, with a missing `new` value then it tries to checkout an empty reference.
I suggest that in that case, kibot just uses the current state of the repository, but considers reporting git hashes, refs, tags for the current git state (rather than None).

> Is that for the if the the checkout corresponds to a tag, and use that tag?

Yes, but it looks like `git describe --tags --dirty` is good enough as it only describes the most recent tag for the checkout if there is one, otherwise it results in 'fatal' in which case I suggest using the hash.

set-soft, 2022-09-06:
> Yes, but it looks like `git describe --tags --dirty` is good enough as it only describes the most recent tag for the checkout if there is one, otherwise it results in 'fatal' in which case I suggest using the hash.

I changed it to use: (first that works)
1. The name of the branch if we are at its head, discarded when detached
2. The name of the tag as you mention
3. The name specified by the user

The format is *SHORT_HASH(NAME)*, so the test is generating: `light_control-diff_pcb_8e9fef3(KIBOT_LAST-1)-4cfa9c0(master).pdf`
The test doesn't have any tag, I'll try to add one, so you get *KIBOT_LAST-1*, and the other point is the HEAD of *master*. The only relevant values are the hashes, the other value is just to help humans to understand what the point is.

Sorry for the parenthesis, but this is git notation. When you run `git log` you get hashes and some text in parenthesis when the hash is associated to a tag, branch, etc. So I want to keep a similar syntax, I guess that parenthesis won't be part of a tag name, as git assumes.

set-soft, 2022-09-06:
> I'll try to add one

After adding a tag I'm getting `light_control-diff_pcb_fe59cb3(v1)-dcc208c(master).pdf` which is quite similar to what `git log` says:

```
commit dcc208ce8d1231bffa19f1f135feb6b85274fe61 (HEAD -> master)
Author: KiBot test <pytest@nowherem.com>
Date:   Tue Sep 6 09:11:25 2022 -0300

    New version

commit 07eff1506cac30d5d3cbb34b2e7ef41196ff6e93
Author: KiBot test <pytest@nowherem.com>
Date:   Tue Sep 6 09:11:25 2022 -0300

    Dummy noise

commit fe59cb3f080cbccb4782e8809d431fc8d644a1f3 (tag: v1)
Author: KiBot test <pytest@nowherem.com>
Date:   Tue Sep 6 09:11:25 2022 -0300

    Reference
```

mdeweerd, 2022-09-06:
Thanks for evaluating this.

I'ld prefer the order:
I changed it to use: (first that works)
1. The name of the last tag valid on the current checkout (even HEAD)
2. The name of the branch if we are at its head, discarded when detached
3. The name specified by the user

And the short hash is always added.

Why the tag first?  When everything is commited, and a tag is applied to the current checkout, then it is logical that the kibot run can be considered a tag related runa and not a master branch run.  Also (in principle), the tag is not supposed to move, while the branch does.

set-soft, 2022-09-06:
> However, when I specify the `new` type as git, with a missing `new` value then it tries to checkout an empty reference.
> I suggest that in that case, kibot just uses the current state of the repository, but considers reporting git hashes, refs, tags for the current git state (rather than None).

I added some logic to skip the stash and checkout. It includes the same naming logic used for the regular git code. The default name is `Dirty` (when modified) or `HEAD`. The code also adds `-dirty` if we use the branch name.

set-soft, 2022-09-06:
> The command you proposed shows the tag even when it is far from the point we are evaluating. So you can't really say this is a diff against the tag.

I added some logic to give priority to tags (43aae8b0feca106caee0988bd143eb9872565d4b)
So now is:

1. The name of the tag if we are exactly sitting at the tag (-dirty if new='' and modified)
2. The name of the branch if we are at its head, discarded when detached (-dirty if new='' and modified)
3. The name of the tag as you mention (it automatically adds dirty)
4. The name specified by the user (Dirty if new='' and is modified, HEAD if new='' and not modified)

So diffs using a tag as reference will we clear

set-soft, 2022-09-06:
> Actually the response indicates the distance from the tag, if the distance is > 0 then it can be considered to be an invalid tag.

Yes, and this is why it isn't relevant. You may be 100 commits away of a tag and it will show-up anyways.

set-soft, 2022-09-06:
> So this a "valid tag":
> 
> ```
> $ git describe --tags
> DUO_V02
> ```
> 
> and this "invalid":
> 
> ```
> $ git describe --tags
> DUO_V02-1-g051d284
> ```

I'm not sure what you mean with this.

mdeweerd, 2022-09-06:
The first response from git describe indicates that this is the tag for the current checkout.

The second one shows the tag is at a distance of "1" from the current checkout.  So that is "invalid" because it can not be used as a perfect reference for the checkout.

So the first tag can be used as "The name of the tag we are exactly sitting at"

set-soft, 2022-09-06:
> The first response from git describe indicates that this is the tag for the current checkout.

Hmmm ... it says what's the closer tag.
 
> The second one shows the tag is at a distance of "1" from the current checkout. So that is "invalid" because it can not be used as a perfect reference for the checkout.

Yes.

> So the first tag can be used as "The name of the tag we are exactly sitting at"

Nope, this will say that the current branch was tagged in the past. To know if we are really at a tag we use: `git describe --exact-match --tags HEAD`.

Adding the last tag to the branch could help, but in this case we must figure out how to add it.

set-soft, 2022-09-06:
To clarify the idea:

- If we say xxxx(TAG) it means we are using the TAG point. TAGs are associated we a fixed hash.
- If we say xxxx(BRANCH) it means we are using the BRANCH branch. A BRANCH is just a point where thing diverted and it has a HEAD.

An option could be: xxxx(BRANCH/TAG)

set-soft, 2022-09-07:
Ok, here is the last change:

1. The name of the tag if we are exactly sitting at the tag (-dirty if new='' and modified)
2. The name of the branch if we are at its head, discarded when detached (-dirty if new='' and modified)
    In addition if we have a tag at this branch it's added as BRANCH[TAG]
4. The name of the tag as you mention (it automatically adds dirty)
5. The name specified by the user (Dirty if new='' and is modified, HEAD if new='' and not modified)

I think this is much more than what the original title asked (current tag or current hash). Now we always have a hash, if available we always has a tag and we know the branch.

closed 2022-09-07 by set-soft

mdeweerd, 2022-09-07:
Not tested yet, but I trust it works, the description is perfect.  More than what requested as you indicate.