INTI-CMNB/KiBot #279: [FEATURE] Copy files from the project directory to the output directory

mdeweerd, 2022-09-04:
Goal: add information to the output directories/packages to make them part of a delivery.

I did not find an option to add (copy) files from the project directory to a target directory.  For instance, I want to add some LCD svgs and private datasheets to the output directory.

Alternatives: #109

set-soft, 2022-09-05:
Currently you can do it only for the `compress` output.
Adding a `copy` output isn't bad idea, but will be very low priority.

set-soft, 2022-09-12:
Hi @mdeweerd !
Is barely tested, but should do the work.

closed 2022-09-12 by set-soft

mdeweerd, 2022-09-12:
Excellent ;-).  It will be useful and I will add it to my setup so I'll let you know!
(I'll also add the variant comparison and let you know about that).

set-soft, 2022-09-19:
@mdeweerd , I'm changing the syntax of this output to allow more uses.

mdeweerd, 2022-09-19:
Thanks for informing, I haven't updated my configuration yet to test, so I'll wait a bit.

set-soft, 2022-09-20:
@mdeweerd I finished the syntax change.
Now you can also export the 3D models, as a plus you can get a PCB modified to use the exported 3D models. The directories layout doesn't allow full customization, and some paths could fail, but I think is a good start for the feature we discussed some time ago.

mdeweerd, 2022-09-20:
I hope that you documented that use case - I did not check it yet, but I wonder if this is for a 3D of the kicad_pcb file when it is copied, or if that means that the step (can) somehow refers to external 3D models.  I remember the feature that we discussed regarding the customisation of which 3D models to use, but not one where copying the 3D model themselves is useful, but it certainly would be usefull for a proper "backup"/"archive".  Exporting the PCB itself as a component/3DModel has been handy at one point where one PCB designed in KiCAD was a board soldered on another design.

set-soft, 2022-09-21:
> I hope that you documented that use case 

Yeap, I pushed the doc changes today, after adding the ${KIPRJMOD} fix

> - I did not check it yet, but I wonder if this is for a 3D of the kicad_pcb file when it is copied, or if that means that the step (can) somehow refers to external 3D models.

This is for the PCB copy, and when you explicitly ask for it. 

> I remember the feature that we discussed regarding the customisation of which 3D models to use, but not one where copying the 3D model themselves is useful, but it certainly would be usefull for a [...]

It can also be used to make a PCB self contained. We always work in this way because we give all the design files to our clients, so we must be sure they can edit the PCB without missing libs and/or 3D models.

set-soft, 2022-09-21:
@mdeweerd, I think this option can do what you mentioned in #278 *Maybe another job for KiBot: verify that all the models are in the project - copy them in and adjust the project if not?*

commit 9a70f2e1 (2022-09-12): New output copy_files

Closes #279