INTI-CMNB/KiBot #281: [FEATURE] Set Edge.Cuts color when `force_edge_cuts: true` is set

cdwilson, 2022-09-05:
**Is your feature request related to a problem? Please describe.**
I may have missed it in the documentation, but I didn't find a way to set the Edge.Cuts color when `force_edge_cuts: true`.

[image]

**Describe the solution you'd like**
I'd like an option to set the color of the Edge.Cuts layer when `force_edge_cuts: true`.

**Describe alternatives you've considered**

A workaround is to add the Edge.Cuts layer manually to each page:

```
        - layers:
            - layer: F.Cu
              color: "#000000"
            - layer: Edge.Cuts
              color: "#000000"
```

[image]

set-soft, 2022-09-06:
Hi @cdwilson !
I added a new option, please check if it works. Is called `forced_edge_cuts_color`.
Note that when this isn't specified the color used comes from KiCad's options.

closed 2022-09-06 by set-soft

cdwilson, 2022-09-06:
Hi @set-soft!  I just tried the new option and it looks like it's working correctly.  Thanks!

commit 22d618a4 (2022-09-06): [PCB Print] Added option to configure the forced edge color.

Closes #281