INTI-CMNB/KiBot #282: [FEATURE] Add board layer control in the `3d_render` output

cdwilson, 2022-09-06:
**Is your feature request related to a problem? Please describe.**
I was trying to generate a 3D render of the bare PCB with KiBot, and I don't think there is any way to turn off all the 3D models, paste, and adhesive layers.  Here's the best I was able to achieve:

```
      no_smd: true
      no_tht: true
```

[image]

Here's an example of what I'd like to achieve:

[image]

While trying to get this working, I realized that there are additional board layers options available in the 3D viewer which don't seem to be supported by KiBot's `3d_render` output.  It would be nice to support these additional options.

[image]

set-soft, 2022-09-12:
Now it should be possible.
Try using `show_solderpaste: false`
Needs KiAuto 2.0.4

closed 2022-09-12 by set-soft

cdwilson, 2022-09-13:
Works great, thanks!

[image]

commit 70a2788d (2022-09-12): [Render 3D] Added options to control render aspects

- To disable some technical layers
- To control the silkscreen clipping

Closes #282