INTI-CMNB/KiBot #283: [BUG] `suffix` option does not work in `pcb_print` output

cdwilson, 2022-09-06:
**Describe the bug**
If the `suffix` option is specified in `pcb_print` output, it appears to be ignored (the default `assembly_page_XX` is always used no matter what).

For example, if the following output is defined,

```
outputs:
  - name: plot_png
    comment: Export PCB plots in PNG format
    type: pcb_print
    dir: PNG
    options:
      format: PNG
      pages:
        - layers:
            - layer: F.Cu
              suffix: foo
```

then the output file suffix will still be `***assembly_page_01.png`, when it should actually be `***foo.png`.

Also, I'm wondering if it makes sense to have a suffix option for each layer included in the same page.  For example, it's not clear to me what the suffix should be in the following output (should it be `***bar.png` or `***baz.png`?):

```
outputs:
  - name: plot_png
    comment: Export PCB plots in PNG format
    type: pcb_print
    dir: PNG
    options:
      format: PNG
      pages:
        - layers:
            - layer: F.Cu
              suffix: bar
            - layer: Edge.Cuts
              suffix: baz
``` 

It seems like the suffix should be an option on the page level, rather than on the layer level, since each page corresponds to a single PNG file, i.e. something like this:

```
outputs:
  - name: plot_png
    comment: Export PCB plots in PNG format
    type: pcb_print
    dir: PNG
    options:
      format: PNG
      pages:
        - layers:
            - layer: F.Cu
          suffix: foo
``` 

**To Reproduce**
```
git clone --branch suffix-not-working git@github.com:cdwilson/test-kibot.git
cd test-kibot
make
# check whether output file in _build/PNG/ is using the correct suffix
```

**Expected behavior**
The specified suffix should be substituted for `%i` in the output file name.

**Environment (please complete the following information):**
Where are you running KiBot:
- Using a docker image? `kicad_auto:dev_k6`

set-soft, 2022-09-06:
Hi @cdwilson !
The `suffix` doesn't have any effect for the `pcb_print` output. This is an attribute of a layer, and you can use a layer list definition in many places. The suffix is useful for outputs derived from KiCad plot (Gerbers, DXF, PDF, SVG, etc.).

If KiBot uses a custom layer definition here you won't be able to reuse an already defined list of layers.

The documentation clearly states: *Filename for the output (%i=assembly, %x=pdf)/(%i=assembly_page_NN, %x=svg)*
It doesn't mention anything about `suffix`, and clearly mentions `assembly_page_NN`

An option to change it could be added, but we must determine its use (to achieve what?)

cdwilson, 2022-09-06:
Hi @set-soft,

> The suffix doesn't have any effect for the pcb_print output.

Got it.  Just FYI, I was confused because `suffix` is listed as an option under the PCB print output, so I assumed it was supported.

[image]

> An option to change it could be added, but we must determine its use (to achieve what?)

I use KiBot to generate the outputs for the bare PCB and the final PCB assemblies separately (bare PCB has a different P/N than the individual PCB assemblies).  One of my outputs for the bare PCB is PNG/SVG of the individual plot layers.  When I export the plots for the bare PCB as PNG or SVG, the default suffix `%i=assembly_page_NN` doesn't make much sense for a bare PCB (there is no assembly involved yet).  I'd like to have the option to provide my own suffix value for `%i` in the output for each of the files that are generated in these outputs.  For example, I'd like to be able to specify something like `suffix: "F_Cu"` for the exported PNG file that contains both the `F.Cu + Edge.Cuts` layers.

set-soft, 2022-09-06:
> Just FYI, I was confused because `suffix` is listed as an option under the PCB print output, so I assumed it was supported.

This is just the *Layer* object, if I use a custom object for `pcb_print` then you won't be able to use YAML anchors to recycle layer lists. Code repetition and YAML limitations, not a good thing.

About the suffix problem: Currently you can just generate individual files and specify the `output` name you want.

cdwilson, 2022-09-06:
> This is just the Layer object, if I use a custom object for pcb_print then you won't be able to use YAML anchors to recycle layer lists. Code repetition and YAML limitations, not a good thing.

Totally agree, being able to reuse layer lists makes a lot of sense (I just didn't realize it when reading the docs).

In the case of the `pcb_print` output, each page has options specific to the page.  Is there any reason why it would not be possible to add a `suffix:` option at the page level to modify the output file name for each page? (I would assume that you could still use layer archors within the page to reuse layer lists).

> About the suffix problem: Currently you can just generate individual files and specify the output name you want.

I didn't think of that, I think that should be OK as a workaround 👍 (feel free to close this out if you don't want to add something like a `suffix:` option at the page level)

closed 2022-09-07 by set-soft

cdwilson, 2022-09-07:
Hi @set-soft,

Just my 2 cents, but this change feels a bit awkward to me.  As far as I can tell, this is the only place where `%x` will behave differently.

> The problem with this particular output is that `%i` changes and has an important information, the page number.

I'm not sure that the page number is always important.  For example, I was originally hoping to be able to generate outputs that look like the following:
```
<some prefix>_F_Cu.png
<some prefix>_B_Cu.png
etc...
```

To me, this feels more intuitive than:
```
<some prefix>_page_01.png
<some prefix>_page_02.png
```

cdwilson, 2022-09-07:
Hi @set-soft,

Just my 2 cents, but this change feels a bit inconsistent to me.  As far as I can tell, this is the only place where `%x` will change it's behavior.

> Currently all outputs has output_id that replaces %I and none can change the %i value.

As far as I can tell, `%I` is the same value for all files generated in a single output (not possible to change on a per-file basis).

There is already precedent to change `%i` in outputs like `gerber` when the `suffix` option is provided for a layer.  This gives the ability to generate different gerber files with a single output definition.

For example, this definition:

```
    layers:
      - layer: F.Cu
        suffix: foo
      - layer: B.Cu
        suffix: bar
    options:
      output: "<some prefix>_%i.%x"
```

can generate this output:

```
<some prefix>_foo.gbr
<some prefix>_bar.gbr
```

In the case of outputs like `gerber` there is a 1:1 mapping between a layer and an output file.  In the case of `pcb_print` output, there is a 1:1 mapping between a page and an output file, which is why I suggested that a page-level suffix could modify `%i` for each file.

> The problem with this particular output is that `%i` changes and has an important information, the page number.

I'm not sure that the page number is always important in all cases.  To me, being able to provide a descriptive suffix for each generated file feels clearer than having page numbers.  

For example, compare this:
```
<some prefix>_top_soldermask.png
<some prefix>_bottom_paste.png
<some prefix>_top_layers.png
etc...
```

to this:

```
<some prefix>_page_01.png
<some prefix>_page_02.png
<some prefix>_page_03.png
```

In the former, it's obvious what the purpose of the file is.  In the latter, you need to open the file to find out what it contains.

I think what I'm trying to say is that `pcb_print` output is very flexible (great!) and can be used to generate many different types of output documentation, not just outputs for `assembly`.  Currently, `%i` always contains the word `assembly` which may not apply in all cases.

cdwilson, 2022-09-07:
Another thought on this: I agree with you that it's important to have an option for including the page number in the output file name, especially when the drawing frame is included in the output.  What I was initially proposing above would totally replace the value of `%i`, so it would not be possible to have a custom suffix _and_ the auto-generated page number at the same time.

As an alternative, you could enable a custom page-level suffix for each page that could include a placeholder for the auto-generated page number, something like `suffix: "_custom_suffix_page_%i"` or `suffix: "_%i_custom_suffix"` (`%i` would be replaced with `01`, `02`, etc).  If the suffix option was defined for a page, it would change the suffix from the default of `assembly_page_NN` to the custom one which could still include the auto-generated page number.  It would also allow the page number to be excluded completely by just removing `%i` from the `suffix:` definition.

This would allow file names like the following:

```
<some prefix>_page_01_top_soldermask.png
<some prefix>_page_02_bottom_paste.png
<some prefix>_page_03_top_layers.png
```

set-soft, 2022-09-08:
I added an option `page_id` that can be used generate different names for each page. You'll most probably want to use something like this:

```yaml
  - name: 'print_front'
    comment: "Experiment"
    type: pcb_print
    options:
      format: 'PNG'
      output: '%f-%I%v%x'
      page_number_as_extension: true
      pages:
        - layers:
            - layer: F.Cu
            - layer: F.Mask
              color: '#14332440'
            - layer: Edge.Cuts
              color: "#004040"
          page_id: F_Cu_color
        - monochrome: true
          layers:
            - layer: F.Cu
            - layer: F.Mask
              color: '#14332440'
            - layer: Edge.Cuts
              color: "#004040"
          page_id: F_Cu_mono
```

A custom `output` pattern is used to remove `assembly_page` and connect the extension in a better way, ids are moved to the `page_number_as_extension` so pages are different and `page_id` is used to get a custom name for each page.

cdwilson, 2022-09-08:
I'm testing out the new changes.  Just want to confirm a couple things about the new changes:
1. With the new `page_id`, it is not possible to have both the auto-generated page number (`01`, `02`, etc) and the custom `page_id` value in the same output filename, right? (for example, `<some prefix>_top_soldermask_page_01.png`)
2. It is not possible to decouple the `page_id` from the extension, right?  (from what I can tell, `page_id` is always fixed to the file extension as `<some prefix>_<page_id>.extension` and cannot be something like `<some prefix>_<page_id>_something_else.extension`)

set-soft, 2022-09-09:
> I'm testing out the new changes. Just want to confirm a couple things about the new changes:
>
> 1. With the new `page_id`, it is not possible to have both the auto-generated page number (`01`, `02`, [...]

Just add `%02d` where you want to get the page number, this is the default value for `page_id`. KiBot always tries to replace the number in the pattern.

> 2. It is not possible to decouple the `page_id` from the extension, right?  (from what I can tell, `page_id` is always fixed to the file extension as `<some prefix>_<page_id>.extension` and cannot be something like `<some prefix>_<page_id>_something_else.extension`)

It depends on what's "_something_else", but in general yes if you use `page_number_as_extension`. Not every name can be created, if you don't want to have `assembly` in the name you have more restrictions. Currently the `%i` values are fixed, and used to differentiate outputs, avoiding overwriting another output.