INTI-CMNB/KiBot #284: [BUG] Dimensions are not included in `dxf` output

cdwilson, 2022-09-06:
**Describe the bug**
When exporting a DXF, the dimension elements are not included in the DXF output.  However, when exporting the same layer as a PDF, the dimension elements are printed correctly.

[image]

[image]

**To Reproduce**
```
git clone --branch dxf-dimensions-not-working git@github.com:cdwilson/test-kibot.git
cd test-kibot
make
# compare DXF vs. PDF outputs in _build/
```

**Expected behavior**
I expect the dimension lines to be included in the DXF output.

**Environment (please complete the following information):**
Where are you running KiBot:
- Using a docker image? `kicad_auto:dev_k6`

set-soft, 2022-09-06:
What do you get using KiCad GUI?

set-soft, 2022-09-06:
I can't even open the DXF files created by KiCad

closed 2022-09-06 by set-soft

cdwilson, 2022-09-06:
Hi @set-soft sorry, you are correct.  I thought this was working in KiCad GUI, but it turns out that it only works if "Plot graphic items using their contours" is unchecked...  When I check this box, it fails in the KiCad GUI also.

[image]

FYI, there is already an issue reported for this: https://gitlab.com/kicad/code/kicad/-/issues/11901

set-soft, 2022-09-07:
Ok, then just use the `polygon_mode: false` option