INTI-CMNB/KiBot #288: [BUG] cannot use `--define` and `--list` together

cdwilson, 2022-09-14:
**Describe the bug**
It appears that the kibot flags `--define` and `--list` are mutually exclusive.  I'm not personally using kibot definitions (`@variable@`) syntax in the YAML `name:` or `comment:` definitions (e.g. `name: some_text_@variable@`), but it may be desirable to allow the `@variable@` replacements to happen before printing out the list of targets.

**Environment (please complete the following information):**
Where are you running KiBot:
- Using a docker image? `kicad_auto:dev_k6`

**Additional context**
Add any other context about the problem here.

set-soft, 2022-09-14:
Thanks! I didn't notice it.
BTW: thanks for the polite *it appears*, but `kibot --help` clearly stated it wasn't possible to use -E and --list together ;-)

closed 2022-09-14 by set-soft

cdwilson, 2022-09-14:
> BTW: thanks for the polite it appears, but kibot --help clearly stated it wasn't possible to use -E and --list together ;-)

Ha, yeah, poor wording on my part...  I assumed they should be allowed together, but when I saw the help I wasn't sure if there was some reason these were mutually exclusive on purpose.  Thx for the quick fix!

commit 23d46b9a (2022-09-14): [Fixed] --define wasn't allowed for --list

- So people couldn't use definitions and use --list

Fixes #288