INTI-CMNB/KiBot #289: requests.exceptions.ConnectionError when downloading specific datasheet

fionn-r, 2022-09-14:
Fill as much as you can, remove the rest

**Describe the bug**
When a component has the following datasheet: https://media.digikey.com/pdf/Data%20Sheets/Samsung%20PDFs/CL_Series_MLCC_ds.pdf you get the following exception:
```
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 699, in urlopen
    httplib_response = self._make_request(
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 445, in _make_request
    six.raise_from(e, None)
  File "<string>", line 3, in raise_from
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 440, in _make_request
    httplib_response = conn.getresponse()
  File "/usr/lib/python3.9/http/client.py", line 1347, in getresponse
    response.begin()
  File "/usr/lib/python3.9/http/client.py", line 307, in begin
    version, status, reason = self._read_status()
  File "/usr/lib/python3.9/http/client.py", line 276, in _read_status
    raise RemoteDisconnected("Remote end closed connection without"
http.client.RemoteDisconnected: Remote end closed connection without response

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/requests/adapters.py", line 439, in send
    resp = conn.urlopen(
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 755, in urlopen
    retries = retries.increment(
  File "/usr/lib/python3/dist-packages/urllib3/util/retry.py", line 532, in increment
    raise six.reraise(type(error), error, _stacktrace)
  File "/usr/lib/python3/dist-packages/six.py", line 718, in reraise
    raise value.with_traceback(tb)
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 699, in urlopen
    httplib_response = self._make_request(
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 445, in _make_request
    six.raise_from(e, None)
  File "<string>", line 3, in raise_from
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 440, in _make_request
    httplib_response = conn.getresponse()
  File "/usr/lib/python3.9/http/client.py", line 1347, in getresponse
    response.begin()
  File "/usr/lib/python3.9/http/client.py", line 307, in begin
    version, status, reason = self._read_status()
  File "/usr/lib/python3.9/http/client.py", line 276, in _read_status
    raise RemoteDisconnected("Remote end closed connection without"
urllib3.exceptions.ProtocolError: ('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/bin/kibot", line 16, in <module>
    main()
  File "/usr/lib/python3/dist-packages/kibot/__main__.py", line 323, in main
    generate_outputs(outputs, args.target, args.invert_sel, args.skip_pre, args.cli_order, args.no_priority)
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 470, in generate_outputs
    run_output(out, dont_stop)
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 397, in run_output
    out.run(get_output_dir(out.dir, out))
  File "/usr/lib/python3/dist-packages/kibot/out_download_datasheets.py", line 161, in run
    self.options.run(output_dir)
  File "/usr/lib/python3/dist-packages/kibot/out_download_datasheets.py", line 125, in run
    name = self.download(c, ds, output_dir, name, known)
  File "/usr/lib/python3/dist-packages/kibot/out_download_datasheets.py", line 68, in download
    r = requests.get(ds, allow_redirects=True, headers={'User-Agent': USER_AGENT}, timeout=20)
  File "/usr/lib/python3/dist-packages/requests/api.py", line 76, in get
    return request('get', url, params=params, **kwargs)
  File "/usr/lib/python3/dist-packages/requests/api.py", line 61, in request
    return session.request(method=method, url=url, **kwargs)
  File "/usr/lib/python3/dist-packages/requests/sessions.py", line 542, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/lib/python3/dist-packages/requests/sessions.py", line 655, in send
    r = adapter.send(request, **kwargs)
  File "/usr/lib/python3/dist-packages/requests/adapters.py", line 498, in send
    raise ConnectionError(err, request=request)
requests.exceptions.ConnectionError: ('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))
```

The link is fine to access with the browser.

I can also reproduce it with the following code:
```
import requests
USER_AGENT = 'Mozilla/5.0 (Windows NT 5.2; rv:2.0.1) Gecko/20100101 Firefox/4.0.1'
r = requests.get("https://media.digikey.com/pdf/Data Sheets/Samsung PDFs/CL_Series_MLCC_ds.pdf", allow_redirects=True, headers={'User-Agent': USER_AGENT}, timeout=20) 
```

**To Reproduce**
Set the above link to be the datasheet for a part then use the following output:
```
- name: 'download_datasheets'
    comment: 'Downloads the datasheets for the project'
    type: 'download_datasheets'
    dir: 'datasheets'
    options:
        dnf: true # Include the DNF components
```

**Expected behavior**
Either the datasheet downloads or fails with a warning.

**Environment (please complete the following information):**
Docker image on Windows
```
docker run -it --rm -v ${PWD}:/powerboard -w /powerboard -e PCB_VERSION="v1.4.0" setsoft/kicad_auto:ki6 kibot --version
KiBot 1.2.0 - Copyright 2018-2022, Salvador E. Tropea/INTI/John Beard - License: GPL v3+
```

fionn-r, 2022-09-14:
I've submitted a pull request #290 (I think I did it right, first time using Github for them...) that catches the error and logs it. I am not sure why it is failing though

set-soft, 2022-09-14:
Hi @fionn-r !
Thanks for the patch. I agree that connection errors should be reported and won't stop the process.
Note that the last release of KiBot (1.3.0) uses a newer user agent id, just to avoid the download problem.

closed 2022-09-14 by set-soft

fionn-r, 2022-09-14:
Ah, I wasn't sure which to do to since I had my issue on the `v2_stable_k6` branch, not on master. Thanks for sorting it out!

commit 446364fb (2022-09-14): [Fixed] Download datasheet abort on connection error

- Now it isn't fatal
- This is PR #290 applied to master

Closes #289