INTI-CMNB/KiBot #291: [FEATURE] import global filters from multiple config files

cdwilson, 2022-09-14:
As far as I can tell, I don't think KiBot currently supports the ability to import/merge global filters from multiple config files.  I'm trying to move some of my generic global filters into a `base.kibot.yaml` which can be reused across projects.  However, it seems like only the last imported `global` section overwrites all other global sections, both imported from other config files and in the current file.

For example, given the following YAML, the global filters from `project.kibot.yaml` replace the global filters from `base.kibot.yaml` and the global filters in the current file.  I'd like to have the ability to import _all_ the global filters from `base.kibot.yaml` and `project.kibot.yaml` and the current file, effectively merging them all together when imported.

```yaml
import:
  - file: base.kibot.yaml
    global: all
  - file: project.kibot.yaml
    global: all

global:
  filters:
    # project-specific warnings
    - number: 13
      regex: "`REF\\*\\*` component in board, but not in schematic"
```

I realize this is probably tricky because you can't merge global options that aren't lists (for example `restore_project: true`).  It feels like the order of precedence for non-list items should be:

1. (1st precedence) Value defined in the current config file
2. (2nd precedence) Imported values, where the last imported file has highest precedence.

However, I noticed that the documentation says:

> Also note that imported globals has more precedence than the ones defined in the same file.

This seems backwards to me at first glance.  Is there a specific reason why the import has higher precedence than the value defined in the current file? (seems like you would want to have the ability to import a "generic" set of global options, and then override a specific subset in the current file)

closed 2022-09-15 by set-soft

set-soft, 2022-09-15:
> why the import has higher precedence than the value defined in the current file?

I think this is because the code is takes the last declaration found, and this is the one in the current file.
Now you can choose to change it using an option in the `kibot` section, not a global option for obvious reasons.
I'm not making it a default to avoid conflicts with files relying on the current behavior.

> As far as I can tell, I don't think KiBot currently supports the ability to import/merge global filters

The last patch should solve it. Now lists are concatenated, but the last one adds its values at the beginning of the list. So they will be used first.
Check if this solves your problem, farther adjusts could be needed.

cdwilson, 2022-09-16:
Thanks, I tested this out and I'm able to import global filters from multiple sources.  Looks like it's working as expected.  I'll let you know if I run into any issues.

commit 5c825151 (2022-09-15): [Import] Global options are now "merged"

Close #291