INTI-CMNB/KiBot #292: [BUG] global dir value is not used for ERC/DRC outputs

cdwilson, 2022-09-15:
**Describe the bug**
The documentation for the default `dir` global option says:

> The default `dir` value for any output is `.`.

I assumed that the `*drc.txt` & `*erc.txt` outputs would use this value, i.e. the path to these reports would be:

`OUT_DIR` / `dir` / `*drc.txt`

However, it appears that these outputs ignore the  default `dir` global option and are output in:

`OUT_DIR` / `*drc.txt`

**Expected behavior**
I expected the ERC/DRC reports to be output in the `dir` path.

**Environment (please complete the following information):**
Where are you running KiBot:
- Using a docker image? `kicad_auto:dev_k6`

set-soft, 2022-09-15:
Technically not a bug, preflights aren't outputs (mushrooms aren't plants ;-)
But It sounds logic.
Now this is the default, but can be disabled.

closed 2022-09-15 by set-soft

cdwilson, 2022-09-15:
@set-soft I'm now getting the following errors when running the ERC/DRC:

```
ERROR:PCB_fabrication/ModST_Mouse_v1_100037_rev_B (kibot.pre_run_erc - pre_run_erc.py:45)
- Running the ERC
ERROR:PCB_fabrication/ModST_Mouse_v1_100037_rev_B (kibot.pre_run_drc - pre_run_drc.py:44)
- Running the DRC
```

Debug output isn't giving me anything more helpful:
```
ERROR:PCB_fabrication/ModST_Mouse_v1_100037_rev_B (kibot.pre_run_erc - pre_run_erc.py:45)
DEBUG:ERC report: /workdir/kibot/_build/PCB_fabrication/ModST_Mouse_v1_100037_rev_B/ModST_Mouse_v1_100037_rev_B_erc.txt (kibot.pre_run_erc - pre_run_erc.py:56)
- Running the ERC
DEBUG:Executing: /usr/bin/eeschema_do -r -vvv run_erc -o /workdir/kibot/_build/PCB_fabrication/ModST_Mouse_v1_100037_rev_B/ModST_Mouse_v1_100037_rev_B_erc.txt /workdir/ModST_Mouse.kicad_sch /workdir/kibot/_build (kibot - kiplot.py:163)
DEBUG:Command line: /usr/bin/eeschema_do -r -vvv run_erc -o /workdir/kibot/_build/PCB_fabrication/ModST_Mouse_v1_100037_rev_B/ModST_Mouse_v1_100037_rev_B_erc.txt /workdir/ModST_Mouse.kicad_sch /workdir/kibot/_build (kibot - kiplot.py:165)
```

The ERC/DRC seems to be running correctly (I see the reports in the output dir) and there are zero violations.

Any idea what might be causing this?

cdwilson, 2022-09-15:
Oh, looks like it might be this:

https://github.com/INTI-CMNB/KiBot/blob/488f2dcbc27d1d457c29a98a552256a55fc10ff0/kibot/pre_run_erc.py#L45

https://github.com/INTI-CMNB/KiBot/blob/488f2dcbc27d1d457c29a98a552256a55fc10ff0/kibot/pre_run_drc.py#L44

set-soft, 2022-09-16:
Oops! I forgot to remove them. Now fixed.

commit 488f2dcb (2022-09-15): [Global options] Now the global `dir` option also applies to the preflights

- Can be disabled using `use_dir_for_preflights`.

Closes #292