INTI-CMNB/KiBot #295: Apply preflight to diff: outputs

mdeweerd, 2022-09-15:
When applying   check_zone_fills: true` as a pre-flight options, the following real issue (via merges into incompatible zone):

[image]

is fixed to:
[image]

However, the diff output does not show any difference and show the original filling.

One of the main purposes of the differences is to detect if the actual manufacturing output changed versus the previous version.

To reproduce this, it should be sufficient to fill the zones in a design, and then move/drag one of the vias and save the design in that state without running any DRC with zone fill.

_Also see https://github.com/INTI-CMNB/KiBot/discussions/294_

closed 2022-09-16 by set-soft

set-soft, 2022-09-16:
Hi @mdeweerd !
I added a new type named `current` it means *the current PCB/SCH in memory*.
Applying the preflight to `file` looks awkward to me, because this isn't the real content of the file (unless you ask for saving it).
And now `current` is the default for the new file.
I tried using:

```yaml
kibot:
  version: 1

preflight:
  check_zone_fills: true

outputs:
  - name: 'diff_pcb'
    comment: "PCB difference with git HEAD"
    type: diff
    layers: ['F.Cu', 'B.Cu']
    options:
      cache_dir: .cache
      # pcb: false
```

And running:

```
src/kibot -c tests/yaml_samples/diff_git_zones_1.kibot.yaml -b tests/board_samples/kicad_6/zone-refill.kicad_pcb -e tests/board_samples/kicad_6/bom.kicad_sch -d pp
```

And got the zone refill:

[image]

set-soft, 2022-09-16:
Hi @mdeweerd !
I think what you want is something like this:

```yaml
kibot:
  version: 1

preflight:
  check_zone_fills: true

outputs:
  - name: 'diff_pcb_show'
    comment: "Show differences for the zone fill"
    type: diff
    layers: copper
    options:
      cache_dir: .cache
      old: ''
      old_type: file
      new_type: current

  - name: 'diff_pcb_check'
    comment: "Check the zone fill doesn't generate important changes"
    type: diff
    layers: copper
    options:
      cache_dir: .cache
      diff_mode: stats
      threshold: 100
      old: ''
      old_type: file
      new_type: current
```

This generates an stereo diff first, useful to check it visually, and then checks if the zone refill generates more than 100 points of difference. This threshold is arbitrary and you should experiment to determine the best value for your project.
Here I'm checking the current file, but you can use the git HEAD.

commit a8e48031 (2022-09-16): [Diff] Added mechanism to specify the PCB/Schematic in memory

- Now this is the default

Closes #295