INTI-CMNB/KiBot #296: [BUG] Can't filter warnings about missing imports

cdwilson, 2022-09-15:
**Describe the bug**
There are some warnings that I'm unable to filter related to missing imports.

For example, assume I have a top level YAML config `bom.kibot.yaml` with the following imports:

```
# bom.kibot.yaml
import:
  - file: base.kibot.yaml
    outputs: [bom_csv, bom_html, bom_xlsx, ibom]
    preflights: all
    filters: all
    variants: all
    global: all
  - file: project.kibot.yaml
    outputs: all
    preflights: all
    filters: all
    variants: all
    global: all
```

The `base.kibot.yaml` defines some filters in case there are missing sections from the `project.kibot.yaml`:

```
# base.kibot.yaml
global:
  filters:
    - number: 68
      regex: "No filters found in"
    - number: 56
      regex: "No outputs found in"
    - number: 94
      regex: "No preflights found in"
    - number: 69
      regex: "No variants found in"
    - number: 72
      regex: "No globals found in"
```

If `project.kibot.yaml` doesn't have anything defined (other than the header), I get the following warnings:

```
WARNING:(W056) No outputs found in `kibot/config/project.kibot.yaml` (kibot - config_reader.py:290)
WARNING:(W094) No preflights found in `kibot/config/project.kibot.yaml` (kibot - config_reader.py:311)
WARNING:(W068) No filters found in `kibot/config/project.kibot.yaml` (kibot - config_reader.py:332)
WARNING:(W069) No variants found in `kibot/config/project.kibot.yaml` (kibot - config_reader.py:353)
WARNING:(W072) No globals found in `kibot/config/project.kibot.yaml` (kibot - config_reader.py:378)
```

**Expected behavior**
I assumed that since the filters were defined in `base.kibot.yaml` and imported into `bom.kibot.yaml`, I should not see any warnings in the output.

**Environment (please complete the following information):**
Where are you running KiBot:
- Using a docker image? `kicad_auto:dev_k6`

set-soft, 2022-09-16:
Hi @cdwilson !
It doesn't work like you think. Things are imported first, and applied after all things are collected. Applying things on-the-fly could generate unintended side effects, so I don't like the idea of applying the imports before we finished importing.
The real solution here is to avoid the warnings in the first place. Why do you need to ask for filters from a file that doesn't contain filters? 
I understand you are creating generic templates, but the use looks dangerous.
In any case, I see that we could have other sources of warnings issued before even having the chance to define the warnings filters, so I added a command line option to list warnings that must be excluded.

closed 2022-09-16 by set-soft

cdwilson, 2022-09-16:
Got it, thanks for the explanation.

> Why do you need to ask for filters from a file that doesn't contain filters?

I was trying to see if it's possible to split out project-specific config vs. generic config in a way that's reusable across multiple projects.  Since each project-specific file may or may not have board-specific filters (e.g. for ERC/DRC), I was trying to see if it's possible to just globally silence the warnings once from the generic config file (instead of having to do it for each project).  As you mentioned, this is probably not a good approach... :)  Right now I'm just experimenting ways to reduce the amount of duplication and increase reuse across the YAML files as much as possible.

> I added a command line option to list warnings that must be excluded

I'll check out the CLI option instead

> I understand you are creating generic templates, but the use looks dangerous.

By "dangerous" are you referring to suppressing the warnings about missing imports or something else?  I'm still new with KiBot, so I want to make sure I'm not doing something really dumb here.

cdwilson, 2022-09-19:
Hi @set-soft I tried out the CLI option, but when I pass the `--no-warn="68,56,94,69,72"` flag I'm getting an error like the following:

```
Traceback (most recent call last):
  File "/usr/local/bin/kibot", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.9/dist-packages/kibot/__main__.py", line 371, in main
    outputs = cr.read(cf_file)
  File "/usr/local/lib/python3.9/dist-packages/kibot/config_reader.py", line 526, in read
    self._parse_import(v1, fstream.name)
  File "/usr/local/lib/python3.9/dist-packages/kibot/config_reader.py", line 457, in _parse_import
    all_collected.outputs.extend(self._parse_import_outputs(outs, explicit_outs, fn_rel, data, imported))
  File "/usr/local/lib/python3.9/dist-packages/kibot/config_reader.py", line 290, in _parse_import_outputs
    logger.warning(W_NOOUTPUTS+"No outputs found in `{}`".format(fn_rel))
  File "/usr/local/lib/python3.9/dist-packages/kibot/log.py", line 96, in warning
    if (f.number == number or f.error == id) and f.regex.search(buf):
AttributeError: 'SimpleFilter' object has no attribute 'error'
```

set-soft, 2022-09-21:
> By "dangerous" are you referring to suppressing the warnings about missing imports or something else? I'm still new with KiBot, so I want to make sure I'm not doing something really dumb here.

Yes, I mean you could think you are importing something, but wonder why you get nothing ;-)

> ... I'm getting an error like the following: ...

I think this is fixed now.

commit 252f10e8 (2022-09-16): Added command line option to specify warnings to be excluded.

- Useful for warnings issued before applying the global options
  (i.e during import).

Fixes #296