INTI-CMNB/KiBot #298: Missing library signature when loading Mechanical.kicad_sym

mmutlu-any, 2022-09-19:
A K5 project merged to K6. Running with `-vvv`
```
DEBUG:Loading library `/usr/share/kicad/symbols/Mechanical.kicad_sym` (kibot - v5_sch.py:684)
ERROR:Trace stack: (kibot - error.py:33)
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 246, in load_any_sch
    sch.load_libs(file)
  File "/usr/lib/python3/dist-packages/kibot/kicad/v5_sch.py", line 1701, in load_libs
    o.load(v, k, self.comps_data)
  File "/usr/lib/python3/dist-packages/kibot/kicad/v5_sch.py", line 689, in load
    raise SchLibError('Missing library signature', line, f)
ERROR:At line 1 of `/usr/share/kicad/symbols/Mechanical.kicad_sym`: Missing library signature (kibot - kiplot.py:251)
ERROR:Line content: `(kicad_symbol_lib (version 20211014) (generator kicad_symbol_editor)` (kibot - kiplot.py:252)
```

I tried it with following version and v1.2 on 2 different Ubuntu 20.04.
```
KiBot installation checker

Core:
Linux: 5.4.0.125 (Linux user 5.4.0-125-generic #141-Ubuntu SMP Wed Aug 10 13:42:03 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux)
Python: 3.8.10 (default, Jun 22 2022, 20:18:18)  [GCC 9.4.0]
KiCad: 6.0.7-f9a2dced07~116~ubuntu20.04.1
Kibot: 1.3.0

Modules:
Colorama: 0.4.3
LXML: 4.5.0
PyYAML: 5.3.1
QRCodeGen: Ok
Requests: 2.22.0
XLSXWriter: 1.1.2

Tools:
Ghostscript: 9.50.0 (9.50)
Git: 2.25.1 (git version 2.25.1)
ImageMagick: 6.9.10.23 (Version: ImageMagick 6.9.10-23 Q16 x86_64 20190101 https://imagemagick.org)
Interactive HTML BoM: 2.5.0 (v2.5.0)
KiBoM: 1.8.0 (KiBOM Version: 1.8.0)
KiCad Automation tools (kiauto): 2.0.4 (pcbnew_do 2.0.4 - Copyright 2018-2022, INTI/Productize SPRL - License: Apache)
KiCad PCB/SCH Diff (kidiff): 2.4.1 (kicad-diff.py 2.4.1 - Copyright 2020-2022, INTI/Salvador E. Tropea - License:)
KiCost: 1.1.8 (KiCost v1.1.8)
Pandoc: 2.5.0 (pandoc 2.5)
PcbDraw: 0.9.0.3 (PcbDraw 0.9.0-3)
RAR: 5.50.0 (RAR 5.50   Copyright (c) 1993-2017 Alexander Roshal   11 Aug 2017)
RSVG tools: 2.48.9 (rsvg-convert version 2.48.9)
```
I can generate kibot outputs for my other projects that are also mostly merged from K5 to K6. I am suspecting a project specific issue. Would you have any tips on what I can check?

set-soft, 2022-09-19:
Hi @mmutlu-any !
If the new project is in KiCad 6 format remove all KiCad 5 files. It looks like you are loading a `.sch` (KiCad 5) file and you have KiCad 6 libs. Move away all `.sch` files and only use the `.kicad_sch` files.

mmutlu-any, 2022-09-20:
Thank you very^2 much! Silly of me. I forgot one .sch file in the repository.

closed 2022-09-20 by mmutlu-any