INTI-CMNB/KiBot #299: [HOW] Make KiBot close a PcbNew warning dialog and continue

matthiesenj, 2022-09-21:
**What do you want to achieve?**
I have a custom NetTie which sits on the inner layers of a multilayer PCB. This is achieved by editing the footprint with a text editor.
This causes Kicad to pop up a dialog complaining about a pad being present on an invalid layer ('could not find valid layer for pad'), but unfortunately there's no other way of putting a net-tie on internal layers.

The dialog causes KiBot to fail with the following message for each KiBot output that deals with opening the pcb in PcbNew :
> ERROR:Unknown KiCad dialog: pcbnew Warning (pcbnew_do - interposer.py:392) (kibot - kiplot.py:124)
> ERROR:Potential dialog messages: {'Details', '14:58:11', 'footprint NT401, pad 2: could not find valid layer for pad', 'OK', 'footprint NT401, pad 1: could not find valid layer for pad', '14:58:12', 'H'} (pcbnew_do - interposer.py:393) (kibot - kiplot.py:124)
> ERROR:/usr/bin/pcbnew_do returned 9 (kibot - out_netlist.py:62)

I'm wondering if you could suggest a way I could hack KiBot in order to ignore/close this dialog?

This is the content of the bottom of the interposer log:
```
GTK:Window Title:pcbnew Warning
KiAuto:Waiting for `['PANGO:']` starts=True times=1
GTK:Label Set Text 2:footprint NT401, pad 1: could not find valid layer for pad
PANGO:footprint NT401, pad 1: could not find valid layer for pad
KiAuto:match
KiAuto:Waiting for `['PANGO:']` starts=True times=1
* wrapping button label
GTK:Button Label:gtk-ok
PANGO:OK
KiAuto:match
KiAuto:Waiting for `['PANGO:']` starts=True times=1
PANGO:Details
KiAuto:match
KiAuto:Waiting for `['PANGO:']` starts=True times=1
PANGO:H
KiAuto:match
KiAuto:Waiting for `['PANGO:']` starts=True times=1
PANGO:footprint NT401, pad 2: could not find valid layer for pad
KiAuto:match
KiAuto:Waiting for `['PANGO:']` starts=True times=1
PANGO:H
KiAuto:match
KiAuto:Waiting for `['PANGO:']` starts=True times=1
PANGO:14:43:15
KiAuto:match
KiAuto:Waiting for `['PANGO:']` starts=True times=1
PANGO:footprint NT401, pad 1: could not find valid layer for pad
KiAuto:match
KiAuto:Waiting for `['PANGO:']` starts=True times=1
PANGO:14:43:15
KiAuto:match
KiAuto:Waiting for `['PANGO:']` starts=True times=1
PANGO:footprint NT401, pad 2: could not find valid layer for pad
KiAuto:match
KiAuto:Waiting for `['PANGO:']` starts=True times=1
PANGO:14:43:15
KiAuto:match
KiAuto:Waiting for `['PANGO:']` starts=True times=1
PANGO:footprint NT401, pad 1: could not find valid layer for pad
KiAuto:match
```

**Environment (please complete the following information):**
Where are you running KiBot:
- Using a docker image `kicad_auto:latest`

set-soft, 2022-09-21:
Hi @matthiesenj !
This should be handled by KiAuto.
Do you have a small example that generates this dialog? I tried using some bizarre layers for a pad and couldn't get it.
The INTI-CMNB/KiAuto@0c8450e2bdba6259f773ae89a5f36593ff6fc81d patch should handle this case, but I couldn't verify it.
Can you try the code from git?

matthiesenj, 2022-09-21:
Hi,
Please find a KiCad5 example attached. It is just an empty project, where I
changed the pcb to 4 layers and added the custom net-tie footprint in the
pcb editor. It should show the warning when you open the pcb (and also
continuously when manipulating the footprint).

I will have a look at how to test the code, I'm using KiBot using your
kicad_auto:latest docker image because I'm using a Windows computer (KiBot
doesn't work well on Windows as you probably know), so I guess I will have
to replace the interposer.py file in the container and see what happens.

KiBot reports version 1.3.0 in the container - I took a look at the KiBot
repository tag 1.3.0, but found no interposer.py file there.

Best regards,
Jesper

matthiesenj, 2022-09-22:
Your suggested patch didn't work, so I added a traceback dump to the unknown_dialog() function:

>   File "/usr/lib/python3/dist-packages/kiauto/interposer.py", line 231, in wait_kicad_ready_i
>     new_res = wait_queue(cfg, 'GLX:Swap', starts=True, timeout=0.1, do_to=False, kicad_can_exit=kicad_can_exit)
>   File "/usr/lib/python3/dist-packages/kiauto/interposer.py", line 197, in wait_queue
>     unknown_dialog(cfg, title)
>   File "/usr/lib/python3/dist-packages/kiauto/interposer.py", line 389, in unknown_dialog
>     traceback.print_stack() (kibot - kiplot.py:124)
> ERROR:Unknown KiCad dialog: pcbnew Warning (pcbnew_do - interposer.py:395) (kibot - kiplot.py:124)
> ERROR:Potential dialog messages: {'Details', 'H', 'OK', '09:15:21', 'footprint NT401, pad 2: could not find valid layer for pad', '09:15:15', 'footprint NT401, pad 1: could not find valid layer for pad'} (pcbnew_do - interposer.py:396) (kibot - kiplot.py:124)
> ERROR:/usr/bin/pcbnew_do returned 9 (kibot - out_netlist.py:62)

As can be seen, the call originates from the wait_queue() function.

set-soft, 2022-09-22:
Hi @matthiesenj !
Thanks for the example file and the backtrace.
I added correct support for this dialog, even when no interposer is available.
This fix is included in KiAuto 2.0.5, now part of `latest`.
I added your example as a regression test for KiCad 5 (KiCad 6 doesn't see it as a problem).

closed 2022-09-22 by set-soft