INTI-CMNB/KiBot #300: [FEATURE] Add "populate" column/value to 'position' output

matthiesenj, 2022-09-26:
**Is your feature request related to a problem? Please describe.**
My assembler would like all footprints/ref designators mentioned in the position file for assembly inspection purposes.

**Describe the solution you'd like**
A column similar to the 'bom' output "Footprint Populate" column could solve this.

**Describe alternatives you've considered**
Alternatively to an extra column, maybe both 'bom' and 'position' outputs should support putting DNF/DNP/etc in the "Value" column.

set-soft, 2022-09-26:
Hi @matthiesenj !
About the column: Did you try using the `bom` output to achieve it? It can currently generate XYRS files, so you can create position files using the BoM. Of course they will be CSV style.
About the DNP: Did you try using the `exclude_value` option of the `generic` filters? It can exclude components with `DNP` as value.

matthiesenj, 2022-09-26:
Hi @set-soft 
I managed to use the bom output to get the desired position format - it took some fiddling with custom filters to get what I want, though.
My comment about the value column was aimed at the output - instead of having a populate yes/no column in the csv-file, DNF could be marked in the value column. But just ignore this, it is not very important.

closed 2022-09-26 by matthiesenj

set-soft, 2022-09-26:
Hi @matthiesenj !
> I managed to use the bom output to get the desired position format - it took some fiddling with custom filters to get what I want, though.

Consider contributing it as an example for other users needing the same columns.

> My comment about the value column was aimed at the output - instead of having a populate yes/no column in the csv-file, DNF could be marked in the value column. But just ignore this, it is not very important.

Ok, we need some filter to apply field transformations.

matthiesenj, 2022-09-26:
> Hi @matthiesenj !
> Consider contributing it as an example for other users needing the same columns.
I would be happy to - should I just add it here as an issue comment?

set-soft, 2022-09-26:
> I would be happy to - should I just add it here as an issue comment?

Yes, but if you know how to handle github repos you could also do a PR with a file added to *docs/samples/*.
In this way the contribution will be more clear.
But feel free to paste it here if you want.

matthiesenj, 2022-09-26:
> > I would be happy to - should I just add it here as an issue comment?
> 
> Yes, but if you know how to handle github repos you could also do a PR with a file added to _docs/samples/_. In this way the contribution will be more clear.

I added a PR.